coolsnowwolf/lede #4021: v2ray 4.23.1 error when makefile

maomaoinchina, 2020-03-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：no matter what I run the make, the v2ray still has error for below reason

### (2) 路由器型号和固件版本

A：mips64 octeonplus

### (3) 详细日志

A：Cleaning module download cache (golang/go#27455)

mv -f /home/vagrant/lede/build_dir/target-mips64_octeonplus_64_musl/v2ray-core-4.23.1/.go_work/build/bin/linux_mips64/main /home/vagrant/lede/build_dir/target-mips64_octeonplus_64_musl/v2ray-core-4.23.1/.go_work/build/bin/linux_mips64/v2ray
/home/vagrant/lede/staging_dir/host/bin/upx --lzma --best /home/vagrant/lede/build_dir/target-mips64_octeonplus_64_musl/v2ray-core-4.23.1/.go_work/build/bin/linux_mips64/v2ray
                       Ultimate Packer for eXecutables
                          Copyright (C) 1996 - 2018
UPX 3.95        Markus Oberhumer, Laszlo Molnar & John Reiser   Aug 26th 2018

        File size         Ratio      Format      Name
   --------------------   ------   -----------   -----------
upx: /home/vagrant/lede/build_dir/target-mips64_octeonplus_64_musl/v2ray-core-4.23.1/.go_work/build/bin/linux_mips64/v2ray: UnknownExecutableFormatException

Packed 0 files.
Makefile:307: recipe for target '/home/vagrant/lede/build_dir/target-mips64_octeonplus_64_musl/v2ray-core-4.23.1/.built' failed
make[3]: *** [/home/vagrant/lede/build_dir/target-mips64_octeonplus_64_musl/v2ray-core-4.23.1/.built] Error 1
make[3]: Leaving directory '/home/vagrant/lede/package/lean/v2ray'
time: package/lean/v2ray/compile#51.60#26.01#50.05
package/Makefile:111: recipe for target 'package/lean/v2ray/compile' failed
make[2]: *** [package/lean/v2ray/compile] Error 2
make[2]: Leaving directory '/home/vagrant/lede'
package/Makefile:107: recipe for target '/home/vagrant/lede/staging_dir/target-mips64_octeonplus_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/vagrant/lede/staging_dir/target-mips64_octeonplus_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/vagrant/lede'
/home/vagrant/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

EstFoxs, 2020-03-26:
报错，跟楼主一样

maomaoinchina, 2020-03-26:
@EstFoxs 搞了半天终于明白了。
make menuconfig 选中SSRplus以及下阶的V2ray。保存退出后，需要手动关闭v2ray 的UPX压缩。
在LEDE根目录下面的那个config文件，
CONFIG_V2RAY_COMPRESS_UPX=y 改成 CONFIG_V2RAY_COMPRESS_UPX=n
保存退出然后 正常make就通过了。
貌似是UPX不支持v2ray core的压缩吧

closed 2020-03-26 by maomaoinchina

kktt007, 2020-04-26:
最近也遇到这问题了,不知道怎么解决的好.不会改
[image]