coolsnowwolf/lede #4023: 如何将xl2tpd离线部署进安装包

Tony-LYX, 2020-03-26:
不少学校里使用的是无加密的L2TP作为上网认证的手段，然而在openwrt并不天生支持。
（用过梅林和Padavan的网友应该知道我在说什么，是连接外部wan时的一个选项）
如果想要安装相关的ipk需要先联网，然而这就出现了一个循环：
没xl2tpd->上不了网->没法自动联网部署xl2tpd->上不了网

解决的唯一方法就是在编译时就部署进去，但是我看咱们这个app库里也没有xl2tpd，不知道该怎么部署？

dongjuanyong, 2020-03-26:
https://github.com/coolsnowwolf/packages/tree/master/net/xl2tpd
feeds里面明明有
还是说你不明白怎么在make menuconfig里选择build-in的package？

Tony-LYX, 2020-03-26:
> https://github.com/coolsnowwolf/packages/tree/master/net/xl2tpd
> feeds里面明明有
> 还是说你不明白怎么在make menuconfig里选择build-in的package？

这个xl2tpd是在coolsnowwolf/packages下有，但是在coolsnowwolf/lede下却没有欸。我在meneconfig下只看到zerotier等等app，没看到x开头的。
我直接git clone lede的话，是没有xl2tpd的，有没有办法引用coolsnowwolf/packages里的内容呢

liwenjie119, 2020-03-26:
> > https://github.com/coolsnowwolf/packages/tree/master/net/xl2tpd
> > feeds里面明明有
> > 还是说你不明白怎么在make menuconfig里选择build-in的package？
> 
> 这个xl2tpd是在coolsnowwolf/packages下有，但是在coolsnowwolf/lede下却没有欸。我在meneconfig下只看到zerotier等等app，没看到x开头的。
> 我直接git clone lede的话，是没有xl2tpd的，有没有办法引用coolsnowwolf/packages里的内容呢

make menuconfig 使善用 / 键进行搜索。xl2tpd包在network-VPN下。可能还需要选中luci-app-pppoe-relay，不过没试过

dongjuanyong, 2020-03-26:
[项目首页](https://github.com/coolsnowwolf/lede)都不看的么：

> ./scripts/feeds update -a 
./scripts/feeds install -a
make menuconfig

Tony-LYX, 2020-03-26:
> [项目首页](https://github.com/coolsnowwolf/lede)都不看的么：
> 
> > ./scripts/feeds update -a
> > ./scripts/feeds install -a
> > make menuconfig

这个配置页面昨天已经进去了，大部分的也都配置过了，只是没在菜单找到xl2tpd的app。
我先按楼上大哥的说法找一找。

Tony-LYX, 2020-03-26:
> make menuconfig 使善用 / 键进行搜索。xl2tpd包在network-VPN下。可能还需要选中luci-app-pppoe-relay，不过没试过

非常感谢，已经搜索到了。依赖的问题我再多研究一下。

Tony-LYX, 2020-03-26:
> [项目首页](https://github.com/coolsnowwolf/lede)都不看的么：
> 
> > ./scripts/feeds update -a
> > ./scripts/feeds install -a
> > make menuconfig

依照楼上大哥的说法，我已经搜到了xl2tpd。
还想问一下Network 和Luci app这两个菜单是什么关系呢？Luci app只是一个界面吗还是说两者是联动的？

liwenjie119, 2020-03-26:
> > [项目首页](https://github.com/coolsnowwolf/lede)都不看的么：
> > > ./scripts/feeds update -a
> > > ./scripts/feeds install -a
> > > make menuconfig
> 
> 依照楼上大哥的说法，我已经搜到了xl2tpd。
> 还想问一下Network 和Luci app这两个菜单是什么关系呢？Luci app只是一个界面吗还是说两者是联动的？

luci是界面，当然选中后肯定有必须的一些依赖包被选中。但是有一些非必须的包需要手动去选。不知道你是否用过ddns，选luci-ddns后，同样也得手动选择network-IP Addresses and Names下的特定ddns-scripts使用。

Tony-LYX, 2020-03-26:
> luci是界面，当然选中后肯定有必须的一些依赖包被选中。但是有一些非必须的包需要手动去选。不知道你是否用过ddns，选luci-ddns后，同样也得手动选择network-IP Addresses and Names下的特定ddns-scripts使用。

我在使用help时能够看到里面有两行：depend 和select，其中有些是n的内容是不是一定要求我们手动去寻找并勾选？

Tony-LYX, 2020-04-08:
@coolsnowwolf 请管理员删除此issue，我已经没有问题了，但是我没有权限删除

closed 2020-07-04 by coolsnowwolf