coolsnowwolf/lede #4027: make download v=s 提示 build failed

a1392429292, 2020-03-26:
编译的过程中，当输入 make download v=s 后
在所有暗黄色的列表中，出现了2个白色的字行，如下：
make tools/download: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
出现这样的提示，请问我需要关注吗？还是直接无视？输入  make -j1 V=s  开始编译？

martin0607, 2020-03-26:
“编译失败”
你都不看看失败的原因。。。

a1392429292, 2020-03-26:
> “编译失败”
> 你都不看看失败的原因。。。

提示我都写上了，可否高人一步到位，告诉我，这串提示是什么意思吗？原因是？？？

EnnawYang, 2020-03-26:
> > “编译失败”
> > 你都不看看失败的原因。。。
> 
> 提示我都写上了，可否高人一步到位，告诉我，这串提示是什么意思吗？原因是？？？

谷歌翻译 请

martin0607, 2020-03-26:
下载失败。。。网络问题

a1392429292, 2020-03-26:
> > > “编译失败”
> > > 你都不看看失败的原因。。。
> > 
> > 
> > 提示我都写上了，可否高人一步到位，告诉我，这串提示是什么意思吗？原因是？？？
> 
> 谷歌翻译 请

> > > “编译失败”
> > > 你都不看看失败的原因。。。
> > 
> > 
> > 提示我都写上了，可否高人一步到位，告诉我，这串提示是什么意思吗？原因是？？？
> 
> 谷歌翻译 请

生成工具/下载：生成失败。请使用-j1 V=s或V=sc重新运行make以获得更高的详细级别，以查看发生了什么

谷歌翻译就是这样说的，那么，我需要如何做呢？重新来一遍？重新输入  make download v=s 这个命令？还是直接 make -j1 V=s ？

a1392429292, 2020-03-26:
> > > “编译失败”
> > > 你都不看看失败的原因。。。
> > 
> > 
> > 提示我都写上了，可否高人一步到位，告诉我，这串提示是什么意思吗？原因是？？？
> 
> 谷歌翻译 请

生成工具/下载：生成失败。请使用-j1 V=s或V=sc重新运行make以获得更高的详细级别，以查看发生了什么

谷歌翻译就是这样说的，那么，我需要如何做呢？重新来一遍？重新输入 make download v=s 这个命令？还是直接 make -j1 V=s ？

a1392429292, 2020-03-26:
> 下载失败。。。网络问题

那么，我应该如何做？重新输入  make download v=s   再跑一遍吗？

martin0607, 2020-03-26:
挂代理再操作一次。。。

jzh33, 2020-03-26:
看起来是你的网络导致的下载失败，还有就是你想看日志就把V大写，小写是没用的，挂着全局梯子再来一遍吧。

closed 2020-03-26 by a1392429292