coolsnowwolf/lede #4034: 编译失败，求指导

a1392429292, 2020-03-26:
touch /home/zooo/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libxtables_installed
echo "libxtables" >> /home/zooo/lede/staging_dir/target-x86_64_musl/pkginfo/iptables.default.install
make[3]: Leaving directory `/home/zooo/lede/package/network/utils/iptables'
time: package/network/utils/iptables/compile#10.69#0.86#15.89
make[3]: Entering directory `/home/zooo/lede/package/lean/openwrt-fullconenat'
touch /home/zooo/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2019-10-21-0cf3b48f/.prepared_ae323cdfd5e6288e130e119249c36acf_6664517399ebbbc92a37c5bb081b5c53_check
. /home/zooo/lede/include/shell.sh; xzcat /home/zooo/lede/dl/fullconenat-2019-10-21-0cf3b48f.tar.xz | tar -C /home/zooo/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2019-10-21-0cf3b48f/.. -xf -
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
make[3]: *** [/home/zooo/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2019-10-21-0cf3b48f/.prepared_ae323cdfd5e6288e130e119249c36acf_6664517399ebbbc92a37c5bb081b5c53] Error 2
make[3]: Leaving directory `/home/zooo/lede/package/lean/openwrt-fullconenat'
time: package/lean/openwrt-fullconenat/compile#0.03#0.01#0.24
make[2]: *** [package/lean/openwrt-fullconenat/compile] Error 2
make[2]: Leaving directory `/home/zooo/lede'
make[1]: *** [/home/zooo/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/zooo/lede'
make: *** [world] Error 2
zooo@ubuntu-linux:~/lede$ 

提示 Error ，不清楚为何

Promix953, 2020-03-26:
fullconenat-2019-10-21-0cf3b48f.tar.xz
tar: This does not look like a tar archive

klever1988, 2020-03-26:
make download一下

waynon, 2020-04-08:
> make download一下

请教一下，make download是干什么用的呀

jdnt, 2020-04-26:
今天刚好也遇到这个问题，在/dl/下面把fullconenat-2019-10-21-0cf3b48f.tar.xz删掉，再make download就可以了

closed 2020-04-29 by a1392429292