coolsnowwolf/lede #4044: 关于 DNS 的问题

LisonFan, 2020-03-26:
这个域名 merlinblog.xyz 不在 GFWList
按照大雕这个 issues -> https://github.com/coolsnowwolf/lede/issues/2551 的说法，应该解析出来的 IP 是正确的，但是解析出来的是被污染的

[image]

是哪配置有问题吗？

shikiroot, 2020-03-27:
dns2socks默认在5335，把dnsmasq的转发改一下，另外最好别开dns加速。

LisonFan, 2020-03-27:
> dnsmasq

好了，非常感谢

LisonFan, 2020-03-27:
> dns2socks默认在5335，把dnsmasq的转发改一下，另外最好别开dns加速。

好的， 非常感谢

LisonFan, 2020-03-27:
> dns2socks默认在5335，把dnsmasq的转发改一下，另外最好别开dns加速。

还是不行，我现在转到 5335 了，出来的还是污染 IP

shikiroot, 2020-03-27:
dns加速关掉，wan口和lan口dns都改成127.0.0.1。

LisonFan, 2020-03-27:
> dns加速关掉，wan口和lan口dns都改成127.0.0.1。

dns 加速关了，wan lan 都改了
这样设置的话所有的 DNS 请求都走代理了

zcgmt, 2020-03-27:
这样设置的行不行

zcgmt, 2020-03-27:
请问访问访问国外网站可以，
国内网站时好时坏是怎么回事？

closed 2020-03-27 by LisonFan