coolsnowwolf/lede #4051: 无法识别网卡驱动

atlantis414, 2020-03-27:
路由器型号和固件版本
A：DIY软路由, 有两网卡:板载和BCM5719
正在学习L神开源OP自行编译OpenWRT, 使用lean源码包编译的镜像和OP官方镜像安装都只识别出了板载网口

然后又分别安装 "欢庆3.1精品小包" 和 "koolshare mod 2.31" 都可以同时识别板载卡和BCM5719

请问是不是在编译菜单中还需要打开额外的支持才可以开启PCIE网卡,如BCM5719

closed 2020-03-27 by atlantis414

DavidNGS, 2020-04-22:
查过了，旧的openwrt把BCM5719认成螃蟹8139了，但能正常使用，反而新的内核认不到5719了。
[    7.120767] 8139too: 8139too fast ethernet driver 0.9.28
[    7.125582] bridge firewalling registered

atlantis414, 2020-04-22:
> 查过了，旧的openwrt把BCM5719认成螃蟹8139了，但能正常使用，反而新的内核认不到5719了。
> [ 7.120767] 8139too: 8139too fast ethernet driver 0.9.28
> [ 7.125582] bridge firewalling registered

怪不得, 驱动全选就行. 选了所有BCM就不行.

DavidNGS, 2020-04-22:
你自己编译成功啦，最新的内核试过吗？

atlantis414, 2020-04-22:
> 你自己编译成功啦，最新的内核试过吗？

2周前的update, 把所有有线驱动都选上可以成功, 所以我也不知道到底认了哪一个...

DavidNGS, 2020-04-24:
发现内核一共调用了这些网卡驱动，你可以尝试编译只加上这些看看？

[   10.620815] 8139too: 8139too Fast Ethernet driver 0.9.28
[   10.628625] Atheros(R) L2 Ethernet Driver - version 2.2.3
[   10.631144] Copyright (c) 2007 Atheros Corporation.
[   10.635677] bnx2x: QLogic 5771x/578xx 10/20-Gigabit Ethernet Driver bnx2x 1.712.30-0 (2014/02/10)
[   10.639818] Bridge firewalling registered
[   10.641383] usbcore: registered new interface driver cdc_wdm
[   10.643836] Loading modules backported from Linux version v4.19.98-0-gd183c8e2647a
[   10.647110] Backport generated by backports.git v4.19.98-1-0-g8204eb99
[   10.649358] e100: Intel(R) PRO/100 Network Driver, 3.5.24-k2-NAPI
[   10.650935] e100: Copyright(c) 1999-2006 Intel Corporation
[   10.654900] e1000e: Intel(R) PRO/1000 Network Driver - 3.2.6-k
[   10.656499] e1000e: Copyright(c) 1999 - 2015 Intel Corporation.
[   10.657990] e1000e 0000:00:19.0: Interrupt Throttling Rate (ints/sec) set to simplified (2000-8000 ints) mode
[   10.663024] e1000e 0000:00:19.0: Interrupt Mode set to 1
[   10.775275] e1000e 0000:00:19.0 0000:00:19.0 (uninitialized): registered PHC clock
[   10.896937] e1000e 0000:00:19.0 eth4: (PCI Express:2.5GT/s:Width x1) f8:b1:56:bb:38:aa
[   10.904811] e1000e 0000:00:19.0 eth4: Intel(R) PRO/1000 Network Connection
[   10.910233] e1000e 0000:00:19.0 eth4: MAC: 10, PHY: 11, PBA No: 1011FF-0FF
[   10.915099] gigaset: Driver for Gigaset 307x
[   10.916778] gigaset: ISDN4Linux interface
[   10.918861] hso: drivers/net/usb/hso.c: Option Wireless
[   10.920801] usbcore: registered new interface driver hso
[   10.924775] i40e: Intel(R) Ethernet Connection XL710 Network Driver - version 2.3.2-k
[   10.926752] i40e: Copyright (c) 2013 - 2014 Intel Corporation.
[   10.930487] i40evf: Intel(R) 40-10 Gigabit Virtual Function Network Driver - version 3.2.2-k
[   10.934049] Copyright (c) 2013 - 2015 Intel Corporation.

zhangnew, 2021-12-22:
Kernel modules -> Network Devices 选中 kmod-tg3 亲测可识别