coolsnowwolf/lede #4054: 编译出现libubox20191228相依性错误

aaronchin, 2020-03-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译过程中出现libubox20191228相依性错误

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for base-files:
 *      libubox20191228
 * opkg_install_cmd: Cannot install package base-files.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for mtd:
 *      libubox20191228
 * opkg_install_cmd: Cannot install package mtd.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for urngd:
 *      libubox20191228
 * opkg_install_cmd: Cannot install package urngd.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/aaronchin/tmp/x86/lede'
package/Makefile:108: recipe for target '/home/aaronchin/tmp/x86/lede/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/aaronchin/tmp/x86/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/aaronchin/tmp/x86/lede'
/home/aaronchin/tmp/x86/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

AmadeusGhost, 2020-03-27:
make clean之后再编译即可

aaronchin, 2020-03-27:
自行更换2020-02-27 Makefile 后 编译正常

closed 2020-03-27 by aaronchin