coolsnowwolf/lede #4058: 求問大神 Luci軟件包問題 ADBlock

bdgsfs, 2020-03-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：ADBlock不小心讓我移除了😵，在軟件包重新添加，顯示已安裝，在服務里面沒有，而且有個錯誤日誌，求問大神如何對敵

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：Collected errors:
 * resolve_conffiles: Existing conffile /etc/config/adblock is different from the conffile in the new package. The new conffile will be placed at /etc/config/adblock-opkg.

shikiroot, 2020-03-27:
你移除的是luci还是本体？安装的是luci还是本体？另外这个日志不用管。

bdgsfs, 2020-03-27:
> 你移除的是luci还是本体？安装的是luci还是本体？另外这个日志不用管。

大神你好，我在軟件包中移除了ADBlock，然後服務中ADBlock就沒有了，重新添加，服務當中還沒有，刷新也不行，多次尝試，服務中就是不顯示ADBlock，但是軟件包顯示已安裝，求問

shikiroot, 2020-03-27:
在system/software里搜adblock，看看有几个。应该有两个或者三个，一个adblock，一个luci-app-adblokc，一个汉化包，如果没有luci-app-adblock那么你需要安装这个才能看见。还有这种问题别在issues里提。送你个链接。

- OP编译官方大群:https://t.me/joinchat/JhKgAA6Hx1uiihA7RaTW1w

- OP编译官方频道:https://t.me/opbypd

- op火种群【QQ】https://jq.qq.com/?_wv=1027&k=5rUFZsI

bdgsfs, 2020-03-27:
> 在system/software里搜adblock，看看有几个。应该有两个或者三个，一个adblock，一个luci-app-adblokc，一个汉化包，如果没有luci-app-adblock那么你需要安装这个才能看见。还有这种问题别在issues里提，close掉吧。给你几个链接。
> 
> * OP编译官方大群:https://t.me/joinchat/JhKgAA6Hx1uiihA7RaTW1w
> * OP编译官方频道:https://t.me/opbypd
> * op火种群【QQ】https://jq.qq.com/?_wv=1027&k=5rUFZsI

多謝🙏

closed 2020-03-27 by bdgsfs