coolsnowwolf/lede #4059: newifi3 D2 R20.3.19 新编译没有LUCI界面，uhttpd未安装。

papaloveyou, 2020-03-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：newifi3 D2 R20.3.19 新编译没有LUCI界面，uhttpd未安装。
同配置编译 R20.3.11版本后有luci，但是5GWiFi设置无法设置，停用后luci无线界面无法获取信息，web界面卡住。

### (2) 路由器型号和固件版本

A：newifi3 D2

### (3) 详细日志

A：

EnnawYang, 2020-03-27:
https://github.com/coolsnowwolf/lede/issues/4050

papaloveyou, 2020-03-27:
3

> #4050

今天下午大概2点的时候新开的小鸡配置的。还是有这个问题，应该是源的问题。

papaloveyou, 2020-03-27:
> #4050

我重新编译一遍试试。看看package里面有没有uhttpd

lphgor, 2020-03-27:
https://github.com/coolsnowwolf/lede/commit/2998f9a2287df630db8290690442d01e843fa9d3 把luci误删掉了

EnnawYang, 2020-03-27:
> [2998f9a](https://github.com/coolsnowwolf/lede/commit/2998f9a2287df630db8290690442d01e843fa9d3) 把luci误删掉了

哪里删了？

lphgor, 2020-03-27:
https://github.com/coolsnowwolf/lede/commit/2998f9a2287df630db8290690442d01e843fa9d3

> > [2998f9a](https://github.com/coolsnowwolf/lede/commit/2998f9a2287df630db8290690442d01e843fa9d3) 把luci误删掉了
> 
> 哪里删了？

删luci-proto-relay的时候，多删了一个luci

papaloveyou, 2020-03-27:
> > [2998f9a](https://github.com/coolsnowwolf/lede/commit/2998f9a2287df630db8290690442d01e843fa9d3) 把luci误删掉了
> 
> 哪里删了？

OK,没问题了，编译无错，运行无错。

papaloveyou, 2020-03-27:
不过有个问题，为什么我17号编译的功能更多，才8M，今天编译了就2个app为什么还有10M呢？

lphgor, 2020-03-27:
> > [2998f9a](https://github.com/coolsnowwolf/lede/commit/2998f9a2287df630db8290690442d01e843fa9d3) 把luci误删掉了
> 
> 哪里删了？

https://github.com/coolsnowwolf/lede/commit/2998f9a2287df630db8290690442d01e843fa9d3 删除luci-proto-relay的时候，多删了luci。
默认的app中，只有`luci-app-autoreboot`和`luci-app-filetransfer`是依赖`luci`的，如果正好取消这两个的话，`luci`和`uhttpd`都会取消选中的。最新的代码测试，就是这样。

`# ./scripts/diffconfig.sh`
```
# CONFIG_PACKAGE_libiwinfo-lua is not set
# CONFIG_PACKAGE_luci is not set
# CONFIG_PACKAGE_luci-app-autoreboot is not set
# CONFIG_PACKAGE_luci-app-filetransfer is not set
# CONFIG_PACKAGE_luci-app-firewall is not set
# CONFIG_PACKAGE_luci-lib-fs is not set
# CONFIG_PACKAGE_luci-proto-ppp is not set
# CONFIG_PACKAGE_luci-theme-bootstrap is not set
# CONFIG_PACKAGE_rpcd-mod-rrdns is not set
# CONFIG_PACKAGE_uhttpd is not set
```

EnnawYang, 2020-03-27:
> > > [2998f9a](https://github.com/coolsnowwolf/lede/commit/2998f9a2287df630db8290690442d01e843fa9d3) 把luci误删掉了
> > 
> > 
> > 哪里删了？
> 
> [2998f9a](https://github.com/coolsnowwolf/lede/commit/2998f9a2287df630db8290690442d01e843fa9d3) 删除luci-proto-relay的时候，多删了luci。
> 默认的app中，只有`luci-app-autoreboot`和`luci-app-filetransfer`是依赖`luci`的，如果正好取消这两个的话，`luci`和`uhttpd`都会取消选中的。最新的代码测试，就是这样。
> 
> `# ./scripts/diffconfig.sh`
> 
> ```
> # CONFIG_PACKAGE_libiwinfo-lua is not set
> # CONFIG_PACKAGE_luci is not set
> # CONFIG_PACKAGE_luci-app-autoreboot is not set
> # CONFIG_PACKAGE_luci-app-filetransfer is not set
> # CONFIG_PACKAGE_luci-app-firewall is not set
> # CONFIG_PACKAGE_luci-lib-fs is not set
> # CONFIG_PACKAGE_luci-proto-ppp is not set
> # CONFIG_PACKAGE_luci-theme-bootstrap is not set
> # CONFIG_PACKAGE_rpcd-mod-rrdns is not set
> # CONFIG_PACKAGE_uhttpd is not set
> ```

看到了

closed 2020-03-28 by coolsnowwolf