coolsnowwolf/lede #4061: GFW列表模式下，闲置一段时间或者跑一下不经过代理的流量后，某些网页无法访问

Pastechn, 2020-03-27:
#### 具体问题：
使用GFW列表模式翻墙，刚设置完一切正常，所有网页均可访问。当闲置一段时间，或者一段时间内的流量都不走代理（以测速或者看视频为主）后，某些网页无法访问，以谷歌系列和Twitter为主。Youtube的现象很奇怪，有时是直接无法访问，有时是能进去，但是视频无法加载，通过F12查阅发现有Network中有大量的Error，且基本都是video playback之类的。此时在Windows上刷新一次DNS缓存可解决问题，但是一段时间后问题仍会复现。绕过大陆IP模式则没有这个问题。
#### 设备型号：
树莓派3B+
#### 操作模式：
旁路由（主路由分配IP，网关与DNS指向旁路由）
#### 出现时间：
从3月中旬开始刷入固件时已存在此问题，此前未知
#### 是否有类似的Issue：
未查询到

ppdragon16, 2020-03-28:
我有类似体会。一觉醒来打开电脑，访问baidu.com秒开，访问google.com浏览器转圈。
有时候多等一下，google页面就出来了，有时候则超时。
如果停用再启用SSR Plus +，就没问题了。

我试过绕过大陆IP模式和GFW列表模式，都是这样。

txdywy, 2020-03-28:
dns污染的样子

ppdragon16, 2020-03-28:
不太像啊，域名解析应该走的梯子。

难道是ISP给wan口重新分配了个IP地址，然后造成了。。。。我瞎猜的。 :-)

我的情况是单臂路由，单网卡，wan口pppoe拨号走vlan。

txdywy, 2020-03-28:
感觉是污染了有缓存，有时也会遇到这种现象。具体什么原因（肯定是dns部分一度不管用了）造成的也还没研究清楚

ppdragon16, 2020-03-28:
弄个定时任务清除dns cache？
感觉系统里dns cache太多，Turbo ACC里的DNS加速有cache，pdnsd有cache，dnsmasq有cache。
要这么多cache干嘛。

hcwang, 2020-03-28:
同样情况。

ppdragon16, 2020-03-28:
我在想有没有可能是因为机场的domain被污了？
我设一个域名劫持试试效果。

dolphinpaopao, 2020-03-28:
目前自用正常。公用流量过大可能会出现断流。

txdywy, 2020-03-28:
感觉有可能

gfonline, 2020-03-28:
我也有这个情况

Wilson-coder, 2020-03-28:
> 我在想有没有可能是因为机场的domain被污了？
> 我设一个域名劫持试试效果。

自建的VPS也有遇到这个现象，感觉不太像你说的这个原因。

txdywy, 2020-03-28:
感觉还是dns卡住了，半天才恢复

z826540272, 2020-03-28:
这就是目前SSR+等路由器翻墙的缺陷,给大神反馈大神就一句你DNS污染了让你重启设备.
实际上是按照设计思想根本不可能存在DNS污染的情况.
电脑上基本遇不到,但是安卓手机上遇到是家常便饭经常play商店无法下载或者YouTube无法播放

z826540272, 2020-03-28:
我可以断定绝非线路问题,手机上直接开翻墙稳如狗一切正常

z826540272, 2020-03-28:
> 我在想有没有可能是因为机场的domain被污了？
> 我设一个域名劫持试试效果。

不是.一般两种情况,DNS返回被墙IP,例如Facebook,Twitter.
或者DNS返回正常IP,但是当前设备无法访问.其他设备可正常访问.

ppdragon16, 2020-03-29:
今早出现手机无法获取IP，禁用再启用wifi后也等了好一会才拿到IP。
感觉dnsmasq可能是重启了。

txdywy, 2020-03-29:
访问不了的时候发现dns无缓存查询卡很长时间，重启dns服务好像也不一定管用

Pastechn, 2020-03-29:
我想调查一下，出现这个问题的路由器是ARM架构的，还是X86和X64都有呢？

hcwang, 2020-03-29:
mips也有

hcwang, 2020-03-31:
禁用了Turbo ACC中的DNS加速，现在过了24小时了，一切正常。IPV6也开着。

fyngd, 2020-04-04:
我的也同样存在这个问题，编译的固件只保留了基本的上网和翻墙功能。只要静待一段时间，感觉墙外就不通了，进入SSR+的服务器节点标签，自动ping后网络马上就通了。

closed 2020-07-04 by coolsnowwolf