coolsnowwolf/lede #4062: ssr有个想法，可以设置某时间段自动切换到某个节点

DongfeiSay, 2020-03-27:
像某些晚高峰期，墙发威时间段，大部分的节点都会有不同程度的速度变慢，稳定性变差等，可以设置这些时间段自动切换上比较稳定的节点(中转，线路质量甚至LPLC等稳定型更好但是费用更高这些)，高峰时间段不强求非常高的网速,流量便宜的节点，这些时间段以稳定性优先是个较好的选择。

1.你可以高峰时间段自己切换啊
：人总是能懒一点就懒一点的

2.你可以自己在计划任务里写上xx时间段切换节点的代码啊
：大群体来说没多少人会，小白还是占比很高的(比如我)，能更简单的实现该功能何乐而不为

stayrabbit, 2020-03-27:
如果可以的話 請加入Netflix自動切換相應的節點

canflyx, 2020-03-28:
我也准备这么干，自己做个定时任务得了

tty228, 2020-03-28:
`0 20 * * * uci set shadowsocksr.@global[0].global_server='cfg094a8f' && uci commit shadowsocksr && /etc/init.d/shadowsocksr  restart`

0 20 * * * 每天 20 点 00 分执行
`uci set shadowsocksr.@global[0].global_server='cfg0c4a8f'` #更改全局服务器为 cfg0c4a8f
`uci commit shadowsocksr` #使设置生效
`/etc/init.d/shadowsocksr  restart` 重启SSR

计划任务计算器：
https://tool.lu/crontab/

tty228, 2020-03-28:
服务器名可以在网页上更改全局服务器，点击保存但不要更改
[image]

canflyx, 2020-03-28:
原来如此，我试下。谢谢

bzzjh, 2020-04-15:
这个想法挺好的 白天可以用HK节点，晚上HK节点比较拥挤，就用欧洲的，欧洲虽然延迟大比较稳定

closed 2020-07-04 by coolsnowwolf