coolsnowwolf/lede #4064: 编译失败, 总是编译到ssl这里就报错, 'TLSv1_client_method' is deprecated

jaxwang, 2020-03-28:
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

### (1) 具体问题
- 使用的分支:  2020年3月28日 最新的master分支
- 系统环境:  Ubuntu 18.04  官方镜像.
- 系统环境是否安装了openssl 和 openssl-dev:  是
- 选择编译的平台系统: x84_64
- 目标路由器型号: 联想ThinkPad L421笔记本 (还没到在电脑安装那一步)
- 问题描述:
最终开始使用`make -j1 V=s`编译的时, 总是编译到ssl这里就报错, 以下是日志信息, 看起来好像是代码中`TLSv1_client_method`这个方法已经被废弃了. 

```
x86_64-openwrt-linux-musl-gcc -c -g -Os -pipe -fno-omit-frame-pointer -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/jax/lede/build_dir/target-x86_64_musl/imap-2007f:imap-2007f -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fPIC -DPIC -D_REENTRANT -DDISABLE_POP_PROXY -fno-strict-aliasing -Wno-pointer-sign -Wno-implicit-function-declaration -Wno-incompatible-pointer-types -DCHUNKSIZE=65536 -I/home/jax/lede/staging_dir/target-x86_64_musl/usr/include -I/home/jax/lede/staging_dir/target-x86_64_musl/usr/include/openssl -DSSL_CERT_DIRECTORY=\"/etc/ssl/certs\" -DSSL_KEY_DIRECTORY=\"/etc/ssl/certs\" -DCREATEPROTO=unixproto -DEMPTYPROTO=unixproto -DMD5ENABLE=\"/etc/cram-md5.pwd\" -DMAILSPOOL=\"/var/spool/mail\" -DANONYMOUSHOME=\"/var/spool/mail/anonymous\" -DACTIVEFILE=\"/var/lib/news/active\" -DNEWSSPOOL=\"/var/spool/news\" -DRSHPATH=\"/usr/bin/rsh\" -DLOCKPGM=\"\" -DLOCKPGM1=\"/usr/libexec/mlock\" -DLOCKPGM2=\"/usr/sbin/mlock\" -DLOCKPGM3=\"/etc/mlock\" -c osdep.c
osdep.c: In function 'ssl_start_work':
osdep.c:425:12: warning: 'TLSv1_client_method' is deprecated [-Wdeprecated-declarations]
            TLSv1_client_method () :
            ^~~~~~~~~~~~~~~~~~~
In file included from /home/jax/lede/staging_dir/target-x86_64_musl/usr/include/openssl/e_os2.h:13:0,
                 from /home/jax/lede/staging_dir/target-x86_64_musl/usr/include/openssl/bio.h:13,
                 from /home/jax/lede/staging_dir/target-x86_64_musl/usr/include/openssl/x509v3.h:13,
                 from osdep.c:232:
/home/jax/lede/staging_dir/target-x86_64_musl/usr/include/openssl/ssl.h:1885:45: note: declared here
 DEPRECATEDIN_1_1_0(__owur const SSL_METHOD *TLSv1_client_method(void))
                                             ^
osdep.c:475:49: error: dereferencing pointer to incomplete type 'X509 {aka struct x509_st}'
     if (scq) return (*scq) (err,host,cert ? cert->name : "???") ? NIL : "";
                                                 ^~
osdep.c: In function 'ssl_server_init':
osdep.c:908:12: warning: 'TLSv1_server_method' is deprecated [-Wdeprecated-declarations]
            TLSv1_server_method () :
            ^~~~~~~~~~~~~~~~~~~
In file included from /home/jax/lede/staging_dir/target-x86_64_musl/usr/include/openssl/e_os2.h:13:0,
                 from /home/jax/lede/staging_dir/target-x86_64_musl/usr/include/openssl/bio.h:13,
                 from /home/jax/lede/staging_dir/target-x86_64_musl/usr/include/openssl/x509v3.h:13,
                 from osdep.c:232:
/home/jax/lede/staging_dir/target-x86_64_musl/usr/include/openssl/ssl.h:1884:45: note: declared here
 DEPRECATEDIN_1_1_0(__owur const SSL_METHOD *TLSv1_server_method(void))
                                             ^
osdep.c: In function 'ssl_genkey':
osdep.c:979:5: warning: 'RSA_generate_key' is deprecated [-Wdeprecated-declarations]
     if (!(key = RSA_generate_key (export ? keylength : 1024,RSA_F4,NIL,NIL))) {
     ^~
In file included from /home/jax/lede/staging_dir/target-x86_64_musl/usr/include/openssl/e_os2.h:13:0,
                 from /home/jax/lede/staging_dir/target-x86_64_musl/usr/include/openssl/bio.h:13,
                 from /home/jax/lede/staging_dir/target-x86_64_musl/usr/include/openssl/x509v3.h:13,
                 from osdep.c:232:
/home/jax/lede/staging_dir/target-x86_64_musl/usr/include/openssl/rsa.h:235:1: note: declared here
 DEPRECATEDIN_0_9_8(RSA *RSA_generate_key(int bits, unsigned long e, void
 ^
Makefile:931: recipe for target 'osdep.o' failed
make[7]: *** [osdep.o] Error 1
make[7]: Leaving directory '/home/jax/lede/build_dir/target-x86_64_musl/imap-2007f/c-client'
Makefile:684: recipe for target 'slx' failed
make[6]: *** [slx] Error 2
make[6]: Leaving directory '/home/jax/lede/build_dir/target-x86_64_musl/imap-2007f/c-client'
Makefile:680: recipe for target 'OSTYPE' failed
make[5]: *** [OSTYPE] Error 2
make[5]: Leaving directory '/home/jax/lede/build_dir/target-x86_64_musl/imap-2007f'
Makefile:311: recipe for target 'slx' failed
make[4]: *** [slx] Error 2
make[4]: Leaving directory '/home/jax/lede/build_dir/target-x86_64_musl/imap-2007f'
Makefile:73: recipe for target '/home/jax/lede/build_dir/target-x86_64_musl/imap-2007f/.built' failed
make[3]: *** [/home/jax/lede/build_dir/target-x86_64_musl/imap-2007f/.built] Error 2
make[3]: Leaving directory '/home/jax/lede/feeds/packages/libs/uw-imap'
time: package/feeds/packages/uw-imap/compile#0.22#0.11#0.35
package/Makefile:111: recipe for target 'package/feeds/packages/uw-imap/compile' failed
make[2]: *** [package/feeds/packages/uw-imap/compile] Error 2
make[2]: Leaving directory '/home/jax/lede'
package/Makefile:107: recipe for target '/home/jax/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/jax/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jax/lede'
/home/jax/lede/include/toplevel.mk:225: recipe for target 'world' failed

```

dongjuanyong, 2020-03-28:
你这是uw-imap包的问题，没必须要的话取消它，必须要的话看看openwrt官方feeds的commit自己改一下：https://github.com/openwrt/packages/commits/master/libs/uw-imap

closed 2020-03-29 by coolsnowwolf

jaxwang, 2020-03-29:
> 你这是uw-imap包的问题，没必须要的话取消它，必须要的话看看openwrt官方feeds的commit自己改一下：https://github.com/openwrt/packages/commits/master/libs/uw-imap

感谢, 已经解决, 用了官方最新的uw-imap 包, 编译通过