coolsnowwolf/lede #4065: 默认配置编译k2p固件没有生成sysupgrade.bin文件

lariboo, 2020-03-28:
使用commits为cdf1ae9的源码，默认配置编译k2p固件，编译完成后没有生成sysupgrade.bin文件。何故？

ecco2005, 2020-03-28:
> 使用commits为cdf1ae9的源码，默认配置编译k2p固件，编译完成后没有生成sysupgrade.bin文件。何故？

我和你的情况一样，我编译的是K2固件，编译完只有一个initramfs的bin文件。不过，前几天编译的时候，是有2个bin的，其中一个就是syspgrade.bin。

这两天试了好多次，都只有1个bin，不知道哪里出问题了。

hxlls, 2020-03-28:
容量超了

kl9800, 2020-03-28:
现在好像编辑出的固件比以前大了，我甚至缩减了以前常编辑的固件，体积还是超了。。。。

892947707, 2020-03-29:
我刚开始也有问题，make clean就好了，不行就删掉整个文件夹重新开始。我现在，嗯，勉强算是熟悉了，编出来的只有8M

kl9800, 2020-03-29:
每次我都make clean，可是不行，删掉整个文件夹太墨迹了，又得6小时起步。。。。

hxlls, 2020-03-29:
> 每次我都make clean，可是不行，删掉整个文件夹太墨迹了，又得6小时起步。。。。

这里不是论坛，提出问题后按照给的步骤解决，才给出反馈，而不是抱怨什么时间太长

kl9800, 2020-03-29:
> > 每次我都make clean，可是不行，删掉整个文件夹太墨迹了，又得6小时起步。。。。
> 
> 这里不是论坛，提出问题后按照给的步骤解决，才给出反馈，而不是抱怨什么时间太长

啥毛病都没有，啥问题也没有，好着呢！！

shu0802, 2020-03-31:
> 我刚开始也有问题，make clean就好了，不行就删掉整个文件夹重新开始。我现在，嗯，勉强算是熟悉了，编出来的只有8M

我编译了多次还是不能缩小固件大小，你是怎么做到的，求指导！

892947707, 2020-03-31:
> > 我刚开始也有问题，make clean就好了，不行就删掉整个文件夹重新开始。我现在，嗯，勉强算是熟悉了，编出来的只有8M
> 
> 我编译了多次还是不能缩小固件大小，你是怎么做到的，求指导！

执行 make clean 和 rm -f .config  
我没看懂你说的“使用commits为cdf1ae9的源码”事什么意思，我是直接拉取最新源码编译的。时不时更新一下源码和软件包。默认勾选的插件很多我也用不到，所以我实际编译的插件最多10来个。有时候也会编出来14M或者超过的，大多数还是8M到10M左右的

kl9800, 2020-03-31:
我是重新

> > 我刚开始也有问题，make clean就好了，不行就删掉整个文件夹重新开始。我现在，嗯，勉强算是熟悉了，编出来的只有8M
> 
> 我编译了多次还是不能缩小固件大小，你是怎么做到的，求指导！

sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib
重新执行以下这个，也好使，我就是这么解决的，要不然插件极少的情况下也会编译出30M固件

shu0802, 2020-04-01:
> 我是重新
>
>
> sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs [...]

用这个命令重头编译？

shu0802, 2020-04-01:
> 我是重新
>
>
> sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs [...]

重头编译还是二次编译时用这个命令

shu0802, 2020-04-01:
> 重头编译还是二次编译时用这个命令
编译时选择插件luci-app-adbyby-plus、luci-app-firewall、luci-app-mwan3、luci-app-oaf、luci-app-ssr-plus、Include Shadowsocks V2ray Plugin、Include V2ray、Include Trojan、Include Redsocks2 、Include ShadowsocksR Server、luci-app-syncdial、luci-app-vlmcsd后，编译出来也有15.7m，是不是还要减少插件个数
[image]

kl9800, 2020-04-01:
二次，就是提前多一步这个

kl9800, 2020-04-01:
先这个然后再pull  make什么的

closed 2020-07-04 by coolsnowwolf