coolsnowwolf/lede #4072: 求助！树莓派4编译后概况界面不显示CPU架构和温度 怎么解决？

IvanSolis1989, 2020-03-28:
树莓派4编译后 概况界面不显示CPU架构和温度 怎么解决？是不是需要选择某个插件才可以？

[image]

主机名 | RaspberryPi4
-- | --
主机型号 | Raspberry Pi 4 Model B Rev 1.2 (CpuMark : 32739.313218 Scores)
架构 | ? x 4
固件版本 | OpenWrt R20.3.19 / LuCI Master (git-20.085.19816-df13074)
内核版本 | 4.19.108
本地时间 | Sat Mar 28 18:33:48 2020
运行时间 | 12h 3m 28s
平均负载 | 0.42, 0.44, 0.39
CPU 使用率（%） | 3 %

EnnawYang, 2020-03-28:
忽略即可

IvanSolis1989, 2020-03-28:
> 忽略即可

但是我想知道下CPU的温度啊 架构不显示都没事

Pastechn, 2020-03-28:
sed -i 's|pcdata(boardinfo.system or "?")|luci.sys.exec("uname -m") or "?"|g' /usr/lib/lua/luci/view/admin_status/index.htm
参考资料：https://tieba.baidu.com/p/6357010977?red_tag=b1848246377

feijj, 2020-04-02:
https://github.com/klever1988/lede/blob/master/enable_autocore.diff
解决方案在此

PR #4720 (2020-06-06): treewide: adjust sysupgrade script
This commit fix sysupgrade failed for bcm27xx, fix #4072, fix #4108 

Run-tested-by: @1715173329 Dalao on Raspberry Pi 3B
Run-tested-by: @EnnawYang Dalao on Raspberry Pi 4B
closed 2020-06-06 by coolsnowwolf