coolsnowwolf/lede #4074: X86 -ppoe频繁重启

ITdesk01, 2020-03-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

最新的固件，经常断网，一个小时7-8次，日志显示是ppoe重启了,有尝试过恢复出厂设置，重新编译不加入多拨插件，问题一样(附件晚点上传老断线)

### (2) 路由器型号和固件版本

A：X86 版本最新OpenWrt R20.3.19 / LuCI Master (git-20.088.29068-cebe120)

### (3) 详细日志

`Sat Mar 28 17:12:54 2020 kern.info kernel: [    4.212382] Button Hotplug driver version 0.4.1
Sat Mar 28 17:12:54 2020 kern.info kernel: [    4.218728] exFAT: Version 1.2.9
Sat Mar 28 17:12:54 2020 kern.info kernel: [    4.223642] usbcore: registered new interface driver uas
Sat Mar 28 17:12:54 2020 user.info kernel: [    4.230107] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
Sat Mar 28 17:12:54 2020 user.info kernel: [    4.247356] init: - preinit -
Sat Mar 28 17:12:54 2020 kern.notice kernel: [    4.412631] random: jshn: uninitialized urandom read (4 bytes read)
Sat Mar 28 17:12:54 2020 kern.notice kernel: [    4.432390] random: jshn: uninitialized urandom read (4 bytes read)
Sat Mar 28 17:12:54 2020 kern.notice kernel: [    4.446198] random: jshn: uninitialized urandom read (4 bytes read)
Sat Mar 28 17:12:54 2020 kern.warn kernel: [    7.578086] urandom_read: 4 callbacks suppressed
Sat Mar 28 17:12:54 2020 kern.notice kernel: [    7.578092] random: jshn: uninitialized urandom read (4 bytes read)
Sat Mar 28 17:12:54 2020 user.info kernel: [    7.643678] mount_root: loading kmods from internal overlay
Sat Mar 28 17:12:54 2020 user.info kernel: [    7.676348] kmodloader: loading kernel modules from //etc/modules-boot.d/*
Sat Mar 28 17:12:54 2020 user.info kernel: [    7.685275] kmodloader: done loading kernel modules from //etc/modules-boot.d/*
Sat Mar 28 17:12:54 2020 user.info kernel: [    7.761487] block: attempting to load /tmp/overlay/upper/etc/config/fstab
Sat Mar 28 17:12:54 2020 user.err kernel: [    7.769353] block: unable to load configuration (fstab: Entry not found)
Sat Mar 28 17:12:54 2020 user.info kernel: [    7.777105] block: attempting to load /tmp/overlay/etc/config/fstab
Sat Mar 28 17:12:54 2020 user.err kernel: [    7.784377] block: unable to load configuration (fstab: Entry not found)
Sat Mar 28 17:12:54 2020 user.info kernel: [    7.792131] block: attempting to load /etc/config/fstab
Sat Mar 28 17:12:54 2020 user.err kernel: [    7.798236] block: unable to load configuration (fstab: Entry not found)
Sat Mar 28 17:12:54 2020 user.err kernel: [    7.805979] block: no usable configuration
Sat Mar 28 17:12:54 2020 kern.notice kernel: [    7.834386] random: procd: uninitialized urandom read (4 bytes read)
Sat Mar 28 17:12:54 2020 user.info kernel: [    7.921599] mount_root: rootdisk overlay filesystem has not been formatted yet
Sat Mar 28 17:12:54 2020 kern.notice kernel: [    8.313037] random: mkfs.ext4: uninitialized urandom read (16 bytes read)
Sat Mar 28 17:12:54 2020 kern.info kernel: [    8.424918] EXT4-fs (loop0): mounted filesystem with ordered data mode. Opts: (null)
Sat Mar 28 17:12:54 2020 user.info kernel: [    8.441286] block: attempting to load /tmp/overlay/upper/etc/config/fstab
Sat Mar 28 17:12:54 2020 user.err kernel: [    8.449147] block: unable to load configuration (fstab: Entry not found)
Sat Mar 28 17:12:54 2020 user.info kernel: [    8.456905] block: attempting to load /tmp/overlay/etc/config/fstab
Sat Mar 28 17:12:54 2020 user.err kernel: [    8.464187] block: unable to load configuration (fstab: Entry not found)
Sat Mar 28 17:12:54 2020 user.info kernel: [    8.471941] block: attempting to load /etc/config/fstab
Sat Mar 28 17:12:54 2020 user.err kernel: [    8.478035] block: unable to load configuration (fstab: Entry not found)
Sat Mar 28 17:12:54 2020 user.err kernel: [    8.485778] block: no usable configuration
Sat Mar 28 17:12:54 2020 user.info kernel: [    8.490959] mount_root: overlay filesystem has not been fully initialized yet
Sat Mar 28 17:12:54 2020 user.info kernel: [    8.499601] mount_root: switching to ext4 overlay
Sat Mar 28 17:12:54 2020 kern.info kernel: [    8.586679] EXT4-fs (sda1): mounted filesystem without journal. Opts: (null)
Sat Mar 28 17:12:54 2020 user.warn kernel: [    8.633968] urandom-seed: Seed file not found (/etc/urandom.seed)
Sat Mar 28 17:12:54 2020 user.info kernel: [    8.657525] procd: - early -
Sat Mar 28 17:12:54 2020 user.info kernel: [    9.316131] procd: - ubus -
Sat Mar 28 17:12:54 2020 user.info kernel: [    9.371704] procd: - init -
Sat Mar 28 17:12:54 2020 user.info kernel: [    9.540122] urngd: v1.0.2 started.
Sat Mar 28 17:12:54 2020 kern.notice kernel: [    9.554224] random: crng init done
Sat Mar 28 17:12:54 2020 user.info kernel: [    9.556656] kmodloader: loading kernel modules from /etc/modules.d/*
Sat Mar 28 17:12:54 2020 kern.info kernel: [    9.571714] NET: Registered protocol family 38
Sat Mar 28 17:12:54 2020 kern.info kernel: [    9.750332] raid6: sse2x1   gen()  2442 MB/s
Sat Mar 28 17:12:54 2020 kern.info kernel: [    9.920328] raid6: sse2x1   xor()  2482 MB/s
Sat Mar 28 17:12:54 2020 kern.info kernel: [   10.090350] raid6: sse2x2   gen()  2245 MB/s
Sat Mar 28 17:12:54 2020 kern.info kernel: [   10.260328] raid6: sse2x2   xor()  2883 MB/s
Sat Mar 28 17:12:54 2020 kern.info kernel: [   10.430343] raid6: sse2x4   gen()  3483 MB/s
Sat Mar 28 17:12:54 2020 kern.info kernel: [   10.600329] raid6: sse2x4   xor()  2126 MB/s
Sat Mar 28 17:12:54 2020 kern.info kernel: [   10.605326] raid6: using algorithm sse2x4 gen() 3483 MB/s
Sat Mar 28 17:12:54 2020 kern.info kernel: [   10.611591] raid6: .... xor() 2126 MB/s, rmw enabled
Sat Mar 28 17:12:54 2020 kern.info kernel: [   10.617365] raid6: using ssse3x2 recovery algorithm
Sat Mar 28 17:12:54 2020 kern.info kernel: [   10.625001] xor: measuring software checksum speed
Sat Mar 28 17:12:54 2020 kern.info kernel: [   10.730329]    prefetch64-sse:  7504.000 MB/sec
[... 592 lines cut ...]
Sat Mar 28 17:24:44 2020 user.info watchcat: stopping all tasks
Sat Mar 28 17:24:44 2020 user.info watchcat: all tasks stopped
Sat Mar 28 17:24:44 2020 user.info watchcat: starting all tasks
Sat Mar 28 17:24:44 2020 user.info watchcat: started task (mode=ping;period=21600;pinghosts=www.baidu.com;pingperiod=1080;forcedelay=30)
Sat Mar 28 17:24:44 2020 user.info watchcat: all tasks started
Sat Mar 28 17:24:44 2020 user.info watchcat: stopping all tasks
Sat Mar 28 17:24:44 2020 user.info watchcat: all tasks stopped
Sat Mar 28 17:24:44 2020 user.info watchcat: starting all tasks
Sat Mar 28 17:24:44 2020 user.info watchcat: started task (mode=ping;period=21600;pinghosts=www.baidu.com;pingperiod=1080;forcedelay=30)
Sat Mar 28 17:24:44 2020 user.info watchcat: all tasks started
Sat Mar 28 17:24:44 2020 user.info watchcat: stopping all tasks
Sat Mar 28 17:24:44 2020 user.info watchcat: all tasks stopped
Sat Mar 28 17:24:44 2020 user.info watchcat: starting all tasks
Sat Mar 28 17:24:44 2020 user.info watchcat: started task (mode=ping;period=21600;pinghosts=www.baidu.com;pingperiod=1080;forcedelay=30)
Sat Mar 28 17:24:44 2020 user.info watchcat: all tasks started
Sat Mar 28 17:24:44 2020 user.info watchcat: stopping all tasks
Sat Mar 28 17:24:44 2020 user.info watchcat: all tasks stopped
Sat Mar 28 17:24:44 2020 user.info watchcat: starting all tasks
Sat Mar 28 17:24:44 2020 user.info watchcat: started task (mode=ping;period=21600;pinghosts=www.baidu.com;pingperiod=1080;forcedelay=30)
Sat Mar 28 17:24:44 2020 user.info watchcat: all tasks started
Sat Mar 28 17:24:44 2020 user.info watchcat: stopping all tasks
Sat Mar 28 17:24:44 2020 user.info watchcat: all tasks stopped
Sat Mar 28 17:24:44 2020 user.info watchcat: starting all tasks
Sat Mar 28 17:24:44 2020 user.info watchcat: started task (mode=ping;period=21600;pinghosts=www.baidu.com;pingperiod=1080;forcedelay=30)
Sat Mar 28 17:24:44 2020 user.info watchcat: all tasks started`

coolsnowwolf, 2020-03-28:
用了第三方的的  app filter 请找原作者解决

ITdesk01, 2020-03-28:
> 用了第三方的的应用过滤器请找原作者解决

大佬能否告知一下是哪个软件引起的，我记得没加什么第三方插件

coolsnowwolf, 2020-03-28:
app filter

coolsnowwolf, 2020-03-28:
和 watchcat

ITdesk01, 2020-03-29:
这是我的固件烦请大佬看下
https://c-t.work/s/4d343c00f8f444 在首页输入取件码：su4fny（24小时内有效）

app filter我逛了一圈没有发现，watchcat我尝试卸载了，但依旧ppoe频繁重启

ITdesk01, 2020-03-29:
更新为最新的源码然后去掉 watchcat，暂时正常了，看看后面怎么说

closed 2020-03-29 by ITdesk01