coolsnowwolf/lede #4076: 编译出错，找不到mac.o文件，但我从google上都找不到mac.o这个文件，只有mac.c和mac.h

Keeper-Samsara, 2020-03-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
编译时好像找不到这个文件，我去目录查看，压根就没有这个ath 目录。

随便问一下，怎么重置  make kernel_menuconfig 这个里面的选项。

make clean 、make distclean和rm -rf ./tmp && rm -rf .config都不起作用

### (2) 路由器型号和固件版本

x64x86
### (3) 详细日志
make[11]: *** [/home/sam/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.98-1/drivers/net/wireless/ath/ath10k/mac.o] Error 1
make[10]: *** [/home/sam/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.98-1/drivers/net/wireless/ath/ath10k] Error 2
make[9]: *** [/home/sam/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.98-1/drivers/net/wireless/ath] Error 2
make[8]: *** [/home/sam/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.98-1/drivers/net/wireless] Error 2
make[7]: *** [_module_/home/sam/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.98-1] Error 2
make[6]: *** [modules] Error 2
make[5]: *** [modules] Error 2
make[4]: *** [modules] Error 2
make[4]: Leaving directory `/home/sam/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.98-1'
make[3]: *** [/home/sam/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.98-1/.built] Error 2
make[3]: Leaving directory `/home/sam/lede/package/kernel/mac80211'
time: package/kernel/mac80211/compile#3.80#1.27#34.19
make[2]: *** [package/kernel/mac80211/compile] Error 2
make[2]: Leaving directory `/home/sam/lede'
make[1]: *** [/home/sam/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/sam/lede'
make: *** [world] Error 2

coolsnowwolf, 2020-03-28:
make clean

Keeper-Samsara, 2020-03-28:
> make clean

谢谢大佬回复   我运行一下重新编译试试看

closed 2020-04-01 by coolsnowwolf