coolsnowwolf/lede #4077: 安装主题后管理页面崩溃

Omoinemie, 2020-03-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [y ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：安装了L大每天编译的x86版本，然后在软件包里面在线下载了一个主题，提示安装出错，管理页面就崩溃了，404错误，进不去

### (2) 路由器型号和固件版本

A：OpenWrt-CI #121

### (3) 详细日志

A：
’
Sat Mar 28 20:42:27 2020 daemon.err uhttpd[1940]: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'sysauth'.
Sat Mar 28 20:42:27 2020 daemon.err uhttpd[1940]: A runtime error occured: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'header'.
Sat Mar 28 20:42:27 2020 daemon.err uhttpd[1940]: A runtime error occured: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'themes/openwrt.org/header'.
Sat Mar 28 20:42:27 2020 daemon.err uhttpd[1940]: A runtime error occured: [string "/usr/lib/lua/luci/view/themes/openwrt.org/h..."]:15: attempt to call field 'menu_json' (a nil value)
Sat Mar 28 20:42:27 2020 daemon.err uhttpd[1940]: stack traceback:
Sat Mar 28 20:42:27 2020 daemon.err uhttpd[1940]: 	[C]: in function 'n'
Sat Mar 28 20:42:27 2020 daemon.err uhttpd[1940]: 	/usr/lib/lua/luci/template.lua:55: in function </usr/lib/lua/luci/template.lua:47>
Sat Mar 28 20:42:27 2020 daemon.err uhttpd[1940]: 	(tail call): ?
Sat Mar 28 20:42:27 2020 daemon.err uhttpd[1940]: 	/usr/lib/lua/luci/dispatcher.lua:319: in function 'dispatch'
Sat Mar 28 20:42:27 2020 daemon.err uhttpd[1940]: 	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>
‘

jzh33, 2020-03-28:
我是直接把软件源#掉了，从不在线安装软件，因为基本上都是你现在的结果，自己去找编译好的离线安装，或者直接编译进去。

coolsnowwolf, 2020-03-28:
不兼容官方的luci，别乱来

Omoinemie, 2020-03-28:
我应该怎么删除并默认主题呢？不想重装了

Myoko, 2020-03-29:
不兼容别装

hcym, 2020-03-29:
ssh进去改吧

agutools, 2020-03-29:
nano /etc/config/luci
config core 'main'
	option mediaurlbase '/luci-static/bootstrap'

Omoinemie, 2020-03-29:
我用ssh重新覆盖安装了主题，现在能进去了

closed 2020-03-29 by Omoinemie

leechaohua, 2021-02-11:
> nano /etc/config/luci
> config core 'main'
> option mediaurlbase '/luci-static/bootstrap'

厉害，跟你这样做，救活了