coolsnowwolf/lede #4079: MT7621 固件体积问题  求助求助！！！

kl9800, 2020-03-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：现在无论插件多少，甚至只编辑1-2个插件进去，squashfs-factory.bin 永远会是30M，刷不进去极路由4增强版，我该怎么办？

### (2) 路由器型号和固件版本

A：ramips-mt7621-hiwifi_hc5962  

### (3) 详细日志

A：

coolsnowwolf, 2020-03-28:
hc5962 的 factory.bin  体积是固定的

kl9800, 2020-03-28:
> hc5962 的 factory.bin 体积是固定的
[image]

可是我2月4号编译的还好好的。。。是近期修改成固定了吗？

zanjie1999, 2020-03-29:
你应该刷sysupgrade.bin

kl9800, 2020-03-29:
但是我记得刷完breed以后，刷sysupgrade，提示好像不让刷，具体提示什么我忘了，一会再去试试吧

closed 2020-03-30 by kl9800

SouHong, 2020-03-30:
[openwrt-ramips-mt7621-hiwifi_hc5962-squashfs-factory.zip](https://github.com/coolsnowwolf/lede/files/4401734/openwrt-ramips-mt7621-hiwifi_hc5962-squashfs-factory.zip)