coolsnowwolf/lede #4080: 添加了Trojan节点以后ShadowSocksR Plus+一直未运行

cuijian83, 2020-03-28:
ShadowsocksR Plus+ 未运行

1.关于你要提交的问题
ShadowsocksR Plus+ 未运行

2. 详细叙述
更新完服务器节点以后，有了Trojan服务器节点，但是选择节点保存以后显示ShadowsocksR Plus+ 未运行；无法编辑节点，编辑后类型会自动变成SSR，变成SSR的节点保存后ShadowsocksR Plus+可以运行，但是无法连接Google

3.路由器型号和固件版本
路由器：NETGEAR WNDR4300 V1
固件：OpenWrt R20.3.19 / LuCI Master (git-20.084.59458-83a4203)

4. 详细日志
2020-03-28 22:07:01 /etc/rc.common: line 391: /usr/sbin/trojan: not found , 1 Threads Started!
2020-03-28 22:06:11 /etc/rc.common: line 391: /usr/sbin/trojan: not found , 1 Threads Started!
2020-03-28 22:05:22 /etc/rc.common: line 391: /usr/sbin/trojan: not found , 1 Threads Started!

EnnawYang, 2020-03-28:
没有安装trojan

cuijian83, 2020-03-28:
> 没有安装trojan

ShadowsocksR Plus+里并没有整合Trojan吗，还是说我的固件版本里被精简掉了？
我的SSR Plus是ShadowSocksR Plus+ 设置（支持SS/SSR/V2RAY/TROJAN/SOCKS5/TUN）这样的

Promix953, 2020-03-28:
编译的时候没有把trojan程序编译进去，如果是自己编译的就去ssrp下面有选择trojan，按空格选成[*]就可以了，如果是下的别人的固件，找编译发布的人，和这边没有关系，不是issue

atlantis414, 2020-03-29:
> 编译的时候没有把trojan程序编译进去，如果是自己编译的就去ssrp下面有选择trojan，按空格选成[*]就可以了，如果是下的别人的固件，找编译发布的人，和这边没有关系，不是issue

Trojan默认就是选中的吧? 但是编译出来也没有

closed 2020-03-30 by coolsnowwolf

ShayneL, 2022-09-16:
（系统 - 启动项 ）里面看看，是不是被禁用了