coolsnowwolf/lede #4087: 绕过中国大陆模式下，SSR翻墙经常出现无法连接部分网站

z826540272, 2020-03-29:
模式为 绕过中国大陆与非常用端口
症状表现，经常性出现手机端无法访问google play商店APP（play.google.com)或者youtube 无法播放视频
或者可以访问谷歌商店但是无法加载出图片内容（lh3.googleusercontent.com）。
[image]

此时手机端查询无法访问的域名的DNS解析记录显示正常，来自美国谷歌公司，但是仍然无法访问，无法加载。
将手机关掉再开启WIFI后，所有无法加载的内容变得立即可用，我可以确定绝非代理服务器问题，手机本地平台使用ssr ,v2ray 稳定性无问题。
电脑端Windows10未发现此问题，手机端每日必然出现，问题伴随我使用lean固件至今。
测试系统从android 6.0到10.0不等均经常发现。
问题出现频率为经常，几乎每次使用谷歌商店都需要重新开关wifi或者等待几分钟自然恢复，然后一段时间后问题又会复现。
无法访问时查询dns未发现解析到facebook，twitter等ip

txdywy, 2020-03-29:
类似情况,还没找到原因

z826540272, 2020-03-29:
有趣的事情发生了
无法访问的时候，使用ping tools app将无法访问的域名http ping 一下，就马上恢复了
测试地址是 
lh3.googleusercontent.com，为play商店图片内容的域名。

z826540272, 2020-03-29:
> 类似情况,还没找到原因
> […](#)

问题暂时测试或许解决，打开游戏模式增强客户端，添加对应游戏模式地址。未经深度测试，请反馈

txdywy, 2020-03-29:
> 无法播放视频 或者可以访问谷歌商店但是无法加载出图片内容（lh3.googleusercontent.com）。 [image:
> Screenshot_2020-03-29-11-31-51]

试试看

z826540272, 2020-03-29:
打开udp模式问题未解决，还是会出现无法访问谷歌的一些网站

snakwu, 2020-03-29:
> 打开udp模式问题未解决，还是会出现无法访问谷歌的一些网站

请问怎么打开udp模式？谢谢！

z826540272, 2020-03-29:
打开udp也还是有这样情况

sirfromprc, 2020-03-29:
有ipv6就会出现上述症状

hcym, 2020-03-29:
鬼话，我都是特地编译ipv6的，何况大雕默认值没有ipv6

swxk521, 2020-03-29:
这个问题长期一直存在

klever1988, 2020-03-29:
#https://github.com/coolsnowwolf/lede/issues/3275

z826540272, 2020-03-29:
> 有ipv6就会出现上述症状

无ipv6,且未编译ipv6

z826540272, 2020-03-29:
> ##3275
我编译特地关掉了ipv6,我再去尝试关闭一下过滤 dns v6解析
不过我记得以前没这个功能的时候,ssr+也是经常出现谷歌无法联网的情况.

closed 2020-03-29 by coolsnowwolf

fhbyljj, 2021-05-26:
https://github.com/coolsnowwolf/lede/issues/4169