coolsnowwolf/lede #4095: 两台MT7621无法进行Client WDS连接

zanjie1999, 2020-03-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用pandavan作为主路由,lede作为附路由,使用 Client(WDS) 时出现 ```wlan1: Authentication with xx:xx:xx:xx:xx:xx timed out.```,使用Client时问题相同

### (2) 路由器型号和固件版本

A：最新pandavan的k2p和使用今晚最新代码编译的lede的newifi d1

### (3) 详细日志

A：

```Sun Mar 29 05:03:25 2020 daemon.notice netifd: Network device 'wlan1' link is up
Sun Mar 29 05:03:25 2020 daemon.notice wpa_supplicant[8858]: wlan1: Associated with xx:xx:xx:xx:xx:xx
Sun Mar 29 05:03:25 2020 daemon.notice wpa_supplicant[8858]: wlan1: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Sun Mar 29 05:03:35 2020 daemon.notice wpa_supplicant[8858]: wlan1: Authentication with xx:xx:xx:xx:xx:xx timed out.
Sun Mar 29 05:03:35 2020 daemon.notice wpa_supplicant[8858]: wlan1: CTRL-EVENT-DISCONNECTED bssid=xx:xx:xx:xx:xx:xx reason=3 locally_generated=1
Sun Mar 29 05:03:35 2020 daemon.notice wpa_supplicant[8858]: wlan1: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="weslie_5G" auth_failures=8 duration=90 reason=CONN_FAILED
Sun Mar 29 05:03:35 2020 daemon.notice netifd: Network device 'wlan1' link is down```

1974294589, 2020-06-06:
好像是MTK 组WDS都有问题，无密码可以成功，有密码就失败
Mon May 25 19:45:31 2020 kern.info kernel: [   37.894921] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
Mon May 25 19:45:31 2020 daemon.notice netifd: wan (1173): udhcpc: started, v1.30.0
Mon May 25 19:45:32 2020 authpriv.warn dropbear[1227]: Failed listening on '22': Error listening: Address not available
Mon May 25 19:45:32 2020 authpriv.info dropbear[1227]: Not backgrounding
Mon May 25 19:45:32 2020 daemon.notice netifd: wan (1173): udhcpc: sending discover
Mon May 25 19:45:32 2020 daemon.notice procd: /etc/rc.d/S60flowoffload: net.ipv4.tcp_congestion_control = cubic
Mon May 25 19:45:33 2020 daemon.notice netifd: radio0 (1122): command failed: Not supported (-122)
Mon May 25 19:45:33 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon May 25 19:45:33 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon May 25 19:45:33 2020 user.notice root: ip link set dev wlan0 up
Mon May 25 19:45:35 2020 kern.info kernel: [   40.626784] ieee80211 phy0: rt2800_rf_self_txdc_cal: Info - RF Tx self calibration start
Mon May 25 19:45:35 2020 kern.info kernel: [   40.635596] ieee80211 phy0: rt2800_rf_self_txdc_cal: Info - RF Tx self calibration end
Mon May 25 19:45:35 2020 daemon.notice procd: /etc/rc.d/S60flowoffload: DNSMASQ change
Mon May 25 19:45:36 2020 kern.info kernel: [   42.769370] ieee80211 phy0: rt2800_loft_iq_calibration: Info - LOFT Calibration Done!
Mon May 25 19:45:36 2020 kern.info kernel: [   42.781325] ieee80211 phy0: rt2800_iq_search: Info - IQCalibration Start!
Mon May 25 19:45:36 2020 kern.info kernel: [   42.802610] ieee80211 phy0: rt2800_iq_search: Info - IQCalibration Done! CH = 0, (gain= 1, phase= 2)
Mon May 25 19:45:36 2020 kern.info kernel: [   42.812842] ieee80211 phy0: rt2800_iq_search: Info - IQCalibration Start!
Mon May 25 19:45:36 2020 kern.info kernel: [   42.834114] ieee80211 phy0: rt2800_iq_search: Info - IQCalibration Done! CH = 1, (gain= 1, phase=3e)
Mon May 25 19:45:36 2020 kern.info kernel: [   42.843466] ieee80211 phy0: rt2800_loft_iq_calibration: Info - TX IQ Calibration Done!
Mon May 25 19:45:36 2020 kern.info kernel: [   42.865165] ieee80211 phy0: rt2800_rxiq_calibration: Info - RXIQ G_imb=1, Ph_rx=3
Mon May 25 19:45:36 2020 kern.info kernel: [   42.883036] ieee80211 phy0: rt2800_rxiq_calibration: Info - RXIQ G_imb=0, Ph_rx=0
Mon May 25 19:45:36 2020 kern.info kernel: [   42.897537] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Mon May 25 19:45:36 2020 daemon.notice netifd: wan (1173): udhcpc: sending discover
Mon May 25 19:45:36 2020 daemon.notice netifd: radio0 (1122): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Mon May 25 19:45:37 2020 daemon.notice netifd: radio0 (1122): Command failed: Invalid argument
Mon May 25 19:45:37 2020 kern.info kernel: [   43.372898] br-lan: port 2(wlan0) entered blocking state
Mon May 25 19:45:37 2020 kern.info kernel: [   43.378416] br-lan: port 2(wlan0) entered disabled state
Mon May 25 19:45:37 2020 kern.info kernel: [   43.384207] device wlan0 entered promiscuous mode
Mon May 25 19:45:38 2020 daemon.notice procd: /etc/rc.d/S90kms: KMS Server has started.
Mon May 25 19:45:38 2020 daemon.notice wpa_supplicant[938]: wlan0: SME: Trying to authenticate with 00:0c:43:76:20:78 (SSID='OpenWrt' freq=2462 MHz)
Mon May 25 19:45:38 2020 kern.info kernel: [   45.088293] wlan0: authenticate with 00:0c:43:76:20:78
Mon May 25 19:45:38 2020 daemon.notice wpa_supplicant[938]: wlan0: Trying to associate with 00:0c:43:76:20:78 (SSID='OpenWrt' freq=2462 MHz)
Mon May 25 19:45:38 2020 kern.info kernel: [   45.120352] wlan0: send auth to 00:0c:43:76:20:78 (try 1/3)
Mon May 25 19:45:38 2020 kern.info kernel: [   45.129994] wlan0: authenticated
Mon May 25 19:45:38 2020 kern.info kernel: [   45.144555] wlan0: associate with 00:0c:43:76:20:78 (try 1/3)
Mon May 25 19:45:38 2020 kern.info kernel: [   45.154547] wlan0: RX AssocResp from 00:0c:43:76:20:78 (capab=0x421 status=0 aid=1)
Mon May 25 19:45:38 2020 kern.info kernel: [   45.162786] wlan0: associated
Mon May 25 19:45:38 2020 daemon.notice wpa_supplicant[938]: wlan0: Associated with 00:0c:43:76:20:78
Mon May 25 19:45:38 2020 daemon.notice wpa_supplicant[938]: wlan0: CTRL-EVENT-CONNECTED - Connection to 00:0c:43:76:20:78 completed [id=0 id_str=]
Mon May 25 19:45:38 2020 daemon.notice netifd: Network device 'wlan0' link is up
Mon May 25 19:45:38 2020 kern.info kernel: [   45.168074] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Mon May 25 19:45:38 2020 kern.info kernel: [   45.174894] br-lan: port 2(wlan0) entered blocking state
Mon May 25 19:45:38 2020 kern.info kernel: [   45.180323] br-lan: port 2(wlan0) entered forwarding state
Mon May 25 19:45:39 2020 daemon.notice wpa_supplicant[938]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Mon May 25 19:45:39 2020 daemon.notice netifd: wan (1173): udhcpc: sending discover
Mon May 25 19:45:40 2020 daemon.notice procd: /etc/rc.d/S96led: setting up led wifi
Mon May 25 19:45:44 2020 cron.info crond[1636]: crond (busybox 1.30.0) started, log level 5
Mon May 25 19:45:46 2020 daemon.info procd: - init complete -
Mon May 25 19:45:50 2020 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Mon May 25 19:45:51 2020 user.notice upnp daemon: external interface not found, not starting
Mon May 25 19:46:18 2020 user.notice firewall: Reloading firewall due to ifup of wan6 (eth0.2)
[... cut ...]
Mon May 25 19:48:28 2020 kern.info kernel: [  104.197678] wlan0: deauthenticated from 00:0c:43:76:20:78 (Reason: 2=PREV_AUTH_NOT_VALID)
Mon May 25 19:48:28 2020 daemon.notice netifd: Network device 'wlan0' link is down
Mon May 25 19:48:28 2020 daemon.notice wpa_supplicant[938]: wlan0: CTRL-EVENT-DISCONNECTED bssid=00:0c:43:76:20:78 reason=2
Mon May 25 19:48:28 2020 daemon.notice wpa_supplicant[938]: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="OpenWrt" auth_failures=2 duration=20 reason=CONN_FAILED

1974294589, 2020-06-06:
可能是现在的代码MTK的WDS都有问题，7620桥接别的AP，没密码的就成功，加密码后就一直报错
Thu Jun  4 20:24:17 2020 daemon.notice wpa_supplicant[3856]: wlan0: SME: Trying to authenticate with aa:0c:43:76:20:78 (SSID='Open' freq=2462 MHz)
Thu Jun  4 20:24:17 2020 kern.info kernel: [ 5452.398702] wlan0: authenticate with aa:0c:43:76:20:78
Thu Jun  4 20:24:17 2020 kern.info kernel: [ 5452.411726] wlan0: send auth to aa:0c:43:76:20:78 (try 1/3)
Thu Jun  4 20:24:17 2020 daemon.notice wpa_supplicant[3856]: wlan0: Trying to associate with aa:0c:43:76:20:78 (SSID='Open' freq=2462 MHz)
Thu Jun  4 20:24:17 2020 kern.info kernel: [ 5452.425214] wlan0: authenticated
Thu Jun  4 20:24:17 2020 kern.info kernel: [ 5452.435266] wlan0: associate with aa:0c:43:76:20:78 (try 1/3)
Thu Jun  4 20:24:17 2020 kern.info kernel: [ 5452.447564] wlan0: RX AssocResp from aa:0c:43:76:20:78 (capab=0x431 status=0 aid=1)
Thu Jun  4 20:24:17 2020 kern.info kernel: [ 5452.463792] wlan0: associated
Thu Jun  4 20:24:17 2020 daemon.notice netifd: Network device 'wlan0' link is up
Thu Jun  4 20:24:17 2020 daemon.notice wpa_supplicant[3856]: wlan0: Associated with aa:0c:43:76:20:78
Thu Jun  4 20:24:17 2020 daemon.notice wpa_supplicant[3856]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Thu Jun  4 20:24:18 2020 daemon.notice hostapd: handle_probe_req: send failed
Thu Jun  4 20:24:18 2020 daemon.notice hostapd: handle_probe_req: send failed
Thu Jun  4 20:24:21 2020 kern.info kernel: [ 5453.680659] wlan0: deauthenticated from aa:0c:43:76:20:78 (Reason: 2=PREV_AUTH_NOT_VALID)
Thu Jun  4 20:24:21 2020 daemon.notice netifd: Network device 'wlan0' link is down
Thu Jun  4 20:24:21 2020 daemon.notice wpa_supplicant[3856]: wlan0: CTRL-EVENT-DISCONNECTED bssid=aa:0c:43:76:20:78 reason=2
Thu Jun  4 20:24:23 2020 daemon.notice wpa_supplicant[3856]: wlan0: SME: Trying to authenticate with aa:0c:43:76:20:78 (SSID='Open' freq=2462 MHz)
Thu Jun  4 20:24:23 2020 kern.info kernel: [ 5453.866990] wlan0: authenticate with aa:0c:43:76:20:78
Thu Jun  4 20:24:23 2020 kern.info kernel: [ 5453.880063] wlan0: send auth to aa:0c:43:76:20:78 (try 1/3)
Thu Jun  4 20:24:23 2020 daemon.notice wpa_supplicant[3856]: wlan0: Trying to associate with aa:0c:43:76:20:78 (SSID='Open' freq=2462 MHz)
Thu Jun  4 20:24:23 2020 kern.info kernel: [ 5453.893468] wlan0: authenticated
Thu Jun  4 20:24:23 2020 kern.info kernel: [ 5453.903535] wlan0: associate with aa:0c:43:76:20:78 (try 1/3)
Thu Jun  4 20:24:23 2020 kern.info kernel: [ 5453.915957] wlan0: RX AssocResp from aa:0c:43:76:20:78 (capab=0x431 status=0 aid=1)
Thu Jun  4 20:24:23 2020 kern.info kernel: [ 5453.931895] wlan0: associated
Thu Jun  4 20:24:23 2020 daemon.notice netifd: Network device 'wlan0' link is up
Thu Jun  4 20:24:23 2020 daemon.notice wpa_supplicant[3856]: wlan0: Associated with aa:0c:43:76:20:78
Thu Jun  4 20:24:23 2020 daemon.notice wpa_supplicant[3856]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Thu Jun  4 20:24:26 2020 daemon.notice hostapd: handle_probe_req: send failed
Thu Jun  4 20:24:26 2020 daemon.notice hostapd: handle_probe_req: send failed
Thu Jun  4 20:24:27 2020 kern.info kernel: [ 5454.368420] wlan0: deauthenticated from aa:0c:43:76:20:78 (Reason: 2=PREV_AUTH_NOT_VALID)
Thu Jun  4 20:24:27 2020 daemon.notice netifd: Network device 'wlan0' link is down
Thu Jun  4 20:24:27 2020 daemon.notice wpa_supplicant[3856]: wlan0: CTRL-EVENT-DISCONNECTED bssid=aa:0c:43:76:20:78 reason=2
Thu Jun  4 20:24:28 2020 daemon.notice wpa_supplicant[3856]: wlan0: SME: Trying to authenticate with aa:0c:43:76:20:78 (SSID='Open' freq=2462 MHz)
Thu Jun  4 20:24:28 2020 kern.info kernel: [ 5454.549892] wlan0: authenticate with aa:0c:43:76:20:78
Thu Jun  4 20:24:28 2020 kern.info kernel: [ 5454.562948] wlan0: send auth to aa:0c:43:76:20:78 (try 1/3)
Thu Jun  4 20:24:28 2020 daemon.notice wpa_supplicant[3856]: wlan0: Trying to associate with aa:0c:43:76:20:78 (SSID='Open' freq=2462 MHz)
Thu Jun  4 20:24:28 2020 kern.info kernel: [ 5454.576674] wlan0: authenticated
Thu Jun  4 20:24:28 2020 kern.info kernel: [ 5454.586764] wlan0: associate with aa:0c:43:76:20:78 (try 1/3)
Thu Jun  4 20:24:28 2020 kern.info kernel: [ 5454.599211] wlan0: RX AssocResp from aa:0c:43:76:20:78 (capab=0x431 status=0 aid=1)
Thu Jun  4 20:24:28 2020 kern.info kernel: [ 5454.615148] wlan0: associated
Thu Jun  4 20:24:28 2020 daemon.notice netifd: Network device 'wlan0' link is up
Thu Jun  4 20:24:28 2020 daemon.notice wpa_supplicant[3856]: wlan0: Associated with aa:0c:43:76:20:78
Thu Jun  4 20:24:28 2020 daemon.notice wpa_supplicant[3856]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Thu Jun  4 20:24:29 2020 daemon.notice hostapd: handle_probe_req: send failed
Thu Jun  4 20:24:29 2020 daemon.notice hostapd: handle_probe_req: send failed
Thu Jun  4 20:24:29 2020 daemon.notice hostapd: handle_probe_req: send failed
Thu Jun  4 20:24:29 2020 daemon.notice hostapd: handle_probe_req: send failed
Thu Jun  4 20:24:29 2020 daemon.notice hostapd: handle_probe_req: send failed
Thu Jun  4 20:24:29 2020 daemon.notice hostapd: handle_probe_req: send failed
Thu Jun  4 20:24:29 2020 daemon.notice hostapd: handle_probe_req: send failed
Thu Jun  4 20:24:31 2020 daemon.notice hostapd: handle_probe_req: send failed
Thu Jun  4 20:24:31 2020 daemon.notice hostapd: handle_probe_req: send failed
Thu Jun  4 20:24:32 2020 kern.info kernel: [ 5455.085307] wlan0: deauthenticated from aa:0c:43:76:20:78 (Reason: 2=PREV_AUTH_NOT_VALID)
[... cut ...]
Thu Jun  4 20:25:46 2020 daemon.notice hostapd: handle_probe_req: send failed
Thu Jun  4 20:25:46 2020 daemon.notice hostapd: handle_probe_req: send failed
Thu Jun  4 20:25:47 2020 daemon.notice hostapd: handle_probe_req: send failed
Thu Jun  4 20:25:47 2020 daemon.notice hostapd: handle_probe_req: send failed
Thu Jun  4 20:25:47 2020 daemon.notice hostapd: handle_probe_req: send failed
Thu Jun  4 20:25:48 2020 daemon.notice hostapd: handle_probe_req: send failed
Thu Jun  4 20:25:48 2020 daemon.notice hostapd: handle_probe_req: send failed

closed 2020-07-04 by coolsnowwolf