coolsnowwolf/lede #4096: 昨日新版固件，rtl8152外置usb网卡无法识别

lonelysoul, 2020-03-30:
Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
昨日程序编译的固件，配置make menuconfig时确认rtl8152芯片选中；
固件安装后，未识别usb网卡；
使用L大的固件（OpenWrt-CI #122: Scheduled），同样无法识别usb网卡。

lonelysoul, 2020-03-30:
补充，config文件中，以下两行包含8152，均为y：

CONFIG_DEFAULT_kmod-usb-net-rtl8152=y
CONFIG_PACKAGE_kmod-usb-net-rtl8152=y

lonelysoul, 2020-03-30:
再补充config文件内容：
CONFIG_DEFAULT_kmod-usb-hid=y
CONFIG_DEFAULT_kmod-usb-net=y
CONFIG_DEFAULT_kmod-usb-net-asix=y
CONFIG_DEFAULT_kmod-usb-net-asix-ax88179=y
CONFIG_DEFAULT_kmod-usb-net-rtl8150=y
CONFIG_DEFAULT_kmod-usb-net-rtl8152=y

closed 2020-04-01 by lonelysoul

crosys, 2020-04-01:
> 再补充config文件内容：
> CONFIG_DEFAULT_kmod-usb-hid=y
> CONFIG_DEFAULT_kmod-usb-net=y
> CONFIG_DEFAULT_kmod-usb-net-asix=y
> CONFIG_DEFAULT_kmod-usb-net-asix-ax88179=y
> CONFIG_DEFAULT_kmod-usb-net-rtl8150=y
> CONFIG_DEFAULT_kmod-usb-net-rtl8152=y

同样无法驱动RTL8152的问题，请问如何解决的？

lonelysoul, 2020-04-01:
自编译固件无法使用，直接用L大官方固件，可以驱动；
你再编译一次看看，新版应该可以了；我现在没有自己编译固件了；

crosys, 2020-04-01:
> 自编译固件无法使用，直接用L大官方固件，可以驱动；
> 你再编译一次看看，新版应该可以了；我现在没有自己编译固件了；

弱弱的问一下，L大官方固件在哪里发布？。。。

lonelysoul, 2020-04-02:
官方GitHub上的actions里面啊

获取 Outlook for iOS<https://aka.ms/o0ukef>

lonelysoul, 2020-04-02:
你用新版代码还是没有驱动吗？

获取 Outlook for iOS<https://aka.ms/o0ukef>