coolsnowwolf/lede #4098: ali ddns不能用

muziling, 2020-03-30:
082537       : http://alidns.aliyuncs.com/?Action=DescribeSubDomainRecords&SubDomain=隐藏.com&Format=JSON&TTL=600&Version=2015-01-09&AccessKeyId=隐藏&SignatureMethod=HMAC-SHA1&Timestamp=2020-03-30T00%3A25%3A37Z&SignatureVersion=1.0&SignatureNonce=2fcc74a3-c196-456c-b19d-5b568177ac7b&Signature=隐藏:
2020-03-30 08:25:37 ERROR 400: Bad Request.
 082538  WARN : 传输失败 - 57/0 在 60 秒后重试

logread里的报错：
Mon Mar 30 09:01:05 2020 user.warn ddns-scripts[9872]: myddns_ipv4: 传输失败 - 92/0 在 60 秒后重试
Mon Mar 30 09:02:05 2020 user.err ddns-scripts[9872]: myddns_ipv4: wget 错误代码: '8'

tone88, 2020-03-30:
你去控制台看一下，应该是已经更新了。   
最近也遇到报错，但是实际上已经是成功解析新地址。

muziling, 2020-03-30:
> 你去控制台看一下，应该是已经更新了。
> 最近也遇到报错，但是实际上已经是成功解析新地址。

控制台看了，并没有更新， 另外我用的KEY是阿里云建议用的子用户 的KEY

tone88, 2020-03-30:
> > 你去控制台看一下，应该是已经更新了。
> > 最近也遇到报错，但是实际上已经是成功解析新地址。
> 
> 控制台看了，并没有更新， 另外我用的KEY是阿里云建议用的子用户 的KEY
这是我的日志报错信息，但是控制台是更新了IP的。    我没用子用户KEY
[image]

helllkz, 2020-03-30:
应该是没安装wget的原因吧

muziling, 2020-03-30:
> 应该是没安装wget的原因吧

有的，这边action里下的固件，默认带

you-mu, 2020-04-01:
上周还是啥时候L大更新过一次这个DDNS, 需要将配置删除重新配置一次.
有尝试过删除重配的操作吗?
我用的是ipv6 , 然后用的ali DDNS , 刚测试可以更新.

muziling, 2020-04-01:
配置是新的， 重置过所有配置， 可能是子用户不行，我有空看看用主账号的用户 密钥 试试

muziling, 2020-04-01:
主账号可以用。

closed 2020-04-01 by muziling