coolsnowwolf/lede #4101: SSR Plus 订阅不能用-linksys EA8500

NorwayLuo, 2020-03-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
A：关于Linksys EA8500的没有提及。

## 2. 详细叙述

A：Linksys EA8500,刷了很多个版本了，SSR的订阅都不能使用，只能手工添加节点。用了passwall订阅功能可用，说明节点订阅没有问题。

### (1) 具体问题
A：先点“更新订阅URL列表”，再点“更新所有服务器节点”，节点一个也不会自动生成。

### (2) 路由器型号和固件版本
A：Linksys EA8500
OpenWrt R20.3.19 / LuCI Master (git-20.085.19816-df13074)

### (3) 详细日志
2020-03-30 10:58:44 V2Ray 4.23.1 (OpenWrt) Lean (go1.14 linux/arm) Started!
2020-03-30 10:58:40 重启服务成功
2020-03-30 10:58:40 发生错误, 正在恢复服务
	[C]: ?
	/usr/share/shadowsocksr/subscribe.lua:434: in main chunk
	[C]: in function 'xpcall'
	/usr/share/shadowsocksr/subscribe.lua:343: in function </usr/share/shadowsocksr/subscribe.lua:294>
	/usr/share/shadowsocksr/subscribe.lua:264: in function 'processData'
	/usr/share/shadowsocksr/subscribe.lua:436: in function </usr/share/shadowsocksr/subscribe.lua:434>
2020-03-30 10:58:40 stack traceback:
2020-03-30 10:58:40 /usr/share/shadowsocksr/subscribe.lua:264: attempt to concatenate field 'server_port' (a nil value)
2020-03-30 10:58:40 成功解析: v2ray 节点, 台湾HInet
2020-03-30 10:58:40 成功解析: v2ray 节点, 🇯🇵 日本高速
2020-03-30 10:58:40 成功解析: v2ray 节点, 🇰🇷 韩国SK
2020-03-30 10:58:40 成功解析: v2ray 节点, 美国
2020-03-30 10:58:40 成功解析: v2ray 节点, 🇺🇸 美国原生-CF中转
2020-03-30 10:58:40 成功解析: v2ray 节点, 🇺🇸  美国原生-CN2中转
2020-03-30 10:58:40 成功解析: v2ray 节点, 🇺🇸  美国CN2
2020-03-30 10:58:40 成功解析: v2ray 节点, 🇭🇰 香港家宽
2020-03-30 10:58:40 成功解析: v2ray 节点, 🇭🇰 香港综合
2020-03-30 10:58:40 成功解析: v2ray 节点, 🇭🇰 香港HKBN
2020-03-30 10:58:40 成功解析: v2ray 节点, 🇭🇰 香港PCCW

coolsnowwolf, 2020-03-30:
鸡场那边的问题，把广告去掉即可

coolsnowwolf, 2020-03-30:
简单来说，鸡场把广告做进了节点里面，但不提供“服务器端口” 这个数值信息，导致回滚

Promix953, 2020-03-30:
用订阅节点关键字过滤功能过滤掉就行了

NorwayLuo, 2020-03-30:
谢谢各位的及时回复，我刚试了把所有的广告过滤功能都禁用了，也没有设任何过滤，还是不能使用，错误如下：
备注：用passwall是可以刷下来的。

2020-03-30 14:07:43 V2Ray 4.23.1 (OpenWrt) Lean (go1.14 linux/arm) Started!
2020-03-30 14:07:30 V2Ray 4.23.1 (OpenWrt) Lean (go1.14 linux/arm) Started!
2020-03-30 14:07:26 重启服务成功
2020-03-30 14:07:26 发生错误, 正在恢复服务
	[C]: ?
	/usr/share/shadowsocksr/subscribe.lua:434: in main chunk
	[C]: in function 'xpcall'
	/usr/share/shadowsocksr/subscribe.lua:343: in function </usr/share/shadowsocksr/subscribe.lua:294>
	/usr/share/shadowsocksr/subscribe.lua:264: in function 'processData'
	/usr/share/shadowsocksr/subscribe.lua:436: in function </usr/share/shadowsocksr/subscribe.lua:434>
2020-03-30 14:07:26 stack traceback:
2020-03-30 14:07:26 /usr/share/shadowsocksr/subscribe.lua:264: attempt to concatenate field 'server_port' (a nil value)

Promix953, 2020-03-30:
是你的订阅下来的节点里面有节点是没有任何设置的，空的节点，没让你关广告过滤，和广告过滤有什么关系？
用浏览器打开订阅链接，看看是哪个节点有问题，然后到：
服务--SSRP+--服务器节点--自动更新的下面的下面的下面-->订阅节点关键字过滤
里面添加需要过滤的关键字（添加格式看输入框下面文字提示）
用软件的时候多看看界面上的字，又不是英文看不懂

NorwayLuo, 2020-03-30:
好的，谢谢。

closed 2020-03-30 by NorwayLuo