coolsnowwolf/lede #4103: Newwifi D2 固件编译失败

caolixiang, 2020-03-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
#3803
## 2. 详细叙述
各种尝试编译都失败，在node这个环节
### (1) 具体问题

A：
Newwifi D2 无法正常编译，尝试了make downlad、make clean
### (2) 路由器型号和固件版本
Newwifi D2
A：

### (3) 详细日志

A：
OSError: [Errno 8] Exec format error
tools/v8_gypfiles/generate_bytecode_builtins_list.target.mk:13: recipe for target '/home/mmzz/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/out/Release/obj/gen/generate-bytecode-output-root/builtins-generated/bytecodes-builtins-list.h' failed
make[5]: *** [/home/mmzz/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/out/Release/obj/gen/generate-bytecode-output-root/builtins-generated/bytecodes-builtins-list.h] Error 1
Makefile:101: recipe for target 'node' failed
make[4]: *** [node] Error 2
make[4]: Leaving directory '/home/mmzz/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0'
Makefile:150: recipe for target '/home/mmzz/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/.built' failed
make[3]: *** [/home/mmzz/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/.built] Error 2
make[3]: Leaving directory '/home/mmzz/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#15.89#29.76#52.71
package/Makefile:111: recipe for target 'package/feeds/packages/node/compile' failed
make[2]: *** [package/feeds/packages/node/compile] Error 2
make[2]: Leaving directory '/home/mmzz/lede'
package/Makefile:107: recipe for target '/home/mmzz/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/mmzz/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/mmzz/lede'
/home/mmzz/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

waynon, 2020-04-04:
我昨天刚刚编译了newifi D2的最新固件，没有任何问题，可以编译。
如果需要无线功能可以使用另一个仓库下面的7615驱动

icefruit-j, 2020-04-04:
> 我昨天刚刚编译了newifi D2的最新固件，没有任何问题，可以编译。
> 如果需要无线功能可以使用另一个仓库下面的7615驱动，直接编译lede仓库下的源码是没有无线的

有无线，node的问题，大概是环境的问题吧，
sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib
更新一下试试啥的...

waynon, 2020-04-04:
(没有无线，需要在package中添加这个 [https://github.com/MeIsReallyBa/k2p-openwrt-mt7615_5.0.2.0)]

Gokou-Ruri, 2020-05-13:
> > > 我昨天刚刚编译了newifi D2的最新固件，没有任何问题，可以编译。
> > > 如果需要无线功能可以使用另一个仓库下面的7615驱动，直接编译lede仓库下的源码是没有无线的
> > 
> > 
> > 有无线，node的问题，大概是环境的问题吧，
> > sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib
> > 更新一下试试啥的...
> 
> 没有无线，需要在package中添加这个 https://github.com/MeIsReallyBa/k2p-openwrt-mt7615_5.0.2.0才会有无线

你们说的无线是说WIFI吗？我用LEDE全默认参数编译了一个D2的固件，给D2刷入后是有WIFI的，其它设备可以正常连接WIFI网络。

closed 2020-07-04 by coolsnowwolf