coolsnowwolf/lede #4104: kcptun-20200321 编译失败

jasonhouang, 2020-03-30:
```bash
go: extracting github.com/coreos/go-iptables v0.4.2
go: extracting github.com/xtaci/kcp-go/v5 v5.5.9
go: downloading github.com/tjfoc/gmsm v1.3.0
go: downloading golang.org/x/net v0.0.0-20200301022130-244492dfa37a
go: downloading github.com/klauspost/reedsolomon v1.9.3
go: extracting github.com/klauspost/reedsolomon v1.9.3
verifying github.com/tjfoc/gmsm@v1.3.0: checksum mismatch
        downloaded: h1:qhgkrZru95jFP9NbVPknJvc9vgkMXhOEzkOASKdc0oQ=
        go.sum:     h1:i7c6Za/IlgBvnGxYpfD7L3TGuaS+v6oGcgq+J9/ecEA=

Building targets
go: finding github.com/xtaci/kcptun/.go_work/build latest
go: finding github.com/xtaci/kcptun/.go_work latest
Fetching https://github.com?go-get=1
Parsing meta tags from https://github.com?go-get=1 (status code 200)
can't load package: package github.com/xtaci/kcptun/.go_work/build: unknown import path "github.com/xtaci/kcptun/.go_work/build": cannot find module providing package github.com/xtaci/kcptun/.go_work/build

Cleaning module download cache (golang/go#27455)
go: finding github.com/xtaci/kcptun/.go_work/build latest
go: finding github.com/xtaci/kcptun/.go_work latest
can't load package: package github.com/xtaci/kcptun/.go_work/build: unknown import path "github.com/xtaci/kcptun/.go_work/build": cannot find module providing package github.com/xtaci/kcptun/.go_work/build

Makefile:61: recipe for target '/home/jason/Workspaces/embedded/openwrt/lede/build_dir/target-mipsel_24kc_musl/kcptun-20200321/.built' failed
make[3]: *** [/home/jason/Workspaces/embedded/openwrt/lede/build_dir/target-mipsel_24kc_musl/kcptun-20200321/.built] Error 1
make[3]: Leaving directory '/home/jason/Workspaces/embedded/openwrt/lede/package/lean/kcptun'
time: package/lean/kcptun/compile#7.68#2.42#33.54
package/Makefile:111: recipe for target 'package/lean/kcptun/compile' failed
make[2]: *** [package/lean/kcptun/compile] Error 2
make[2]: Leaving directory '/home/jason/Workspaces/embedded/openwrt/lede'
package/Makefile:107: recipe for target '/home/jason/Workspaces/embedded/openwrt/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/jason/Workspaces/embedded/openwrt/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jason/Workspaces/embedded/openwrt/lede'
/home/jason/Workspaces/embedded/openwrt/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```

lxwq, 2020-03-30:
阿里云
    配置如下：
export GOPROXY=https://mirrors.aliyun.com/goproxy/

    nexus社区提供的
    配置如下：
export GOPROXY=https://gonexus.dev

    goproxy.io 的
    配置如下：
export GOPROXY=https://goproxy.io/

    基于athens的公共服务
    配置如下：
export GOPROXY=https://athens.azurefd.net

    官方提供的(jfrog,golang)
export GOPROXY=https://gocenter.io
export GOPROXY=https://proxy.golang.org

    七牛云赞助支持的
export GOPROXY=https://goproxy.cn

jasonhouang, 2020-03-30:
手动修改了go.sum先让它编译过，然后又遇到
```bash
go: extracting golang.org/x/net v0.0.0-20190311183353-d8887717615a
go: extracting golang.org/x/crypto v0.0.0-20190308221718-c2843e01d9a2
go: downloading golang.org/x/text v0.3.0
go: extracting golang.org/x/sys v0.0.0-20190215142949-d0b11bdaac8a
go: extracting google.golang.org/genproto v0.0.0-20180831171423-11092d34479b
go: extracting golang.org/x/text v0.3.0
build v2ray.com/core/main: cannot load crypto/ed25519: cannot find module providing package crypto/ed25519

Building targets
Fetching https://v2ray.com/core/.go_work/build?go-get=1
https fetch failed: Get https://v2ray.com/core/.go_work/build?go-get=1: dial tcp: lookup v2ray.com on 30.208.4.8:53: read udp 192.168.1.3:37332->30.208.4.8:53: read: connection refused
Fetching https://v2ray.com/core/.go_work?go-get=1
Parsing meta tags from https://v2ray.com/core/.go_work?go-get=1 (status code 200)
get "v2ray.com/core/.go_work": found meta tag get.metaImport{Prefix:"v2ray.com/core", VCS:"git", RepoRoot:"https://github.com/v2ray/v2ray-core"} at https://v2ray.com/core/.go_work?go-get=1
get "v2ray.com/core/.go_work": verifying non-authoritative meta tag
Fetching https://v2ray.com/core?go-get=1
Parsing meta tags from https://v2ray.com/core?go-get=1 (status code 200)
go: finding v2ray.com/core/.go_work latest
Fetching https://v2ray.com?go-get=1
Parsing meta tags from https://v2ray.com?go-get=1 (status code 200)
can't load package: package v2ray.com/core/.go_work/build: unknown import path "v2ray.com/core/.go_work/build": cannot find module providing package v2ray.com/core/.go_work/build

Cleaning module download cache (golang/go#27455)
go: finding v2ray.com/core/.go_work/build latest
go: finding v2ray.com/core/.go_work latest
can't load package: package v2ray.com/core/.go_work/build: unknown import path "v2ray.com/core/.go_work/build": cannot find module providing package v2ray.com/core/.go_work/build

Makefile:307: recipe for target '/home/jason/Workspaces/embedded/openwrt/lede/build_dir/target-mipsel_24kc_musl/v2ray-core-4.23.1/.built' failed
make[3]: *** [/home/jason/Workspaces/embedded/openwrt/lede/build_dir/target-mipsel_24kc_musl/v2ray-core-4.23.1/.built] Error 1
make[3]: Leaving directory '/home/jason/Workspaces/embedded/openwrt/lede/package/lean/v2ray'
time: package/lean/v2ray/compile#18.61#5.19#60.16
package/Makefile:111: recipe for target 'package/lean/v2ray/compile' failed
make[2]: *** [package/lean/v2ray/compile] Error 2
make[2]: Leaving directory '/home/jason/Workspaces/embedded/openwrt/lede'
package/Makefile:107: recipe for target '/home/jason/Workspaces/embedded/openwrt/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/jason/Workspaces/embedded/openwrt/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jason/Workspaces/embedded/openwrt/lede'
/home/jason/Workspaces/embedded/openwrt/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```

jasonhouang, 2020-03-30:
解决方法：
更新 go 版本`1.12.1`至`1.13.3`

操作：
修改`feeds/packages/lang/golang/golang-version.mk`
> GO_VERSION_MAJOR_MINOR:=1.13
> GO_VERSION_PATCH:=3
修改`feeds/packages/lang/golang/golang/Makefile`
> PKG_HASH:=4f7123044375d5c404280737fbd2d0b17064b66182a65919ffe20ffe8620e3df

编译顺利通过~

AmadeusGhost, 2020-03-31:
是不是没update feeds？package那go已经是1.14了 https://github.com/coolsnowwolf/packages/commit/066cb5691162939bf189289b9cd0eba8e045b216

jasonhouang, 2020-03-31:
> 是不是没update feeds？package那go已经是1.14了 [coolsnowwolf/packages@066cb56](https://github.com/coolsnowwolf/packages/commit/066cb5691162939bf189289b9cd0eba8e045b216)

应该是的这个原因，很久没有`update feeds`了，不知道是不是增加了自己的`package`，执行`./scripts/feeds update -a` 没有更新

jasonhouang, 2020-03-31:
> > 是不是没update feeds？package那go已经是1.14了 [coolsnowwolf/packages@066cb56](https://github.com/coolsnowwolf/packages/commit/066cb5691162939bf189289b9cd0eba8e045b216)
> 
> 应该是的这个原因，很久没有`update feeds`了，不知道是不是增加了自己的`package`，执行`./scripts/feeds update -a` 没有更新

找到原因：默认的`feeds.conf` 被我修改过，需要用`feeds.conf.default`替换下，然后
> ./scripts/feeds update -a
> ./scripts/feeds install -a

jasonhouang, 2020-04-13:
不修改`go.sum`就编译不过kcptun
```bash
make[3]: Entering directory '/home/jason/Workspaces/embedded/openwrt/lede/package/lean/kcptun'
rm -f /home/jason/Workspaces/embedded/openwrt/lede/build_dir/target-mips_24kc_musl/kcptun-20200409/.built
touch /home/jason/Workspaces/embedded/openwrt/lede/build_dir/target-mips_24kc_musl/kcptun-20200409/.built_check
( cd /home/jason/Workspaces/embedded/openwrt/lede/build_dir/target-mips_24kc_musl/kcptun-20200409/.go_work/build ; export               GOOS=linux GOARCH=mips GO386= GOARM= GOMIPS=softfloat GOMIPS64= CGO_ENABLED=1 CC=mips-openwrt-linux-musl-gcc CXX=mips-openwrt-linux-musl-g++ CGO_CFLAGS="-Os -pipe -mno-branch-likely -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/jason/Workspaces/embedded/openwrt/lede/build_dir/target-mips_24kc_musl/kcptun-20200409:kcptun-20200409 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/jason/Workspaces/embedded/openwrt/lede/staging_dir/target-mips_24kc_musl/usr/include -I/home/jason/Workspaces/embedded/openwrt/lede/staging_dir/target-mips_24kc_musl/include -I/home/jason/Workspaces/embedded/openwrt/lede/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/usr/include -I/home/jason/Workspaces/embedded/openwrt/lede/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include/fortify -I/home/jason/Workspaces/embedded/openwrt/lede/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include" CGO_CXXFLAGS="-Os -pipe -mno-branch-likely -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/jason/Workspaces/embedded/openwrt/lede/build_dir/target-mips_24kc_musl/kcptun-20200409:kcptun-20200409 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/jason/Workspaces/embedded/openwrt/lede/staging_dir/target-mips_24kc_musl/usr/lib -L/home/jason/Workspaces/embedded/openwrt/lede/staging_dir/target-mips_24kc_musl/lib -L/home/jason/Workspaces/embedded/openwrt/lede/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/usr/lib -L/home/jason/Workspaces/embedded/openwrt/lede/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib -znow -zrelro"   GOPATH=/home/jason/Workspaces/embedded/openwrt/lede/build_dir/target-mips_24kc_musl/kcptun-20200409/.go_work/build GOCACHE=/home/jason/Workspaces/embedded/openwrt/lede/build_dir/target-mips_24kc_musl/kcptun-20200409/.go_work/cache GOENV=off ; echo "Finding targets" ; targets=$(go list github.com/xtaci/kcptun/...) ; for pattern in ; do targets=$(echo "$targets" | grep -v "$pattern") ; done ; echo ; if [ "" = 1 ]; then echo "Calling go generate" ; go generate -v  $targets ; echo ; fi ; if [ "" != 1 ]; then echo "Building targets" ; case mips in arm)             installsuffix="v" ;; mips|mipsle)     installsuffix="softfloat" ;; mips64|mips64le) installsuffix="" ;; esac ; ldflags="-linkmode external -extldflags '-L/home/jason/Workspaces/embedded/openwrt/lede/staging_dir/target-mips_24kc_musl/usr/lib -L/home/jason/Workspaces/embedded/openwrt/lede/staging_dir/target-mips_24kc_musl/lib -L/home/jason/Workspaces/embedded/openwrt/lede/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/usr/lib -L/home/jason/Workspaces/embedded/openwrt/lede/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib -Wl,-z,now -Wl,-z,relro'" ; pkg_gcflags="" ; pkg_ldflags="-s -w -X 'main.VERSION=20200409-1 for OpenWrt'" ; for def in ; do pkg_ldflags="$pkg_ldflags -X $def" ; done ; go install  ${installsuffix:+-installsuffix $installsuffix} -trimpath -ldflags "all=$ldflags" -v ${pkg_gcflags:+-gcflags "$pkg_gcflags"} ${pkg_ldflags:+-ldflags "$pkg_ldflags $ldflags"}  $targets ; retval=$? ; echo ; if [ "$retval" -eq 0 ] && [ -z "$(find /home/jason/Workspaces/embedded/openwrt/lede/build_dir/target-mips_24kc_musl/kcptun-20200409/.go_work/build/bin/linux_mips -maxdepth 0 -type d \! -empty 2>/dev/null)" ]; then echo "No binaries were generated, consider adding GO_PKG_SOURCE_ONLY:=1 to Makefile" ; echo ; fi ; echo "Cleaning module download cache (golang/go#27455)" ; go clean -modcache ; echo ; fi ; exit $retval ; )
Finding targets
go: downloading github.com/xtaci/kcp-go v5.4.20+incompatible
go: downloading github.com/pkg/errors v0.9.1
go: downloading golang.org/x/crypto v0.0.0-20200320181102-891825fb96df
go: downloading github.com/xtaci/tcpraw v1.2.25
go: downloading github.com/golang/snappy v0.0.1
go: downloading github.com/urfave/cli v1.21.0
go: downloading github.com/xtaci/smux v1.5.12
go: downloading github.com/google/gopacket v1.1.17
go: downloading github.com/coreos/go-iptables v0.4.2
go: downloading github.com/xtaci/kcp-go/v5 v5.5.11
go: downloading github.com/klauspost/reedsolomon v1.9.3
go: downloading golang.org/x/net v0.0.0-20200320220750-118fecf932d8
go: downloading github.com/tjfoc/gmsm v1.3.0
go: downloading github.com/templexxx/xorsimd v0.4.1
verifying github.com/tjfoc/gmsm@v1.3.0: checksum mismatch
        downloaded: h1:qhgkrZru95jFP9NbVPknJvc9vgkMXhOEzkOASKdc0oQ=
        go.sum:     h1:i7c6Za/IlgBvnGxYpfD7L3TGuaS+v6oGcgq+J9/ecEA=

SECURITY ERROR
This download does NOT match an earlier download recorded in go.sum.
The bits may have been replaced on the origin server, or an attacker may
have intercepted the download attempt.

For more information, see 'go help module-auth'.

Building targets
[... cut ...]
make[2]: Leaving directory '/home/jason/Workspaces/embedded/openwrt/lede'
package/Makefile:107: recipe for target '/home/jason/Workspaces/embedded/openwrt/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/jason/Workspaces/embedded/openwrt/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jason/Workspaces/embedded/openwrt/lede'
/home/jason/Workspaces/embedded/openwrt/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```

hzjnet, 2020-05-24:
最后如何处理喃？

closed 2020-07-04 by coolsnowwolf