coolsnowwolf/lede #4108: 树莓派4 sysupgrade固件包在线升级问题

IvanSolis1989, 2020-03-30:
我之前编译了一个openwrt烧写到树莓派4正常使用。后来调整了一些插件和驱动选项重新编译。固件文件分区大小没有调过，但是用sysupgrade包上传刷写，路由器10秒左右就重新进入登录页面，然后系统里面的插件也没有增减，请问什么情况下才能用sysupgrade直接升级？不用每次都重新烧录整个系统？

AmadeusGhost, 2020-03-31:
给个升级时候的系统日志？

EnnawYang, 2020-03-31:
至少你还能开机，我在线升级之后从来没有成功开过机，每次都要取卡刷写（

AmadeusGhost, 2020-03-31:
[image]
试一下这样改下源码？本机的`/lib/upgrade/platform.sh`也照样改下，有些选项我不确定就暂时没改，先这样改下试试吧。

coolsnowwolf, 2020-04-03:
具体日志有么

AmadeusGhost, 2020-04-03:
除了之前有一部分未被定义的变量，主要还有`package/base-files/files/lib/upgrade/common.sh`的更改

Troyshen94, 2020-04-04:
至少你还能开机，我在线升级之后就会进入无限重启....

AmadeusGhost, 2020-04-04:
> 具体日志有么

这个issue可以关了，没人愿意测试或者给日志，也就是说这个问题不存在。

Troyshen94, 2020-04-04:
> > 具体日志有么
> 
> 这个issue可以关了，没人愿意测试或者给日志，也就是说这个问题不存在。

抱歉 我也希望能够给出日志
但是实际遇到的问题是 引导之后的页面都进不去 直接连树莓派可以看到跑了一会儿代码就自动重启
没有办法导出日志
我晚一点试试能不能拍下来

AmadeusGhost, 2020-04-04:
> 抱歉 我也希望能够给出日志

这个日志是挺麻烦的，好像只能接显示器看，有个可能有问题的地方是官方大改了upgrade的commom.sh，虽然官方x86支持uefi了，没设备测试改起来也是挺蛋疼的。

PR #4720 (2020-06-06): treewide: adjust sysupgrade script
This commit fix sysupgrade failed for bcm27xx, fix #4072, fix #4108 

Run-tested-by: @1715173329 Dalao on Raspberry Pi 3B
Run-tested-by: @EnnawYang Dalao on Raspberry Pi 4B
closed 2020-06-06 by coolsnowwolf