coolsnowwolf/lede #4113: 编译出来的img可以直接web更新使用，但是.vmdk的虚拟机文件，在虚拟机里不能正常启动，用img转换的vmdk也试过

0jack-lei0, 2020-03-30:
[image]

hhmaomao, 2020-03-30:
考虑一下vm里面的虚拟网卡转换一下，把主机的虚拟网卡绑定到vmnet1或其他vmnet，然后vmnet的模式设置为host only，更改后再试试就可以了，一般这个问题存在都是vm的问题，更改一下网卡后一般都没问题

hue715, 2020-03-31:
有安装open-vm-tools试试吗？

0jack-lei0, 2020-04-01:
> 有安装open-vm-tools试试吗？

这个openwrt里不能上网要如何安装tools

0jack-lei0, 2020-04-01:
哎，搞一天，还是没有办法

hue715, 2020-04-01:
> > 有安装open-vm-tools试试吗？
> 
> 这个openwrt里不能上网要如何安装tools

可以考虑在编译时选择open-vm-tools这个包

Greeed-islanl, 2020-04-02:
最好你用软件转一下嘛好像叫“V2V”

lonwern, 2020-04-03:
遇到过类似的问题，在虚拟机设置里把虚拟磁盘扩展到1G就可以了

[image]

d975xbx, 2020-04-03:
vmkfstools -i centos-original_xxx.vmdk -d thin centos-original_0_新的xxx.vmdk.
转换以后要有两个文件才ok，一个是xxx.vmdk，另一个是xxx flat.vmdk.一起上传到esxi。

luxiaok, 2020-06-16:
在VM配置里把硬盘类型改成 `IDE` 试试~

closed 2020-07-04 by coolsnowwolf