coolsnowwolf/lede #4115: 3月份以后编译的版本状态页wan口显示的是lan口信息

badcrazy, 2020-03-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue
具体什么时候出现不清楚了，讲真没去关注，但是看了下2月24日的一个截图还是正常的，肯定是之后，这个具体是什么原因？真心求教
[image]

coolsnowwolf, 2020-03-30:
废话，你的 LAN IP 的 网关也是自己

badcrazy, 2020-03-31:
> 废话，你的 LAN IP 的 网关也是自己

感谢lean大回复，我就是想知道怎么能让这里正确显示wan口，谢谢

niceguy3-31, 2020-03-31:
我也碰到过，这样还会无法上网，重置下好了

shu0802, 2020-03-31:
与你的一样
[image]

shu0802, 2020-03-31:
> 我也碰到过，这样还会无法上网，重置下好了

重置后是什么状态

shu0802, 2020-03-31:
原来是设置LAN口处原因，这样设置就正常了
[image]

xayane, 2020-03-31:
master/package/lean/default-settings/files/zzz-default-settings
可以修改默认配置，可以根据自己的需求自定义。

closed 2020-03-31 by coolsnowwolf