coolsnowwolf/lede #4116: 请教lean大或者其他大神一个问题，希望能在百忙中拨冗解答一下

norikohxb, 2020-03-30:
[image]
我的旁路由是lean大的op，最新的版本。就编译了adbyby和ssrplus，连upnp都不编译。
旁路由ip是192.168.16.248，旁路由的lan口直接设置网关是ros 192.168.16.252
这种情况下，是不是旁路由不编译upnp是没有任何影响的。
而且旁路由的防火墙的菜单设置不设置其实对于旁路由的效能一点影响都没有
而且是不是旁路由的fullcone-nat其实开启或者不开启也是一点影响都没有的
因为压根儿数据就没有从wan口出去，也不做ip的伪装
NAT= 路由+伪装
而在这种旁路由下，其实都不存在伪装了。因为都是内网同一个网段，旁路由其实有点像是一个二级交换机。
我这样理解对吗？

hxlls, 2020-03-30:
你数据走OP的lan 出去肯定会过WAN ，你的图上都存在了2个区域，也就是存在WAN到LAN的转发，NAT是地址转发，248-252也是地址转换，所以这个“旁路由”你看成是一个DHCP和一个网段的2级路由就完事了

norikohxb, 2020-03-31:
> 你数据走OP的lan 出去肯定会过WAN ，你的图上都存在了2个区域，也就是存在WAN到LAN的转发，NAT是地址转发，248-252也是地址转换，所以这个“旁路由”你看成是一个DHCP和一个网段的2级路由就完事了

因为op的wan口甚至可以删除是虚拟的，所以其实是不存在wan口，只是在lan口的网卡直接指定了ros作为网关，这样一来其实应该并没有NAT的行为吧，有的只是路由的行为。既然不存在WAN口，WAN口可以删除，防火墙这块其实是没有用到，就算你设置了，但是其实并没有效果，因为他是作为LAN-》WAN的ip伪装和防止攻击的。其实整个旁路由说白了，是一个彻彻底底的路由器，只是通过SSR的科学上网走了trojan或者v2ray出去。

norikohxb, 2020-04-11:
> 你数据走OP的lan 出去肯定会过WAN ，你的图上都存在了2个区域，也就是存在WAN到LAN的转发，NAT是地址转发，248-252也是地址转换，所以这个“旁路由”你看成是一个DHCP和一个网段的2级路由就完事了

感谢你啊，我发的帖子好像都是你在解答

closed 2020-04-11 by norikohxb