coolsnowwolf/lede #4118: 求助-对unraid-linux qemu的支持

vshen009, 2020-03-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在unraid平台下的虚拟机里面使用，

- 发现img可以启动，但是不能正常生成 /etc/config/network 文件， 

- 用vi /erc/config/network 发现不能保存文件。

- 网卡没有自动加载， 用过手动命令ifconfig 可以指定ip，并且连接到web页面

- 登陆web后找到网络接口部分，提示原始配置文件找不到，也不能保存。

### (2) 路由器型号和固件版本
我用的是 unraid 系统，基于liunx的x86系统，安装在我的i3 cpu 笔记本。虚拟机技术栈应该是 linux下的qemu
搜索了openwrt论坛也没找到更多信息 不过发现一个插件叫 qemu-qa 安装之后也没什么变化。
请问需要怎么排除错误。

A：

### (3) 详细日志

A：

muziling, 2020-03-31:
我的unraid正常， 网络上几篇unraid跑op的文章，也没人有反映这个问题
unraid笔记本支持很有限吧，至少我的笔记本启动不进去

vshen009, 2020-03-31:
> 我的unraid正常， 网络上几篇unraid跑op的文章，也没人有反映这个问题
> unraid笔记本支持很有限吧，至少我的笔记本启动不进去

我其他都正常的 unriad 运行没问题，也能正常启动其他虚拟vm ，包括linux windows

vshen009, 2020-03-31:
> 我的unraid正常， 网络上几篇unraid跑op的文章，也没人有反映这个问题
请教下 opt编译出来的版本 选什么文件格式？  用seabios 的固件启动么

muziling, 2020-03-31:
选squashfs 的img，  ext4 img应该也可以，  不能选seabios

vshen009, 2020-03-31:
> 选squashfs 的img， ext4 img应该也可以， 不能选seabios

感谢我回去试试，还有一个选项 和硬件相关 是选i440fx 还是另外一个

muziling, 2020-03-31:
q35 4.2

closed 2020-07-04 by coolsnowwolf