coolsnowwolf/lede #4123: 建议MWAN3分流助手增加更新的功能

gzxiexl, 2020-03-31:
我的使用场景是单软路由，联通+电信双线接入，不做多拨，仅策略分流，联通ip走联通，其他走电信，v2ray走联通。
但是mwan3的分流助手无法更新ip表，建议大佬能新增这个功能，感谢！！

coolsnowwolf, 2020-04-01:
这个功能没什么人用，已经打算放弃了

waynon, 2020-04-08:
这个功能其实还是有些用的，如果家里有两条宽带的话。
联通线路使用非cn2的代理一般都比电信好

rshi0212, 2020-04-08:
> 我的使用场景是单软路由，联通+电信双线接入，不做多拨，仅策略分流，联通ip走联通，其他走电信，v2ray走联通。
> 但是mwan3的分流助手无法更新ip表，建议大佬能新增这个功能，感谢！！

我是目前联通100M和移动200M（4拨）用mwan3均衡（20%x5），同时用ssr+，看到mwan3分流助手说和ssr+冲突，从来没用过mwan3分流助手。

Readeon, 2020-04-09:
L大别放弃啊！我正在处理一个场景，用旁路由兼顾做IPTV的分流，正需要mwan3。

gzxiexl, 2020-04-10:
求大佬能继续优化，双线接入的家庭还是不少的，否则也不会这么多双软路由的人。

xpc0938, 2020-04-10:
我目前软路由上方案: EXSI虚拟+爱快+LEDE
爱快负责多播、DDNS和端口转发，LEDE用来爬墙和samba。另外还有ac86u+ac1900P mesh组网做二级路由，提供WiFi。多系统或者多设备独立，各做各的事，网络还算比较稳定。

rshi0212, 2020-04-11:
> 我目前软路由上方案: EXSI虚拟+爱快+LEDE
> 爱快负责多播、DDNS和端口映射，LEDE用来爬墙和samba。另外还有ac86u+ac1900P mesh组网做二级路由，提供WiFi。多系统或者多设备独立，各做各的事，网络还算比较稳定。给并发多播的同学提供下思路。

不明白为什么要那么复杂，那么多层NAT，是软路由性能不够嘛？

rshi0212, 2020-04-11:
> 我目前软路由上方案: EXSI虚拟+爱快+LEDE
> 爱快负责多播、DDNS和端口映射，LEDE用来爬墙和samba。另外还有ac86u+ac1900P mesh组网做二级路由，提供WiFi。多系统或者多设备独立，各做各的事，网络还算比较稳定。给并发多播的同学提供下思路。

除非是网络终端数量大于一个网段内的ip数量，或者说有特殊管理需求需要划分vlan隔离终端设备，或者有设备硬件能力的限制，一般家庭情况网络架构都是从简设计，何况ESXi服务器用的还是同设备上的硬件资源，后面的WiFi硬路由组网能当做AP就做AP用。家用这点负载也看不出性能的损失。

chenxuan5858, 2020-05-19:
同意，希望L大能把插件继续维持下去，确实两条宽带很需要！！

closed 2020-07-04 by coolsnowwolf

wenpo1975, 2022-09-30:
> 我的使用场景是单软路由，联通+电信双线接入，不做多拨，仅策略分流，联通ip走联通，其他走电信，v2ray走联通。 但是mwan3的分流助手无法更新ip表，建议大佬能新增这个功能，感谢！！

https://ispip.clang.cn/
这里有最新得运营商IP地址段，TXT格式下和分流助手生成得IP地址段格式排版都是一样的
个人认为是可以到相应目录下直接替换
另外IP段确实有些变化