coolsnowwolf/lede #4124: zerotier遭遇运营商UDP QOS

TXMAN, 2020-03-31:
感谢Lean大佬集成zerotier，非常好用，内网用户的福音。但是由于zerotier基于UDP打洞，而国内运营商广泛存在UDP QOS，所以在一些情况下，尤其是跨营运商时，存在ping值低（10ms）但丢包率很高(>50%)的情况，导致实际使用起来卡顿，大流量使用时更是只能做3秒“真男人”的窘状。

TXMAN, 2020-03-31:
大佬能否考虑将zerotier流量伪装成TCP，来绕过UDP QOS？

TXMAN, 2020-03-31:
例如参考这个项目：https://github.com/wangyu-/udp2raw-tunnel

ZqinKing, 2020-03-31:
大哥ZeroTier走的是点对点隧道，点对点怎么走TCP？你可以自建moon节点，UDP打洞不顺畅的时候会自行切换至moon节点走TCP转发。

ZqinKing, 2020-03-31:
openWrt的ZeroTier节点中要放置moon节点的配置文件，需要修改/etc/zerotier文件。具体怎么修改自行百度。

ZqinKing, 2020-03-31:
不过关于moon的倒是可以让大雕支持一下，搞个上传框。把moon节点的配置文件上传一下。

TXMAN, 2020-03-31:
> 不过关于moon的倒是可以让大雕支持一下，搞个上传框。把moon节点的配置文件上传一下。

点对点也是可以技术上伪装成tcp的，详见我发的链接中udp2raw项目。QOS带来的问题不是打洞不畅，而是直连后丢包严重，这种情况下moon服务器可能也不会被触发转发。 另外，lean大已经把zerotier的配置文件夹放到ect路径下了，moon服务配置起来不难，我已经配置了。但是zerotier优势就在于直连，如果依赖服务器转发的话，一方面使用国内服务器的话流量很贵，二是使用国外服务器的话延迟和稳定性又差。所以还是的优先通过技术手段伪装udp使用直连更好。

badcrazy, 2020-03-31:
我一直以为是握手服务器被墙造成的，原来是这个原因啊

nftbty, 2020-03-31:
可是伪装TCP的话，这个应该发到zerotier那边去，发在这不起作用啊，就算这里改了，和其他客户端也是不兼容的啊。不过伪装TCP确实是个很好的建议，在有些网络条件下应该很有用，真的可以反馈给官方试试。我一直使用zerotier连接两个局域网互通，因为是同一运营商宽带，连接质量很好，就是受限上传带宽，传文件慢。不过有段时间也几乎快不能用。有时在外用手机流量连接的话，时好时坏的，很多时候只是勉强能通，但是很慢，和宽带还是同一运营商的。
有谁英语好的，可以去zerotier one那边提个issue，绝对顶贴支持。

TXMAN, 2020-03-31:
> 可是伪装TCP的话，这个应该发到zerotier那边去，发在这不起作用啊，就算这里改了，和其他客户端也是不兼容的啊。不过伪装TCP确实是个很好的建议，在有些网络条件下应该很有用，真的可以反馈给官方试试。我一直使用zerotier连接两个局域网互通，因为是同一运营商宽带，连接质量很好，就是受限上传带宽，传文件慢。不过有段时间也几乎快不能用。有时在外用手机流量连接的话，时好时坏的，很多时候只是勉强能通，但是很慢，和宽带还是同一运营商的。
> 有谁英语好的，可以去zerotier one那边提个issue，绝对顶贴支持。

如果zerotier官方支持tcp当然最好不过了。目前在官方不支持的情况下，参考udp2raw项目的方法，利用udp2raw tunnel来伪装zerotier的udp的流量应该也是现阶段一个可行的思路。

coolsnowwolf, 2020-03-31:
tcp 是无法打洞的

coolsnowwolf, 2020-03-31:
因为TCP和UDP在打洞行为是是不同。这是

Socket（标准socket规范）的API造成的:
UDP的socket允许多个socket绑定到同一个本地端口，而TCP的socket则不允许

这是这样一个意思：A B要连接到S，肯定首先A B双方都会在本地创建一个socket，
去连接S上的socket。创建一个socket必然会绑定一个本地端口（就算应用程序里面没写
端口，实际上也是绑定了的，至少java确实如此），假设为8888，这样A和B才分别建立了到
S的通信信道。接下来就需要打洞了，打洞则需要A和B分别发送数据包到对方的公网IP。但是
问题就在这里：因为NAT设备是根据端口号来确定session，如果是UDP的socket，A B可以
分别再创建socket，然后将socket绑定到8888，这样打洞就成功了。

但是如果是TCP的socket，则不能再创建socket并绑定到8888了，这样打洞就无法成功。

TXMAN, 2020-03-31:
> 因为TCP和UDP在打洞行为不同，这是Socket（标准socket规范）的API造成的:
> 
> UDP的socket允许多个socket绑定到同一个本地端口，而TCP的socket则不允许
> 
> 这是这样一个意思：A B要连接到S，肯定首先A B双方都会在本地创建一个socket，
> 去连接S上的socket。创建一个socket必然会绑定一个本地端口（就算应用程序里面没写
> 端口，实际上也是绑定了的），假设为8888，这样A和B才分别建立了到S的通信信道。
> 
> 接下来就需要打洞了，打洞则需要A和B分别发送数据包到对方的公网IP。但是问题就在这里：因为NAT设备是根据端口号来确定session，如果是UDP的socket，A B可以分别再创建socket，然后将socket绑定到8888，这样打洞就成功了。
> 
> 但是如果是TCP的socket，则不能再创建socket并绑定到8888了，这样打洞就无法成功。

感谢lean大亲自从技术原理上解答！我大概明白了TCP打洞的不可行性。不过还有一点想请教：从udp2raw tunnel的项目介绍来看，udp2raw tunnel的原理是“通过raw socket给UDP包加上TCP或ICMP header，进而绕过UDP屏蔽或QoS，或在UDP不稳定的环境下提升稳定性。可以有效防止在使用kcptun或者finalspeed的情况下udp端口被运营商限速。”看上去它只是给UDP的socket加上了TCP的header，这种做法是不是并没有完全把UDP的socket变为TCP的socket？有没有可能在UDP socket中加入TCP的header的行为不会影响到您所说的“UDP的多个socket绑定到同一个本地端口”呢？

TXMAN, 2020-03-31:
或者是，如果UDP伪装TCP的方案技术上不可行，也可以考虑参考https://github.com/wangyu-/UDPspeeder/blob/branch_libev/doc/README.zh-cn.md  借助UDPspeeder通过冗余数据来对抗网络的丢包，据它的项目介绍，能达到“用1.5倍的流量，就可以把10%的丢包率降低到万分之一以下”的效果，也不失为一种对抗运营商UDP QOS的解决方案。

Promix953, 2020-03-31:
> 或者是，如果UDP伪装TCP的方案技术上不可行，也可以考虑参考https://github.com/wangyu-/UDPspeeder/blob/branch_libev/doc/README.zh-cn.md 借助UDPspeeder通过冗余数据来对抗网络的丢包，据它的项目介绍，能达到“用1.5倍的流量，就可以把10%的丢包率降低到万分之一以下”的效果，也不失为一种对抗运营商UDP QOS的解决方案。

UDPspeeder能抗QoS的话，还要开发udp2raw干嘛，两还是一个人开发的。。。可以去装个RouterOS玩玩，了解下QoS原理，针对UDP的QoS，不是简单的多倍发包就能暴力解决的

ZqinKing, 2020-04-01:
> > 不过关于moon的倒是可以让大雕支持一下，搞个上传框。把moon节点的配置文件上传一下。
> 
> 点对点也是可以技术上伪装成tcp的，详见我发的链接中udp2raw项目。QOS带来的问题不是打洞不畅，而是直连后丢包严重，这种情况下moon服务器可能也不会被触发转发。 另外，lean大已经把zerotier的配置文件夹放到ect路径下了，moon服务配置起来不难，我已经配置了。但是zerotier优势就在于直连，如果依赖服务器转发的话，一方面使用国内服务器的话流量很贵，二是使用国外服务器的话延迟和稳定性又差。所以还是的优先通过技术手段伪装udp使用直连更好。

研究了一下你说的udp2raw。它支持的VPN都是有中心服务器的（流量也走服务器），也就是要一个带公网IP的服务器。然而ZeroTier的P2P隧道并在建立后并不需要中心话服务器。ZeroTier的中心服务器只为UDP打洞和TCP转发。

nftbty, 2020-04-01:
@coolsnowwolf 查了一些资料，TCP打洞其实是有的，但比UDP要困难复杂，原因也就是你说的socket端口不能复用的问题。虽然标准socket不支持，但是操作系统支持SO_REUSEADDR参数的话，可以通过setsockopt设置这个参数，就能监听、外连绑定到同一个port，其他就跟UDP差不多了，当然实现起来更复杂，也许成功率也低些。现在主流操作系统都是支持SO_REUSEADDR的，bsd系统还多个SO_REUSEPORT。主流编程语言应该也都支持。

nftbty, 2020-04-01:
不知道QUIC正式发布RFC之后，运营商对quic不知会不会比对其他UDP好一些，如果是的话，可以UDP打洞成功后使用QUIC来传输或者伪装QUIC。
其实还蛮期待DNSoverQUIC的。

Promix953, 2020-04-01:
> @coolsnowwolf 查了一些资料，TCP打洞其实是有的，但比UDP要困难复杂，原因也就是你说的socket端口不能复用的问题。虽然标准socket不支持，但是操作系统支持SO_REUSEADDR参数的话，可以通过setsockopt设置这个参数，就能监听、外连绑定到同一个port，其他就跟UDP差不多了，当然实现起来更复杂，也许成功率也低些。现在主流操作系统都是支持SO_REUSEADDR的，bsd系统还多个SO_REUSEPORT。主流编程语言应该也都支持。

这些意见应该去zerotier项目提issue，固件只是集成和提供前端操作界面

nftbty, 2020-04-01:
@Promix953  这个当然，看我第一条回复，就是建议题主应该发到zerotier那边。只不过看到后面大雕说TCP不能打洞，才回复一下。其实现在有了IPv6，在外面基本用不上zt了。

closed 2020-07-04 by coolsnowwolf

maryjeck, 2020-07-21:
tinc可以打洞吧，貌似基于tcp的

piiiiiiiiiiiiiii, 2020-12-01:
参见问题：怎样给zerotier套上udp2raw-tunnel?

https://github.com/wangyu-/udp2raw-tunnel/issues/332

piiiiiiiiiiiiiii, 2020-12-01:
> tinc可以打洞吧，貌似基于tcp的

tinc也会被搞，我用阿里云，经常用一用就不通，换个端口才行。

牛b吧?

coppercash, 2020-12-18:
> tinc可以打洞吧，貌似基于tcp的

tinc 没打洞成功过，tcp udp 都不行。
zerotier 倒是打洞成功了，但是被疯狂 qos。
现在主要在用的还是 tinc tcp。
但是我的中转节点只有 5m 带宽，卡的很。
也许要搞起 ipv6 了。

piiiiiiiiiiiiiii, 2021-01-16:
> > tinc可以打洞吧，貌似基于tcp的
> 
> tinc 没打洞成功过，tcp udp 都不行。
> zerotier 倒是打洞成功了，但是被疯狂 qos。
> 现在主要在用的还是 tinc tcp。
> 但是我的中转节点只有 5m 带宽，卡的很。
> 也许要搞起 ipv6 了。

你这是配置问题，而非不能用

zclikezc, 2022-08-16:
大佬们是在说给zerotier架设自己的服务器？
现在想给自己搞一个没有公网ip的服务器
然后来用来架设zerotier中转站？
反正搞不懂咋弄了，不管咋样好像都得有个公网ip，不知道可不可以直接用ipv6？在中转的时候？
能不能用cpolar这种把服务器暴露在公网来做中转服务器？
哎想科学上网咋就这么复杂

Cathgao, 2022-08-17:
你把UDP伪装成TCP之后，那谁去拆包成UDP呢？总不能让zerotier官方去专门做个udp2raw吧？如果你要专门给zerotier伪装开个公网服务器，那zerotier的意义在哪？

maryjeck, 2022-08-17:
> 或者是，如果UDP伪装TCP的方案技术上不可行，也可以考虑参考https://github.com/wangyu-/UDPspeeder/blob/branch_libev/doc/README.zh-cn.md 借助UDPspeeder通过冗余数据来对抗网络的丢包，据它的项目介绍，能达到“用1.5倍的流量，就可以把10%的丢包率降低到万分之一以下”的效果，也不失为一种对抗运营商UDP QOS的解决方案。

这个估计是不行的，我测试过，电信的网络，Qos是直接给你换ip，说白了，就是闪断重联，zerotier又要重新打洞

maryjeck, 2022-08-17:
> 

zerotier数据不加密吧，用于科学上网，那直接给你gfw了，