coolsnowwolf/lede #4126: 这种日志能否改成一秒钟只报一次？

wlshdjj, 2020-03-31:
Tue Mar 31 18:23:39 2020 kern.err /usr/bin/ssr-redir[30758]: getpeername: Socket not connected
Tue Mar 31 18:23:39 2020 kern.err /usr/bin/ssr-redir[30758]: getpeername: Socket not connected
Tue Mar 31 18:23:39 2020 kern.err /usr/bin/ssr-redir[30760]: getpeername: Socket not connected
Tue Mar 31 18:23:39 2020 kern.err /usr/bin/ssr-redir[30758]: getpeername: Socket not connected
Tue Mar 31 18:23:39 2020 kern.err /usr/bin/ssr-redir[30760]: getpeername: Socket not connected
Tue Mar 31 18:23:39 2020 kern.err /usr/bin/ssr-redir[30758]: getpeername: Socket not connected
Tue Mar 31 18:23:39 2020 kern.err /usr/bin/ssr-redir[30760]: getpeername: Socket not connected
Tue Mar 31 18:23:39 2020 kern.err /usr/bin/ssr-redir[30758]: getpeername: Socket not connected
Tue Mar 31 18:23:39 2020 kern.err /usr/bin/ssr-redir[30758]: getpeername: Socket not connected
Tue Mar 31 18:23:39 2020 kern.err /usr/bin/ssr-redir[30758]: getpeername: Socket not connected
Tue Mar 31 18:23:39 2020 kern.err /usr/bin/ssr-redir[30758]: getpeername: Socket not connected
Tue Mar 31 18:23:39 2020 kern.err /usr/bin/ssr-redir[30760]: getpeername: Socket not connected
Tue Mar 31 18:23:39 2020 kern.err /usr/bin/ssr-redir[30758]: getpeername: Socket not connected

通屏在刷哈

wlshdjj, 2020-03-31:
Tue Mar 31 14:59:57 2020 kern.err /usr/bin/ssr-redir[15862]: accept: No file descriptors available
Tue Mar 31 14:59:57 2020 kern.err /usr/bin/ssr-redir[15862]: accept: No file descriptors available
Tue Mar 31 14:59:57 2020 kern.err /usr/bin/ssr-redir[15862]: accept: No file descriptors available
Tue Mar 31 14:59:57 2020 kern.err /usr/bin/ssr-redir[15862]: accept: No file descriptors available
Tue Mar 31 14:59:57 2020 kern.err /usr/bin/ssr-redir[15862]: accept: No file descriptors available
Tue Mar 31 14:59:57 2020 kern.err /usr/bin/ssr-redir[15862]: accept: No file descriptors available
Tue Mar 31 14:59:57 2020 kern.err /usr/bin/ssr-redir[15862]: accept: No file descriptors available
Tue Mar 31 14:59:57 2020 kern.err /usr/bin/ssr-redir[15862]: accept: No file descriptors available
Tue Mar 31 14:59:57 2020 kern.err /usr/bin/ssr-redir[15862]: accept: No file descriptors available
Tue Mar 31 14:59:57 2020 kern.err /usr/bin/ssr-redir[15862]: accept: No file descriptors available
Tue Mar 31 14:59:57 2020 kern.err /usr/bin/ssr-redir[15862]: accept: No file descriptors available
Tue Mar 31 14:59:57 2020 kern.err /usr/bin/ssr-redir[15862]: accept: No file descriptors available
Tue Mar 31 14:59:57 2020 kern.err /usr/bin/ssr-redir[15862]: accept: No file descriptors available

这个更加离谱，刷满全部日志

coolsnowwolf, 2020-03-31:
这个是由socks 底层决定的

closed 2020-04-01 by coolsnowwolf

Format-DJ, 2020-05-21:
> 这个是由socks 底层决定的

请问这个错误的产生是因为什么呢？每天7点我的路由器都会报这个错误 届时SSR服务器的连接会被短暂断开