coolsnowwolf/lede #4140: 一个ROS+OP的迷幻问题

KleinerSource, 2020-04-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述
op 除了 某+服务以外 其他的服务均未开启 
### (1) 具体问题

A：
主路由是routeros 拿op做旁路 另有个wifi路由做AP
dhcp 为ros 设定默认网关到op
但是 只要是 op启动之后  有设备wifi连接到 AP (部分设备 例如小米网关 灯 笔记本电脑等)
就会不停的释放 和获取dhcp地址(15秒左右一次) 无限循环释放获取ip 不管绑定或未绑定都一样
但是在过程中 只要重启op 就都正常了 只要op 重启好之后 连接的部分设备 就会有这种情况发送
op重启过程中 连接wifi设备 都正常.
### (2) 路由器型号和固件版本

A：
软路由 大雕很多个版本都试过 结果一样, 但是试过K某 好像没有这个问题
### (3) 详细日志

A：

iyuangang, 2020-04-01:
跟我家的方案差不多，差异为我的中心思想为一个设备负责任务不能与其他设备重复。
routeros 负责拨号，关闭dhcp服务，除wan口外其他口bridge，分配ip地址R1。
op单口连接routeros lan口，设置ip R2及默认网关为R1，设置DNS， 开启op dhcp。
Wifi 关闭Lan口 dhcp ，设定LAN口ip R3，网线连接wifi lan口及routeros LAN 口。
其中 IP R1 R2 R3 均在同一网段，DHCP 也在同一网段。

KleinerSource, 2020-04-01:
我就是因为喜欢 ros的 window 方式管理  看的直观 操作方便 所以 把dhcp 拨号 等功能全都交给他
op只做一项服务 但是这个dhcp问题一直无解

hue715, 2020-04-01:
网络-接口-br-lan-DHCP，把OP的DHCP关闭了吗？

KleinerSource, 2020-04-01:
dhcp当然关了

nichwang88, 2020-04-06:
我也是主路由ROS（DHCP服务器在ROS中），旁路由openwrt，若干无线路由器为AP，没有出现任何DHCP问题。
你的问题出现在WIFI上，检查你的AP设置是否有问题，是否设置AP模式，或者还是路由模式，DHCP没有关的可能性。

KleinerSource, 2020-04-06:
> 我也是主路由ROS（DHCP服务器在ROS中），旁路由openwrt，若干无线路由器为AP，没有出现任何DHCP问题。
> 
> 你的问题出现在WIFI上，检查你的AP设置是否有问题，是否设置AP模式，或者还是路由模式，DHCP没有关的可能性。

我ap也是同版op 设置也一样。而且只用ks版的lede是没有这问题的 我也很纳闷

KillingJacky, 2020-04-07:
禁用FullCone NAT重启路由器解决。
可能还需要重启运行dhcp client的程序 or 虚机。

KleinerSource, 2020-04-07:
> 禁用FullCone NAT重启路由器解决。
> 
> 可能还需要重启运行dhcp client的程序 or 虚机。

关闭 fullcore 并重启路由器和wifi 均无效

KillingJacky, 2020-04-07:
如果你的ROS是装在某种虚拟环境中，需要重启ROS所在的虚机。
我猜你的某些申请不到IP的WiFi设备是DHCP Discovery的时候带了让DHCP Server以广播方式回应，而你的广播包在层层转发中丢了。
你可以用WireShark抓一下包。

KleinerSource, 2020-04-07:
> 如果你的ROS是装在某种虚拟环境中，需要重启ROS所在的虚机。
> 
> 我猜你的某些申请不到IP的WiFi设备是DHCP Discovery的时候带了让DHCP Server以广播方式回应，而你的广播包在层层转发中丢了。
> 
> 你可以用WireShark抓一下包。

我ros是物理机 只有op是虚拟机 
而且 我之前 ros+ks版lede+lean版op（wifi）是没有这个问题的
换成ros+lean版op+lean版op（wifi）
才出现这个奇怪的问题
而且出问题的设备 基本上都是 米家网关和灯
像手机 平板 笔记本这类东西都没问题

KillingJacky, 2020-04-08:
> > 如果你的ROS是装在某种虚拟环境中，需要重启ROS所在的虚机。
> > 我猜你的某些申请不到IP的WiFi设备是DHCP Discovery的时候带了让DHCP Server以广播方式回应，而你的广播包在层层转发中丢了。
> > 你可以用WireShark抓一下包。
> 
> 我ros是物理机 只有op是虚拟机
> 而且 我之前 ros+ks版lede+lean版op（wifi）是没有这个问题的
> 换成ros+lean版op+lean版op（wifi）
> 才出现这个奇怪的问题
> 而且出问题的设备 基本上都是 米家网关和灯
> 像手机 平板 笔记本这类东西都没问题

我的情况是，lean OP装在pve虚拟环境中作主路由，pve bridge出来接交换机，接AP（普通路由器），也是某些设备无法申请到IP，WiFi association 阶段是可以过的，抓包发现OP发出的dhcp offer、dhcp ack等包根本无法发出OP（在pve bridge中dump的包），确定了OP设置的问题。
同时，pve中还有另外一个隔壁家的lede，替换掉Lean OP，没有任何问题，对比lede与OP的设置发现，OP使能了FullCone NAT，关闭后，dhcp offer包可以正常发出，问题解决。
以上，供您参考。
细看了一下您的情况，好像不太一样，你不是申请不到IP，是反复申请，怀疑还是LAN中有2个DHCP Server。

KleinerSource, 2020-04-08:
> > > 如果你的ROS是装在某种虚拟环境中，需要重启ROS所在的虚机。
> 
> > > 我猜你的某些申请不到IP的WiFi设备是DHCP Discovery的时候带了让DHCP Server以广播方式回应，而你的广播包在层层转发中丢了。
> 
> > > 你可以用WireShark抓一下包。
> 
> > 
> 
> > 我ros是物理机 只有op是虚拟机
> 
> > 而且 我之前 ros+ks版lede+lean版op（wifi）是没有这个问题的
> 
> > 换成ros+lean版op+lean版op（wifi）
> 
> > 才出现这个奇怪的问题
> 
> > 而且出问题的设备 基本上都是 米家网关和灯
> 
> > 像手机 平板 笔记本这类东西都没问题
> 
> 
> 
> 我的情况是，lean OP装在pve虚拟环境中作主路由，pve bridge出来接交换机，接AP（普通路由器），也是某些设备无法申请到IP，WiFi association 阶段是可以过的，抓包发现OP发出的dhcp offer、dhcp ack等包根本无法发出OP（在pve bridge中dump的包），确定了OP设置的问题。
> 
> 同时，pve中还有另外一个隔壁家的lede，替换掉Lean OP，没有任何问题，对比lede与OP的设置发现，OP使能了FullCone NAT，关闭后，dhcp offer包可以正常发出，问题解决。我这边也是有线设备正常，WiFi AP下设备异常，猜测是Linux FullCone实现不够完美，UDP广播包在穿越多级交换后到达Linux Kernel，FullCone之后的包，出不去Linux Bridge，链路比较复杂没有去细究了。
> 
> 以上，供您参考。
> 
> 细看了一下您的情况，好像不太一样，你不是申请不到IP，是反复申请，怀疑还是LAN中有2个DHCP Server。

请问 抓包 再哪个设备上执行呢 
我现在这几个这些小米设备的ip都是10秒一次
ros日志是 dhcp assigned 然后dhcp deassigned

KleinerSource, 2020-04-08:
> 我的情况是，lean OP装在pve虚拟环境中作主路由，pve bridge出来接交换机，接AP（普通路由器），也是某些设备无法申请到IP，WiFi association 阶段是可以过的，抓包发现OP发出的dhcp offer、dhcp ack等包根本无法发出OP（在pve bridge中dump的包），确定了OP设置的问题。
>
> 同时，pve中还有另外一个隔壁家的lede，替换掉Lean [...]

我在ros 有3个段的ip 刚才测试了一下 把有问题的设备调到 1段 就没有这个问题 调回原来的ip段 就这样 我感觉好迷茫

closed 2020-07-04 by coolsnowwolf

norikohxb, 2021-02-11:
我家也是接近250个智能家居博联的设备确实会出现频繁的释放ip地址和获得ip地址，最后好像让ros都错乱了。应该和op没有关系。

norikohxb, 2021-02-11:
哈哈，确实是op引起了ros下面的智能家居不能获取到ip地址。我按照你的方法关闭了op的虚拟机，我的博联sp4l就可以获得ip地址了，太奇葩了。

norikohxb, 2021-02-11:
忘记告诉你了，我的op和ros都是虚拟机上面，也是一堆的智能家居（我有255个博联）都是不停释放和获得ip地址

norikohxb, 2021-02-11:
> 反馈bug/问题模板，提建议请删除
>
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
>
> * [ x ] 没有类似的issue
>
> ## 2. 详细叙述
> op 除了 某+服务以外 其他的服务均未开启
>
> ### (1) 具体问题
> A：
> 主路由是routeros 拿op做旁路 另有个wifi路由做AP
> dhcp 为ros 设定默认网关到op
> 但是 只要是 op启动之后 [...]

请问你现在是怎么解决，智能家居+op+ros的问题。确实不停的释放ip地址

norikohxb, 2021-02-11:
我刚才不死心又重新在esxi上面关闭了op（lean大的，我自己编译的）我的博联的sp4l就真的可以找到ip了，然后我再把op在虚拟机开起来了。现在目前来说正常，但是可以肯定是op导致了。ros无法给这个博联分配ip

norikohxb, 2021-02-11:
> 反馈bug/问题模板，提建议请删除
>
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
>
> * [ x ] 没有类似的issue
>
> ## 2. 详细叙述
> op 除了 某+服务以外 其他的服务均未开启
>
> ### (1) 具体问题
> A：
> 主路由是routeros 拿op做旁路 另有个wifi路由做AP
> dhcp 为ros 设定默认网关到op
> 但是 只要是 op启动之后 [...]

你有没有其他联系方式。我家里智能家居很多，用了接近200多个ip走了。
所以我的电脑手机监控是一个网段，另外一个网段是给博联智能家居而且他们因为不需要翻墙也不用走op他们网关就是ros

norikohxb, 2021-02-11:
[image]
OP没有开dhcp但是好像也干扰了ros分配ip

qq280579056, 2021-05-10:
老哥解决了吗 我也遇到相同的问题

norikohxb, 2021-05-10:
没有解决，偶尔还是会有博联智能家居的设备丢失ip，出于闪灯状态。我家里的博联智能家居已经独享一个192.168.8.x的网段，目前占据了250个设备了
没有多少ip可以用了

IriKaQ, 2021-09-03:
可能IP不够了，可以尝试扩大局域网网段，比如使用172.16.0.0/16。

norikohxb, 2021-12-25:
> 我在ros 有3个段的ip 刚才测试了一下 把有问题的设备调到 1段 就没有这个问题 调回原来的ip段 就这样 我感觉好迷茫

你最后解决了吗？我的智能家居几乎是几分钟就释放和获得一次，我自己修改了lease time到4天也是如此。

norikohxb, 2021-12-25:
> 我在ros 有3个段的ip 刚才测试了一下 把有问题的设备调到 1段 就没有这个问题 调回原来的ip段 就这样 我感觉好迷茫

检查一下你的wifi路由器的wifi的2.4g的设置
是不是启用了 启动 WMM No-Acknowledgement  这个不能启用

lingfeng0303, 2022-10-27:
这个问题最后解决了吗？是原因？我是有线设备一直分配ip

KleinerSource, 2022-11-13:
> 这个问题最后解决了吗？是原因？我是有线设备一直分配ip

哇 2年前的问题。。。我都不记得怎么解决的了 早就恢复正常了 现在依旧是ros加op

gitous, 2023-12-11:
应该是ros的问题，我没用openwrt，就ros配ap，dhcp固定几个设备尤其是手机开了网络调试软件，就能看到隔几分钟切到5g然后切回Wi-Fi