coolsnowwolf/lede #4143: 移植SSR plus到原版OpenWrt SNAPSHOT r12787下的问题

tinnlux, 2020-04-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：移植ssr plus到原版OpenWrt SNAPSHOT，新增节点的 “导入配置信息”功能失效，在lede下无问题。

### (2) 路由器型号和固件版本

A：测试手头x86、Newifi-D2、lenovo-y1、xiaomi-nano等设备在OpenWrt SNAPSHOT下都有此问题。

### (3) 详细日志

A：无错误日志，无报错。

hcym, 2020-04-01:
是滴，导入无效，其它还行，可能用1907分支更好

dongjuanyong, 2020-04-02:
以下patch实验成功：
```
diff --git a/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.lua b/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.lua
index 0ae7bbe688..7b05ce5cfc 100644
--- a/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.lua
+++ b/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.lua
@@ -19,7 +19,7 @@ local cache = {}
 local nodeResult = setmetatable({}, { __index = cache })  -- update result
 local name = 'shadowsocksr'
 local uciType = 'servers'
-local ucic = uci.cursor()
+local ucic = luci.model.uci.cursor()
 local proxy = ucic:get_first(name, 'server_subscribe', 'proxy', '0')
 local switch = ucic:get_first(name, 'server_subscribe', 'switch', '1')
 local subscribe_url = ucic:get_first(name, 'server_subscribe', 'subscribe_url', {})
```

hcym, 2020-04-02:
牛叉啦，不过原版还是用原版ss和v2合适，对ipv6友好，就是少了trojian

dongjuanyong, 2020-04-02:
你说的ipv6友好是指哪方面，dnsmasq屏蔽aaaa查询么，patch都是现成的，考过去改改语句就能用
除了部分内核驱动（不兼容snapshot的5.4内核）别的问题都不大

dongjuanyong, 2020-04-02:
话说我只用v2（mkcp），配合udp2raw防qos，dnscrypt-proxy做doh查询和aaaa屏蔽，稳定好用

haxc, 2020-04-03:
> 以下patch实验成功：
>
> ```
> diff --git a/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.lua b/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.lua
> index [...]

可以提交一下pr

tinnlux, 2020-04-03:
@dongjuanyong 实测失败，你这个补丁是修复不能更新订阅的问题的。修改的文件也是订阅相关。

hcym, 2020-04-03:
原版用哪个passwall也不错，刚刚用snapshot版本编译了，除了没有内置广告其它和官方融合没问题

tty228, 2020-04-03:
`sed -i "s/'cbid.shadowsocksr.'/'widget.cbid.shadowsocksr.'/g" /usr/lib/lua/luci/view/shadowsocksr/ssrurl.htm`

dongjuanyong, 2020-04-04:
subscribe.lua的修改应该对lede没什么影响，ssrurl.htm就不知道了

dongjuanyong, 2020-04-04:
又看了一眼，input的加个widget.能解决，checkbox的id根本对不上，想正确使用的话必须改用getElementsByName

iyuangang, 2020-04-04:
> 又看了一眼，input的加个widget.能解决，checkbox的id根本对不上，想正确使用的话必须改用getElementsByName

经测试，可修复。
[#issues7](https://github.com/iyuangang/openwrt/issues/7)

dongjuanyong, 2020-04-04:
> > 又看了一眼，input的加个widget.能解决，checkbox的id根本对不上，想正确使用的话必须改用getElementsByName
> 
> 经测试，可修复。
> [#issues7](https://github.com/iyuangang/openwrt/issues/7)
> [image]

你再试试，提示成功没问题，但打勾的那种设置（checkbox）都不会生效，因为id根本不对：
[image]
上面那个tls的checkbox的id是cbd31fabcd，你用getElementById抓不到，只能getElementsByName

iyuangang, 2020-04-04:
> > > 又看了一眼，input的加个widget.能解决，checkbox的id根本对不上，想正确使用的话必须改用getElementsByName
> > 
> > 
> > 经测试，可修复。
> > [#issues7](https://github.com/iyuangang/openwrt/issues/7)
> > [image]
> 
> 你再试试，提示成功没问题，但打勾的那种设置（checkbox）都不会生效，因为id根本不对：
> [image]
> 上面那个tls的checkbox的id是cbd31fabcd，你用getElementById抓不到，只能getElementsByName

用的getElementsByName

dongjuanyong, 2020-04-04:
> > > > 又看了一眼，input的加个widget.能解决，checkbox的id根本对不上，想正确使用的话必须改用getElementsByName
> > > 
> > > 
> > > 经测试，可修复。
> > > [#issues7](https://github.com/iyuangang/openwrt/issues/7)
> > > [image]
> > 
> > 
> > 你再试试，提示成功没问题，但打勾的那种设置（checkbox）都不会生效，因为id根本不对：
> > [image]
> > 上面那个tls的checkbox的id是cbd31fabcd，你用getElementById抓不到，只能getElementsByName
> 
> 用的getElementsByName

那就没问题了，有可能的话希望用lede原版（非19.07或snapshot）的朋友试试PR #4154 能兼容的话就方便了

PR #4154 (2020-04-08): luci-app-ssr-plus: OpenWrt SNAPSHOT compatibility
Fix #4143

Q：你知道这是`pull request`吗？(使用 "x" 选择)
* [x] 我知道
commit ae7a5d70 (2020-04-08): luci-app-ssr-plus: OpenWrt SNAPSHOT compatibility (#4154)

* luci-app-ssr-plus: OpenWrt SNAPSHOT compatibility

Fix #4143

* Modify getElementById to getElementsByName

Luci modification for OpenWrt 19.07 or higher
closed 2020-04-08 by coolsnowwolf