coolsnowwolf/lede #4146: x64 无法生成sysupgrade.bin固件吗？factory.bin也没，只有img

zuo000, 2020-04-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

you-mu, 2020-04-02:
For x86 systems there is no “sysupgrade” image, just be sure the new firmware image has the same family of filesystem as your old one. 

https://openwrt.org/docs/guide-user/installation/sysupgrade.cli

zuo000, 2020-04-02:
> For x86 systems there is no “sysupgrade” image, just be sure the new firmware image has the same family of filesystem as your old one.
> 
> https://openwrt.org/docs/guide-user/installation/sysupgrade.cli

thanks!

closed 2020-04-02 by zuo000

AmibaWorld, 2020-04-02:
下载并验证OpenWrt固件升级映像
仅下载和使用以“ -sysupgrade.bin”结尾的OpenWrt固件映像进行命令行升级。
### **对于x86系统，没有“ sysupgrade”映像，只需确保新的固件映像具有与旧系统相同的文件系统家族即可。**