coolsnowwolf/lede #4147: 软路由刷openwrt遇到的奇怪的现象

a1392429292, 2020-04-02:
刷过的固件有：（全部是最新的固件）lede自己编译的，lede别人编译的，openwrt.org官网的，koolshare的；
软路由硬件：G5420+H310软路由主板+8GDDR4
宽带：200M电信单线；
环境：因为发现了问题，所有的包都是刷好设置了lan口和wan口就没有在配置其他的，怕是设置出现了问题；网线线材均是相同；拨号方式也相同；
全部出现同一现象：在测网速的平台上（speedtest.cn/net），均出现极不稳定的现象；大部分的线路测试结果只有30-100M左右，只有极少数的线路是可以跑满在220M左右的
初步判断：可能是线路问题造成的？我把有问题的线路都记录了下来；
验证判断：我将软路由换成了硬路由，而然记录下来跑不满的线路在硬路由上均全部跑满；
再次判断：难道是软路由的硬件有问题了？
验证判断：我将软路由装了Win10系统测宽带在230M左右，跑局域网内也跑100M/s；
------
这真的让我不知道该如何判断问题在哪了，有相关经验的大佬帮分析一下？（排除最基本的问题，不是网线问题，不是水晶头问题）

a1392429292, 2020-04-02:
一个不是专业的怀疑是：软路由在处理小包上性能偏弱，所以，部分的测速线路是采用的小包测验，所以，软路由就显得性能不足了？而这些有问题的线路在硬路由测速起来就显得十分的轻松？
这只是我一个小白的怀疑。。。

fonlan, 2020-04-02:
我之前和你类似的情况，软路由里同时虚拟了爱快和OpenWrt，爱快自带测速轻松500M跑满，但是用OpenWrt上400M都困难，那时候都是用的网上下载的别人做好的包。后来试了下自己编译，只保留自己需要的几个软件，相当精简，结果就稳稳的跑满500M了。所以我怀疑可能某些软件哪怕没有启用，只要编译进去了就会对转发性能有影响，你可以试试编译一个最精简的固件，看看速度能不能上去。

a1392429292, 2020-04-02:
> 我之前和你类似的情况，软路由里同时虚拟了爱快和OpenWrt，爱快自带测速轻松500M跑满，但是用OpenWrt上400M都困难，那时候都是用的网上下载的别人做好的包。后来试了下自己编译，只保留自己需要的几个软件，相当精简，结果就稳稳的跑满500M了。所以我怀疑可能某些软件哪怕没有启用，只要编译进去了就会对转发性能有影响，你可以试试编译一个最精简的固件，看看速度能不能上去。

我刷了openwrt.org最纯净的x86-64的包（只有4M左右）的也出现这样的情况。。。

iyuangang, 2020-04-02:
感觉是硬件的问题。
我的是i5 4590 + 华硕B85 + 两块双口千兆PCIE x1 网卡。自编译OpenWrt 19.07.2 + L大 某+ 包
跑满600M 很稳定。
[image]

fonlan, 2020-04-02:
> > 我之前和你类似的情况，软路由里同时虚拟了爱快和OpenWrt，爱快自带测速轻松500M跑满，但是用OpenWrt上400M都困难，那时候都是用的网上下载的别人做好的包。后来试了下自己编译，只保留自己需要的几个软件，相当精简，结果就稳稳的跑满500M了。所以我怀疑可能某些软件哪怕没有启用，只要编译进去了就会对转发性能有影响，你可以试试编译一个最精简的固件，看看速度能不能上去。
> 
> 我刷了openwrt.org最纯净的x86-64的包（只有4M左右）的也出现这样的情况。。。

那我估计你大概率线路问题，Speedtest毕竟也没能力满足所有地区的测速需求，要选择和自己相同运营商且距离最近的测速点才能达到峰值，其他测速点你不用考虑。

ElectronicElephant, 2020-04-02:
只要有线路可以跑满200M就可以了，不用考虑其他地区的。

over

a1392429292, 2020-04-02:
> 只要有线路可以跑满200M就可以了，不用考虑其他地区的。
> 
> over

可以确定的是，迅雷下载文件，基本都稳定在25+M/s，除非有些文件本身速度就不高的。只要速度高的，全程都是满速在下载，并没有出现   大幅度的 变动。。。。

a1392429292, 2020-04-02:
感谢各位的回答，问题找到了，h310 itx软路由主板 的4个网口，之间有问题，或者说有干扰，只用其中2个是可以满速的，所有测速节点都满速；而如果4个口全都用起来，就变成了30M-100M左右了。。。
这也就解释了我遇到的问题的解释了。。。

coolsnowwolf, 2020-04-03:
这真是玄学了

dolphinpaopao, 2020-04-04:
可能是驱动或者网卡芯片共用了pcie通道。

Aes64X, 2020-04-04:
能否告知具体主板型号？和网卡型号？存在以下可能：
不同的网卡芯片支持的PCIE标准不同，通常PCIE 2.0 x4 可以跑满4口千兆网卡，我仅知晓如下型号：intel 82574、intel82576、intel i350
但如果一个4口网卡被设计为 PCIE x1插槽，通常两个网口工作时尙没什么问题，4口全部工作时PCIE带宽将被均分。。。  已知的一个4口螃蟹网卡会这样。

a1392429292, 2020-04-04:
> 能否告知具体主板型号？和网卡型号？存在以下可能：
> 不同的网卡芯片支持的PCIE标准不同，通常PCIE 2.0 x4 可以跑满4口千兆网卡，我仅知晓如下型号：intel 82574、intel82576、intel i350、intel i340
> 但如果一个4口网卡被设计为 PCIE x1插槽，通常两个网口工作时尙没什么问题，4口全部工作时PCIE带宽将被均分。。。 已知的一个4口螃蟹网卡会这样。

主板：H310 itx 4网口软路由主板
网卡：i211 4口
这个板子以前是双网口的，后期升级可以四网口了，感觉锅就在这个主板上。。。

ggllioio, 2020-04-04:
> 能否告知具体主板型号？和网卡型号？存在以下可能：
> 不同的网卡芯片支持的PCIE标准不同，通常PCIE 2.0 x4 可以跑满4口千兆网卡，我仅知晓如下型号：intel 82574、intel82576、intel i350、intel i340
> 但如果一个4口网卡被设计为 PCIE x1插槽，通常两个网口工作时尙没什么问题，4口全部工作时PCIE带宽将被均分。。。 已知的一个4口螃蟹网卡会这样。

看起来是泽创伟业的板子，双网口四网口可选：

规格 | ZC-DN-H310QH多网口网安主板参数
-- | --
CPU | 支持Intel第九代、第八代Coffee Lake LGA1151系列CPU，支持赛扬、奔腾、I3，I5，I7，I9等处理器。
芯片组 | INTEL H310
显卡 | Intel UHD核心显卡(取决于CPU型号）
内存 | 2*SO-DIMM，DDR4 Memory，Max.64GB （内存需另行购买）
网卡 | 4*千兆网卡，Intel I211网卡（可选2LAN版本）
音频 | 集成6声道音效ALC662声卡芯片
USB | 6*USB 其中4个标准 USB3.0 ，2个为USB2.0插针
PCIe | 1*PCIe 4X
存储接口 | 2* SATA3.0和1*MSATA
显示 | DP1.2+HDMI1.4 (可订制VGA版本)
外部I/O接口 | 1*DC-IN 电源输入
2*USB3.0
4*RJ45（Intel 1000M LAN）
1*VGA
1*3.5mm Audio（SPK+MIC）
1*开关按键&电源指示灯（前端）
2*USB 3.0(前端）
内部I/O接头 | 1*LGA1151 CPU插座
2*SATA 3.0
1* F-Panel前置面板插针（9pin）
2* USB2.0插针（9pin）
2*RS232串口插针（2*9pin)
1*GPIO（2*6PIN）
1*2PIN（机箱防盗插针）
1*PCIE 4X插槽
1*MSATA
1*硬件通电自启跳帽
2*SO-DIMM内存插槽(DDR4)
1*M.2 2230插槽支持Wifi模块
1*TPM插针
1*SPDIF插针
1*前置音频插针（2*5抽8）
供电 | DC12-DC19V宽电压
板型 | Mini-ITX 17x17cm(不含散热器厚度约20mm）
温度 | 工作：0-60摄氏度        储存：-20—80摄氏度（无冷凝）

板子上没看到桥接芯片，4个i211网卡应该不是一个pci-e x1 拆分的。

ElectronicElephant, 2020-04-09:
爱莫能助，感觉很玄学，不过可以分享一点经验和看法。测速网站是不会用小包测试的，肯定是大包。我其实一向认为，如果没有链路聚合的需求，软路由只要双网口（1 WAN + 1 LAN）就足矣。没有必要把局域网流量也到CPU去绕一圈。交换的事情就让专业的交换机来做。

a1392429292, 2020-04-09:
感谢各位；已经证实了，是主板问题，主板网卡的问题，现在主板研发的说，给主板加几颗滤波电容就可以解决，但是，思考了一下，还是不想要了，就退掉了。。。

closed 2020-04-09 by a1392429292