coolsnowwolf/lede #4149: ssr plus+ 服务端socks5代理是否真能连上？代理验证是摆设？

DongfeiSay, 2020-04-02:
服务端里的socks5代理是强制需要输入用户密码的，问题来了，究竟该怎么连上呢？正常浏览器代理服务器里输入ip+端口肯定是不行的了，装上SwitchyOmega插件能正常使用，这是之前高级设置里的socks5代理不用也不能设置代理认证的使用情况，现在那个socks5代理是全局代理的了，我就只能使用服务端的socks5代理了，SwitchyOmega代理是不支持用户认证的(给我介绍一款支持SwitchyOmega用户认证的浏览器？)，行⑧，我开始认为这怕不是只能tg上用的，tg填入socks5ip用户密码，哎能连上了，可是。。。就算我没输入用户密码也能连上啊，代理验证是摆设？(之前高级设置里的socks能代理验证的时候也是一样)，如果这不是bug那求各位高人指点江山

ElectronicElephant, 2020-04-02:
不勾选不就行了
[image]

socks5 的验证是在第一次连接的时候用明文发送的，所以其实也没什么卵用

DongfeiSay, 2020-04-02:
> 不勾选不就行了
> [image]
> 
> socks5 的验证是在第一次连接的时候用明文发送的，所以其实也没什么卵用

你先更新到最新源码在看看吧，现在这里的socks5是全局代理而且也没有启用代理验证这个选项了，我说的是服务端里的socks5

ElectronicElephant, 2020-04-02:
> 
> 
> > 不勾选不就行了
> > [image]
> > socks5 的验证是在第一次连接的时候用明文发送的，所以其实也没什么卵用
> 
> 你先更新到最新源码在看看吧，现在这里的socks5是全局代理而且也没有启用代理验证这个选项了，我说的是服务端里的socks5，服务端里的socks5偏要输入用户密码才能保存我能怎么办

Okay, 不好意思，是我搞错了。😂

liwenjie119, 2020-04-02:
我用最新源码编译newifi3，socks还在高级设置，是这样的，但是之前就说改位置了，你们的和我的也不一样。。。
[image]

EnnawYang, 2020-04-02:
> 我用最新源码编译newifi3，socks还在高级设置，是这样的，但是之前就说改位置了，你们的和我的也不一样。。。
> [image]

你这个就是最新的

babylife, 2020-04-03:
我的完全正常非问题

DongfeiSay, 2020-04-03:
> 我的完全正常非问题

嗯，那请教，你是怎么连上服务端 服务端类型socks5必须输入用户密码的需代理认证socks5的？浏览器代理服务器输入ip+端口打开个网页直接弹出来要你输入用户密码？装了某插件插件里面输入ip+端口+用户密码连上？

Mattraks, 2020-04-03:
服务端里面的服务是对外网开放的，为了安全强制启用密码。
局域网内使用 **高级设置>SOCKS5 代理服务端（全局）**

coolsnowwolf, 2020-04-03:
解释下，那个对外的 sock5 认证，为了兼容 chrome 这类的不支持认证的浏览器，默认启用了 OTA 一次认证

认证通过后，会自动对这个IP通过，无需再次认证

Zack0617, 2020-04-05:
服务端那个我没有成功开启过socks5，高级选项下面那个正常

closed 2020-04-08 by coolsnowwolf

yanmaiouc, 2020-06-21:
mark

qiaoyun680, 2022-01-04:
> 服务端里的socks5代理是强制需要输入用户密码的，问题来了，究竟该怎么连上呢？正常浏览器代理服务器里输入ip+端口肯定是不行的了，装上SwitchyOmega插件能正常使用，这是之前高级设置里的socks5代理不用也不能设置代理认证的使用情况，现在那个socks5代理是全局代理的了，我就只能使用服务端的socks5代理了，SwitchyOmega代理是不支持用户认证的(给我介绍一款支持SwitchyOmega代理认证的浏览器？)，行⑧，我开始认为这怕不是只能tg上用的，tg填入socks5ip用户密码，哎能连上了，可是。。。就算我没输入用户密码也能连上啊，代理认证是摆设？还使SwitchyOmega也连不上了(之前高级设置里的socks能代理验证的时候也是一样)，如果这不是bug那求各位高人指点江山

请问最后是怎么处理chrome浏览器密码认证的呀…