coolsnowwolf/lede #4150: airplay2的问题

kongzheng333, 2020-04-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：虽然在openwrt控制台能看到音频设备 如下图
[image]
但是接上耳机却无法发声 系统日志显示HW0 not found

### (2) 路由器型号和固件版本

A：X86 4.1 action

### (3) 详细日志

A：

archyc, 2020-04-02:
也是一样的问题，用的usb声卡，也是没声

elonj, 2020-04-02:
手机上可以连上并且播放，但是没有声音。有如下日志：
kern.info kernel: [  419.998574] snd_hda_intel 0000:00:1b.0: IRQ timing workaround is activated for card #1. Suggest a bigger bdl_pos_adj.

qiankan520, 2020-04-03:
我也这样 没有声音 不知道怎么回事

dhlsam, 2020-04-05:
同样的问题，可以识别 但是没有声音
root@OpenWrt:~# aplay -l
 List of PLAYBACK Hardware Devices 
card 1: Pebbles [JBL Pebbles], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

kongzheng333, 2020-04-07:
music center已发布

closed 2020-04-07 by kongzheng333

LXY631, 2021-12-05:
> music center 已发布
大佬你好，看到你用的和我用的同系列声卡（我的usb alc5686声卡），不知现在你问题解决了没有，使用什么方法解决的，当前我编译的固件中包括kmod-usb-audio，手机播放音乐投放过去后无法播放，后台查询提示信息如下：

root@OpenWrt:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Audio [Realtek USB2.0 Audio], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

mbskter, 2024-03-28:
我也遇到了问题...