coolsnowwolf/lede #4152: 有没有人可以指点我一下，为什么在trojan服务器开启了tcp fast open，客户端youtube app无法播放了

norikohxb, 2020-04-02:
有没有人可以指点我一下，为什么在trojan服务器开启了tcp fast open，客户端youtube app无法播放了
我自己实践
以为那个tcp fast open是好事情，毕竟有个fast，所以就特地去把自己的trojan的服务器的server.conf
里面的tcp fast open都由false修改为true
在电脑上面chrome网页打开youtube没有任何问题
就是在安卓手机的youtube的apk，或者是youtube vanced的apk，无一例外全部都无法打开视频

折腾了半天最后才发现是开启了 tcp fast open，有大神可以指导我一下这个是为何会影响 安卓手机的youtube的apk访问视频吗？

Promix953, 2020-04-03:
[https://github.com/trojan-gfw/trojan/issues](url)
针对你三天两头提的issue：
1.哪个算是真正的issue？
2.google下很累吗？
3.试一下很累吗？
4.学一学很累吗？
5.当[https://github.com/coolsnowwolf/lede/issues/2648](url)里面：
```
这种不愿提供足够的信息，截图日志什么都没有，也不愿写一些自己尝试的过程供后人参考。issues 那边，一堆随随便便的一句话问题，却想着别人认认真真的去帮他分析解决的，比比皆是。
```
这样的人很爽吗？

norikohxb, 2020-04-03:
> [https://github.com/trojan-gfw/trojan/issues](url)
> 针对你三天两头提的issue：
> 1.哪个算是真正的issue？
> 2.google下很累吗？
> 3.试一下很累吗？
> 4.学一学很累吗？
> 5.当[https://github.com/coolsnowwolf/lede/issues/2648](url)里面：
> 
> ```
> 这种不愿提供足够的信息，截图日志什么都没有，也不愿写一些自己尝试的过程供后人参考。 
> issues 那边，一堆随随便便的一句话问题，却想着别人认认真真的去帮他分析解决的，比比皆是。
> ```
> 
> 这样的人很爽吗？
这个没有log，老板，这个不是lean大的问题，只是trojan的tcp fastopen，我并不是说是lean大的ssr的问题。所以我提供纪录也没有用，并不是说lean大的ssr有问题，你不需要那么怼我吧。你觉得喜欢可以回答不喜欢就不回答，这个只是我想要问一下有人去折腾tcp fast open的问题。你息怒咯

dolphinpaopao, 2020-04-03:
就事论事，拿issue当论坛版聊的，作者很早前就提醒过了。issue是issue不是论坛，有问题先去论坛QQ群等渠道获取信息，issue一大堆提建议问问题的，所以作者鸟都不鸟了，好自为之。

norikohxb, 2020-04-03:
那就关闭咯，没有问题的。

norikohxb, 2020-04-03:
这个有什么好不爽的，感觉好像影响到你什么一样，

closed 2020-04-03 by norikohxb

Promix953, 2020-04-03:
> 这个有什么好不爽的，感觉好像影响到你什么一样，

一周发的issue比别人几个月的都多，天天拿这里当聊天版，你不只影响到我，你影响了每一个为这个项目做出帮助的人。
之前帮你处理了一个问题，结果不停的问东问西，最后来一句从我这能免费学知识。。。你要真好学就去自己学去，大家帮助的是这个项目，不是你的免费家教！