coolsnowwolf/lede #4155: enhancement： 列表显示服务器地址的需求

babylife, 2020-04-03:
由于使用订阅地址，多个服务器所有备注都一样 ，如下截图，希望能够在选择节点的列表下拉框上节点后面加上实际服务器地址，也就是把真实地址写在下拉列表中后面紧跟显示，这样可以知道实际的服务器地址以便进行切换选择。附上截图对比，后面是passwall的给老大参考下。 希望老大可以采纳
[image]

[image]

Promix953, 2020-04-03:
服务器列表直接点击应用，更直观，还能看延时

iyuangang, 2020-04-03:
[image]

这边感觉更好一点儿

babylife, 2020-04-03:
恩 也是个方法 谢谢

babylife, 2020-04-03:
> 服务器列表直接点击应用，更直观，还能看延时

服务器列表也是一样的描述 不知道哪个节点

Promix953, 2020-04-03:
> > 服务器列表直接点击应用，更直观，还能看延时
> 
> 服务器列表也是一样的描述 不知道哪个节点

忘了服务器列表的确看不到具体服务器地址。。。
但是为什么会出现节点别名和延迟没有参考意义，需要通过具体地址选择节点的情况？

coolsnowwolf, 2020-04-03:
[image]

其实挺难看的

babylife, 2020-04-04:
> 
> 
> [image]
> 
> 
> 
> 
> 
> 其实挺难看的。。

老大 其实是有点难看，不过是不是加个复选框，显示服务器地址？服务器列表那里也是 别名一列全部一样哈哈

qianxu2001, 2020-04-04:
建议换机场

babylife, 2020-04-07:
> 建议换机场

换机场代价有点大

Promix953, 2020-04-09:
> > 建议换机场
> 
> 换机场代价有点大

所以还是那个问题：为什么节点名字和延迟不能作为选择依据，一定要看机场实际地址？

babylife, 2020-04-10:
> > > 建议换机场
> > 
> > 
> > 换机场代价有点大
> 
> 所以还是那个问题：为什么节点名字和延迟不能作为选择依据，一定要看机场实际地址？

老大说的也不无道理，

babylife, 2020-04-14:
机场已更新解决，谢谢！

closed 2020-04-14 by babylife