coolsnowwolf/lede #4156: ShadowSocksR Plus +广告屏蔽功能可能会导致无法加载部分网站

wsndshx, 2020-04-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在开启了SSR Plus+的广告屏蔽功能后,今日头条文章流无法加载.具体表现为域名`sf1-ttcdn-tos.pstatp.com`无法解析,同时小米AD浏览器中部分文章域名也无法解析.
[image]

### (2) 路由器型号和固件版本

A：Newifi-D2 - OpenWrt R20.3.19

## 疑问
A：开了广告拦截这些被拦截下来也能理解,但疑惑的是为何会走代理这边,因为我同时还开启了adbyby plus+,但关闭adbyby plus+后问题依旧,只有关闭ssr plus+中的广告拦截才能恢复访问.是这些域名被加入到GWF列表中了还是ssr plus+的广告拦截功能是和adbyby plus+的作用范围是重复的?
(之前理解的是adbyby plus+负责直连流量的广告,而无法处理代理流量中的广告;ssr plus+则处理经过代理的流量中的广告)

wsndshx, 2020-04-03:
## 补充
adbyby plus+的过滤模式为`Plus + 模式`, XXR Plus+ 的代理模式为GFW模式.
[image]
[image]

shikiroot, 2020-04-03:
adbyby和ssr的ad block功能和run mode(gfw, ip router. etc.)无关。所有的屏蔽都是基于规则的。发现误屏蔽自己改规则或者去相关规则的repo提issue。

babylife, 2020-04-03:
去广告建议用KP，专业的

wsndshx, 2020-04-03:
> adbyby和ssr的ad block功能和运行模式（gfw，ip路由器等）无关。所有的屏蔽都是基于规则的。发现错误屏蔽自己改规则或者去相关规则的repo提问题。

emmmm好的蟹蟹.

closed 2020-04-03 by wsndshx