coolsnowwolf/lede #4161: 编译的K2P总是无缘无故断网，接口显示IP

segadrive, 2020-04-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
无缘无故无网络
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本
K2P
A：

### (3) 详细日志
Fri Apr  3 20:56:50 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 20:56:50 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.115.115" on interface wan (pppoe-wan)
Fri Apr  3 20:56:50 2020 user.info mwan3track[2928]: Lost 1 ping(s) on interface wan (pppoe-wan)
Fri Apr  3 20:56:55 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 20:56:55 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.115.115" on interface wan (pppoe-wan)
Fri Apr  3 20:57:00 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 20:57:00 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.115.115" on interface wan (pppoe-wan)
Fri Apr  3 20:57:05 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 20:57:05 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.115.115" on interface wan (pppoe-wan)
Fri Apr  3 20:57:10 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 20:57:10 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.115.115" on interface wan (pppoe-wan)
Fri Apr  3 20:57:17 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.115.115" on interface wan (pppoe-wan)
Fri Apr  3 20:57:24 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.115.115" on interface wan (pppoe-wan)
Fri Apr  3 20:57:26 2020 cron.info crond[6452]: crond (busybox 1.30.0) started, log level 5
Fri Apr  3 20:57:39 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 20:57:39 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.115.115" on interface wan (pppoe-wan)
Fri Apr  3 20:57:39 2020 user.info mwan3track[2928]: Lost 4 ping(s) on interface wan (pppoe-wan)
Fri Apr  3 20:57:42 2020 cron.info crond[6639]: crond (busybox 1.30.0) started, log level 5
Fri Apr  3 20:57:44 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 20:57:44 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.115.115" on interface wan (pppoe-wan)
Fri Apr  3 20:57:49 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 20:57:49 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.115.115" on interface wan (pppoe-wan)
Fri Apr  3 20:57:50 2020 cron.info crond[6777]: crond (busybox 1.30.0) started, log level 5
Fri Apr  3 20:57:54 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 20:57:54 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.115.115" on interface wan (pppoe-wan)
Fri Apr  3 20:57:54 2020 cron.info crond[6955]: crond (busybox 1.30.0) started, log level 5
Fri Apr  3 20:57:59 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 20:58:06 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 20:58:06 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.115.115" on interface wan (pppoe-wan)
Fri Apr  3 20:58:06 2020 user.info mwan3track[2928]: Lost 1 ping(s) on interface wan (pppoe-wan)
Fri Apr  3 20:58:11 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 20:58:11 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.115.115" on interface wan (pppoe-wan)
Fri Apr  3 20:58:19 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.115.115" on interface wan (pppoe-wan)
Fri Apr  3 20:58:24 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 20:58:24 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.115.115" on interface wan (pppoe-wan)
Fri Apr  3 20:58:24 2020 user.info mwan3track[2928]: Lost 1 ping(s) on interface wan (pppoe-wan)
Fri Apr  3 20:58:29 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 20:58:29 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.115.115" on interface wan (pppoe-wan)
Fri Apr  3 20:58:34 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 20:58:34 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.115.115" on interface wan (pppoe-wan)
Fri Apr  3 20:58:55 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 20:58:55 2020 daemon.err uhttpd[1359]: sh: iwconfig: not found
Fri Apr  3 20:58:56 2020 daemon.err uhttpd[1359]: sh: iwconfig: not found
Fri Apr  3 20:58:57 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.115.115" on interface wan (pppoe-wan)
Fri Apr  3 20:58:58 2020 daemon.notice netifd: Network device 'rax0' link is up
Fri Apr  3 20:58:58 2020 daemon.notice netifd: Network device 'apclix0' link is up
Fri Apr  3 20:58:58 2020 daemon.err uhttpd[1359]: sh: iwconfig: not found
Fri Apr  3 20:58:58 2020 daemon.err uhttpd[1359]: sh: iwconfig: not found
Fri Apr  3 20:59:12 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 20:59:12 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.115.115" on interface wan (pppoe-wan)
Fri Apr  3 20:59:12 2020 user.info mwan3track[2928]: Lost 2 ping(s) on interface wan (pppoe-wan)
Fri Apr  3 20:59:17 2020 user.info mwan3track[2928]: Check (ping) success for target "114.114.114.114" on interface wan (pppoe-wan)
[... 150 lines cut ...]
Fri Apr  3 21:04:07 2020 user.info mwan3[10045]: connection tracking not flushed on interface wan (pppoe-wan) ifup
Fri Apr  3 21:04:07 2020 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Fri Apr  3 21:04:09 2020 cron.info crond[10994]: crond (busybox 1.30.0) started, log level 5
Fri Apr  3 21:04:09 2020 user.info mwan3track[10701]: Check (ping) failed for target "114.114.115.115" on interface wan (pppoe-wan)
Fri Apr  3 21:04:10 2020 cron.info crond[11108]: crond (busybox 1.30.0) started, log level 5
Fri Apr  3 21:04:13 2020 user.notice ddns-scripts[11289]: myddns_ipv4: PID '11289' started at 2020-04-03 21:04
Fri Apr  3 21:04:16 2020 user.warn ddns-scripts[11289]: myddns_ipv4: Service section disabled! - TERMINATE
Fri Apr  3 21:04:16 2020 user.info mwan3track[10701]: Check (ping) failed for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 21:04:16 2020 user.warn ddns-scripts[11289]: myddns_ipv4: PID '11289' exit WITH ERROR '1' at 2020-04-03 21:04
Fri Apr  3 21:04:21 2020 user.info mwan3track[10701]: Lost 2 ping(s) on interface wan (pppoe-wan)
Fri Apr  3 21:04:22 2020 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Fri Apr  3 21:04:24 2020 cron.info crond[12705]: crond (busybox 1.30.0) started, log level 5
Fri Apr  3 21:04:25 2020 cron.info crond[12822]: crond (busybox 1.30.0) started, log level 5
Fri Apr  3 21:04:40 2020 cron.info crond[13222]: crond (busybox 1.30.0) started, log level 5
Fri Apr  3 21:04:44 2020 user.info mwan3track[10701]: Check (ping) failed for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 21:04:51 2020 user.info mwan3track[10701]: Check (ping) failed for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 21:04:56 2020 user.info mwan3track[10701]: Lost 2 ping(s) on interface wan (pppoe-wan)
Fri Apr  3 21:05:08 2020 user.info mwan3track[10701]: Check (ping) failed for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 21:05:10 2020 user.info mwan3track[10701]: Check (ping) failed for target "114.114.115.115" on interface wan (pppoe-wan)
[... cut ...] heck (ping) failed for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 21:05:22 2020 user.info mwan3track[10701]: Lost 3 ping(s) on interface wan (pppoe-wan)
Fri Apr  3 21:05:50 2020 user.info mwan3track[10701]: Check (ping) failed for target "114.114.114.114" on interface wan (pppoe-wan)
Fri Apr  3 21:05:55 2020 user.info mwan3track[10701]: Lost 1 ping(s) on interface wan (pppoe-wan)
Fri Apr  3 21:06:08 2020 cron.info crond[13525]: crond (busybox 1.30.0) started, log level 5
A：

coolsnowwolf, 2020-04-03:
明显是要删除 mwan3 嘛

segadrive, 2020-04-03:
> 明显是要删除 mwan3 嘛

好的，谢谢大佬

closed 2020-04-08 by coolsnowwolf