coolsnowwolf/lede #4167: 编译树莓派4b出现如下编译错误

LIGHT1213, 2020-04-04:
## 1.关于你要提交的问题
编译树莓派4b出现如下编译错误
Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
error2错误，之前有成功编译过，这次更新后二次编译就有问题了
A：

### (2) 路由器型号和固件版本
树莓派4b
最新的源码

### (3) 详细日志
( cd /home/ubuntu/lede/build_dir/hostpkg/go-1.14/src ; GOROOT_FINAL=/home/ubuntu/lede/staging_dir/hostpkg/lib/go-cross  GOROOT_BOOTSTRAP=/home/ubuntu/lede/build_dir/hostpkg/go-1.14/.go_bootstrap GOCACHE=/home/ubuntu/lede/build_dir/hostpkg/go-1.14/.go_work/cache CC=gcc CXX=g++  bash make.bash --no-banner ; )
Building Go cmd/dist using /home/ubuntu/lede/build_dir/hostpkg/go-1.14/.go_bootstrap. (go1.4-bootstrap-20170531 linux/amd64)
Building Go toolchain1 using /home/ubuntu/lede/build_dir/hostpkg/go-1.14/.go_bootstrap.
go build bootstrap/cmd/compile/internal/ssa: /home/ubuntu/lede/build_dir/hostpkg/go-1.14/.go_bootstrap/pkg/tool/linux_amd64/6g: signal: killed
go tool dist: FAILED: /home/ubuntu/lede/build_dir/hostpkg/go-1.14/.go_bootstrap/bin/go install -gcflags=-l -tags=math_big_pure_go compiler_bootstrap bootstrap/cmd/...: exit status 1
Makefile:327: recipe for target '/home/ubuntu/lede/build_dir/hostpkg/go-1.14/.built' failed
make[3]: *** [/home/ubuntu/lede/build_dir/hostpkg/go-1.14/.built] Error 2
make[3]: Leaving directory '/home/ubuntu/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#57.80#8.82#72.90
package/Makefile:111: recipe for target 'package/feeds/packages/golang/host/compile' failed
make[2]: *** [package/feeds/packages/golang/host/compile] Error 2
make[2]: Leaving directory '/home/ubuntu/lede'
package/Makefile:107: recipe for target '/home/ubuntu/lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile' failed
make[1]: *** [/home/ubuntu/lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
/home/ubuntu/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

LIGHT1213, 2020-04-04:
此次编译的全部日志已经push到https://github.com/LIGHT1213/txt

lphgor, 2020-04-04:
内存不够，进程被杀了
`go build bootstrap/cmd/compile/internal/ssa: /home/ubuntu/lede/build_dir/hostpkg/go-1.14/.go_bootstrap/pkg/tool/linux_amd64/6g: signal: killed`

LIGHT1213, 2020-04-04:
> 内存不够，进程被杀了
> `go build bootstrap/cmd/compile/internal/ssa: /home/ubuntu/lede/build_dir/hostpkg/go-1.14/.go_bootstrap/pkg/tool/linux_amd64/6g: signal: killed`

可能欸，这次的多加了模块，我先试试4g内存的ecs

closed 2020-07-04 by coolsnowwolf