coolsnowwolf/lede #4169: 路由器翻墙Google Play服务不稳定分析

z826540272, 2020-04-04:
使用SSR+翻墙经常发现谷歌商店不稳定，无法下载，无法刷新，无法显示图像等
经过调查，无法显示图像的域名是
lh3.googleusercontent.com
这时候我在手机端 https ping一下这个域名，图像显示即可恢复

这种间歇性的不稳定非线路问题，只要使用谷歌服务的用户是很容易出现的，目前未发现叫好的解决办法

GWONGJYULEI, 2020-04-04:
使用过程种，暂无发现上述问题

lizhenglin2002, 2020-04-08:
我也有这情况，经常性play服务这些下载不了应用，而在电脑和Apple这边没问题

waynon, 2020-04-08:
是的，我这里无法在谷歌play商店中下载应用，会一直显示“正在等待下载”

lizhenglin2002, 2020-04-08:
你现在解决了吗🥶

waynon, 2020-04-08:
> […](#)

时好时坏，现在又好了。
我有以下几种推测：
1. IPv6冲突。如果你同时有IPv4和IPv6地址，而只代理了IPv4地址，Google Play Store下载时会首先尝使用IPv6地址，但是其实你的IPv6是被墙了的，需要等待几分钟直至IPv6下载尝试超时后Google Play Store才会使用你的IPv4地址进行下载。
2. UDP未代理。谷歌部分使用QUIC协议，而QUIC协议是基于UDP的，如果UDP未代理结果可想而知。
3. ​在网关使用透明代理导致的副作用：服务器期望收到的https请求是：CONNECT $host ... ，然后才是 ssl 握手，而通过iptables 规则转发过去的请求，直接就是ssl 握手。参考 https://github.com/XX-net/XX-Net/issues/8516
4. 尝试更新一下GFW列表和国内IP

lizhenglin2002, 2020-04-08:
>    握手，而通过iptables 规则转发过去的请求，直接就是ssl 握手。参考 XX-net/XX-Net#8516

对，我是把禁止解析ipv6 dns开着，然后Google Play要等两分钟这样才能下载应用，可是我把这个关了也是这样的情况

Aaron-lv, 2020-04-08:
是否为小米或者华为手机，这两款手机会导致你说的问题，可刷第三方或国际版系统解决

lizhenglin2002, 2020-04-08:
是小米，是国际版的

Aaron-lv, 2020-04-08:
> […](#)

MIUI国际官网是否有官方国际版适配，如果没有刷所谓的欧洲版也是不行的，如果没有官方国际版适配那欧洲版的固件就是用国内固件修改的。    所以这个导致Google服务不稳定的问题依旧没解决

waynon, 2020-04-09:
> > […](#)
> 
> MIUI国际官网是否有官方国际版适配，如果没有刷所谓的欧洲版也是不行的，如果没有官方国际版适配那欧洲版的固件就是用国内固件修改的。 所以这个导致Google服务不稳定的问题依旧没解决

我的手机刷写了国外的第三方固件(Pixel Experience, LineageOS, etc.)以及一加手机的国外版固件(OxygenOS)也会出现这样的情况，这种情况 (谷歌Play商店下载应用一直显示"等待下载中...“) 集中出现在路由器刚刚刷机完成或重置路由器设置的一段时间，即使重启路由器仍然出现，但是过一两天这个问题就消失了。

而且手机直接使用VPN不会出现这个问题，因此我不认为是手机系统导致的，而应该是OpenWRT这边的问题

Aaron-lv, 2020-04-09:
> > > […](#) 
> > 
> > 
> > MIUI国际官网是否有官方国际版适应，如果没有刷所谓的欧洲版本也是不行的，如果没有官方国际版适应那欧洲版的固件就是用国内固件修改的。依旧没解决
> 
> 我的手机刷写了国外的第三方固件（Pixel Experience，LineageOS等）以及一加手机的国外版固件（OxygenOS）也会出现这样的情况
> 
> 这种情况（谷歌Play商店下载应用一直显示“等待下载中...”）集中出现在路由器刚刚刷机完成或重置路由器设置的重置，重新启动路由器仍然出现，但是过一两天这个问题就消失了。
> 
> 而且手机直接使用VPN不会出现这个问题，因此我不认为是手机系统导致的，而应该是OpenWRT这边的问题

自行判断吧，反正我刷了魔趣OS是已经好了，版主的问题也不仅仅是等待下载，他是图片无法显示，估计看YouTube视频也会卡住或显示无网络连接，而且iPhone和PC都没问题，所以不是OpenWRT的问题。

lizhenglin2002, 2020-04-09:
你说的这个情况我都有，现在还是挂vpn算了

Aaron-lv, 2020-04-09:
> […](#)

小米手机和华为手机还是刷第三方系统吧

fonlan, 2020-04-09:
有时候是手机DNS没刷新，且手机流量和宽带不同运营商，比如用流量且没开VPN的时候手机后台尝试过访问Google，那么可能就会解析到一个错误的IP地址上，由于DNS是有缓存的，在缓存失效前就会使用这个错误的IP尝试连接，自然是连不上的。
所以可以尝试清空手机缓存再试，我所知的方法试彻底结束Google Play，然后开关一次飞行模式，再开Google Play试一下？

Aaron-lv, 2020-04-09:
> 有时候是手机DNS没刷新，且手机流量和宽带不同运营商，比如用流量且没开VPN的时候手机后台尝试过访问Google，那么可能就会解析到一个错误的IP地址上，由于DNS是有缓存的，在缓存失效前就会使用这个错误的IP尝试连接，自然是连不上的。
> 所以可以尝试清空手机缓存再试，我所知的方法试彻底结束Google Play，然后开关一次飞行模式，再开Google Play试一下？

如果是其它手机系统你说的方法确实可以解决，但是小米和华为自带污染的，飞行多少次都是不可以的，只能手机自己开VPN，强制流量走代理通道才能解决，不然就刷第三方系统。

fonlan, 2020-04-09:
> > 有时候是手机DNS没刷新，且手机流量和宽带不同运营商，比如用流量且没开VPN的时候手机后台尝试过访问Google，那么可能就会解析到一个错误的IP地址上，由于DNS是有缓存的，在缓存失效前就会使用这个错误的IP尝试连接，自然是连不上的。
> > 所以可以尝试清空手机缓存再试，我所知的方法试彻底结束Google Play，然后开关一次飞行模式，再开Google Play试一下？
> 
> 如果是其它手机系统你说的方法确实可以解决，但是小米和华为自带污染的，飞行多少次都是不可以的，只能手机自己开VPN，强制流量走代理通道才能解决，不然就刷第三方系统。

不用啊，我就是华为，路由器上SSR Plus+把googleapis.cn加域名黑名单就好了

lizhenglin2002, 2020-04-09:
你也有这个问题？

fonlan, 2020-04-09:
> […](#)

偶尔会出现，飞行模式开关一下就好了，极少数需要重启手机

lizhenglin2002, 2020-04-09:
为啥你说的这个googleapis.cn ping了没反应

Aaron-lv, 2020-04-09:
> 不用啊，我就是华为，路由器上SSR Plus+把googleapis.cn加域名黑名单就好了

你说的这个域名已经默认走代理了，在GFWlist中，反正小米手机是不行的，不是谷歌服务的问题，我刷了第三方系统其它没有任何变动就一切正常了。

fonlan, 2020-04-09:
> […](#)

域名解析正常就行，为啥不能ping你得去问谷歌

fonlan, 2020-04-09:
> 你说的这个域名已经默认走代理了，在GFWlist中，反正小米手机是不行的，不是谷歌服务的问题，我刷了第三方系统其它没有任何变动就一切正常了。

各人情况不一样吧，不过建议还是大陆白名单，GFW模式确实会有一些奇怪的问题

lizhenglin2002, 2020-04-09:
是用大陆白名单啊😭

z826540272, 2020-04-09:
显然不是，我三星 miui  原生都这样

z826540272, 2020-04-09:
不是ipv6的问题，我压根没用ipv6,手机也没v6地址，在手机上使用v2ray或者ssr play商店非常稳定

z826540272, 2020-04-09:
我开启了udp代理问题依然出现
不通的时候，只要手机上https ping 一下相关域名马上就通了

z826540272, 2020-04-09:
你的推理是错误的，dns解析是谷歌地址，只是就是不通超时，这时候https ping一下，整个系统恢复通畅。

z826540272, 2020-04-09:
不实信息。

z826540272, 2020-04-09:
开关一下wifi就解决了，但是时长出现很烦人

z826540272, 2020-04-09:
大陆白名单也写问题，唯一就全局没试过了

lizhenglin2002, 2020-04-09:
不想弄了，今天把Openwrt重刷了。现在还算稳定

fonlan, 2020-04-09:
> 不实信息。
> […](#)

得，当我好心喂驴了，好心给你提供个可能的思路你还不实信息了，你NB你自己慢慢折腾吧

z826540272, 2020-04-09:
不仅谷歌不通，手机待机一段时间后重连telegram也不通了，开关wifi才能解决

z826540272, 2020-04-09:
> […](#)

我开全局也不稳，准备换passwall试试了

jikkyfu, 2020-04-29:
@z826540272 你最终怎么解决的？

z826540272, 2020-04-29:
passwall tcp开全部端口可以缓解

lizhenglin2002, 2020-04-29:
SSR+应该怎样解决

z826540272, 2021-02-17:
> […](#)

初步定位到原因,请问你当时是否使用网关模式,非路由器直接拨号.

z826540272, 2021-02-17:
> @z826540272 你最终怎么解决的？

初步定位到原因,请问你当时是否使用网关模式,非路由器直接拨号

ChingLin123, 2021-02-17:
> > […](#)
> 
> 初步定位到原因,请问你当时是否使用网关模式,非路由器直接拨号.

是的

z826540272, 2021-02-17:
> 我也有这情况，经常性play服务这些下载不了应用，而在电脑和Apple这边没问题

现在发现大概原因了,当图片不加载的时候,手机端traceroute发现,连接跳过网关,数据直接走向内网光猫.
可能是Android或者linux内核的bug.
我使用的是网关转发模式,非路由器拨号,
请问你们遇到这种故障的是不是也是使用网关模式?

z826540272, 2021-02-17:
> > > […](#)
> > 
> > 
> > 初步定位到原因,请问你当时是否使用网关模式,非路由器直接拨号.
> 
> 是的

你的是什么手机,Android版本和Linux内核版本分别是多少,
目前猜测解决办法是不要用网关模式,否则会触发这个不稳定的bug

ChingLin123, 2021-02-17:
> > > > […](#)
> > > 
> > > 
> > > 初步定位到原因,请问你当时是否使用网关模式,非路由器直接拨号.
> > 
> > 
> > 是的
> 
> 你的是什么手机,Android版本和Linux内核版本分别是多少,
> 目前猜测解决办法是不要用网关模式,否则会触发这个不稳定的bug

之前用的小米手机国际版ROM,安卓7.0.直接指向旁路网关经常上不了外网，现在刷了其他Rom，已经没问题了

z826540272, 2021-02-17:
> 是的，我这里无法在谷歌play商店中下载应用，会一直显示“正在等待下载”

你用的是不是网关模式

muzi9527, 2021-03-04:
> 不用啊，我就是华为，路由器上SSR Plus+把googleapis.cn加域名黑名单就好了

牛逼，解决了

fhbyljj, 2021-05-26:
> 大陆白名单也写问题，唯一就全局没试过了
> […](#)

压根就不关ROM事，我华为和小米去澳门，Play都能正常下载，你试试用passwall吧，TCP我设置绕过大陆IP UDP我设置GFW列表

lxhill, 2021-06-17:
如果是小米手机，只要把google play服务的省电策略的智能省电关闭，就正常了

closed 2021-06-20 by coolsnowwolf

wnarutou, 2022-06-05:
> 如果是小米手机，只要把google play服务的省电策略的智能省电关闭，就正常了

赞！！！小米确实如此，是什么原理

jiangjiangflies, 2024-10-27:
> > […](#)
> 
> 我现在时好时坏，现在又好了。 我有以下几种推测：
> 
> 1. **IPv6冲突**。如果你同时有IPv4和IPv6地址，而只代理了IPv4地址，Google Play Store下载时会首先尝使用IPv6地址，但是其实你的IPv6是被墙了的，需要等待几分钟直至IPv6下载尝试超时后Google Play Store才会使用你的IPv4地址进行下载。
> 2. **UDP未代理**。谷歌部分使用QUIC协议，而QUIC协议是基于UDP的，如果UDP未代理结果可想而知。
> 3. ​**在网关使用透明代理导致的副作用**：服务器期望收到的https请求是：CONNECT $host ... ，然后才是 ssl 握手，而通过iptables 规则转发过去的请求，直接就是ssl 握手。参考 [teredo 配合 DNAT透明代理上网，https 出现ssl 连接出错 XX-net/XX-Net#8516](https://github.com/XX-net/XX-Net/issues/8516)
> 4. 尝试更新一下GFW列表和国内IP

2024，终于找到了问题的答案，就是第一个推测！我把路由器（拨号上网）和网关的ipv6统统禁用掉之后play store下载就正常了，包括一些其他地方应用类似问题也不复存在了！现在感觉透明代理无比顺畅！