coolsnowwolf/lede #4174: RP-N53刷了此固件后，5g Wifi不能使用

tjlive, 2020-04-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
RP-N53刷了此固件后，5g Wifi不能使用。
* [X] 没有类似的issue

## 2. 详细叙述
RP-N53刷了此固件后，5g Wifi不能使用，Reset按件需要定制修改源码。

### (2) 路由器型号和固件版本
路由器，华硕RP-N53；
架构：MediaTek MT7620A ver:2 eco:3 x 1
固件：OpenWrt R20.3.19 / LuCI Master (git-20.087.30210-56fe7ae)

### (3) 详细日志

hcwang, 2020-04-05:
小米R3P也有类似问题，应该是新的MAC80211模块升级导致的。在核心日志中有如下信息：

```
[10031.886744] ------------[ cut here ]------------
[10031.891549] WARNING: CPU: 0 PID: 0 at target-mipsel_24kc_musl/linux-ramips_mt7621/backports-5.4.27-1/net/mac80211/sta_info.c:1937 ieee80211_sta_update_pending_airtime+0x104/0x204 [mac80211]
[10031.908437] STA 80:5e:4f:de:36:b3 AC 0 txq pending airtime underflow: 4294967032, 264
[10031.908447] Modules linked in: xt_FULLCONENAT pppoe ppp_async pppox ppp_mppe ppp_generic mt7615e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_socket xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_bpf xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm slhc nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv4 nf_conntrack_rtcache
[10031.986852]  nf_conntrack_netlink macvlan iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat fuse tcp_bbr ledtrig_usbport cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6 nf_nat nf_conntrack nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 sit tunnel4 ip_tunnel tun vfat fat ntfs hfsplus hfs nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb authenc usb_storage
[10032.058287]  leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd sd_mod scsi_mod gpio_button_hotplug f2fs ext4 mbcache jbd2 usbcore nls_base usb_common crc32c_generic crc32_generic
[10032.074291] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.14.172 #0
[10032.080355] Stack : 00000000 00000200 60000211 8006c638 805b0000 80557c20 00000000 00000000
[10032.088688]         80522578 8fc09c5c 80590d3c 805908c7 8051d3b0 00000001 8fc09c00 1cc2824c
[10032.097020]         00000000 00000000 806f0000 00007c58 00000000 0000018b 00000008 00000000
[10032.105350]         00000000 80590000 00012233 00000000 00000000 805b0000 00000000 8f3077d8
[10032.113680]         00000009 00000791 8de54480 00000200 00000000 802a4898 00000000 806f0000
[10032.122010]         ...
[10032.124445] Call Trace:
[10032.126907] [<8000c6f0>] show_stack+0x58/0x100
[10032.131363] [<8045b484>] dump_stack+0xa4/0xe0
[10032.135705] [<8002aec8>] __warn+0xe0/0x114
[10032.139781] [<8002af2c>] warn_slowpath_fmt+0x30/0x3c
[10032.144848] [<8f3077d8>] ieee80211_sta_update_pending_airtime+0x104/0x204 [mac80211]
[10032.152609] [<8f303458>] ieee80211_tx_monitor+0xf50/0x11b0 [mac80211]
[10032.159152] ---[ end trace debe7b9a376c04ce ]---
```

AmadeusGhost, 2020-04-05:
mt76请去官方仓库提issue
https://github.com/openwrt/mt76

tjlive, 2020-04-10:
看来驱动的问题没人关注，不知道应从哪儿引用解决。。。

closed 2020-07-04 by coolsnowwolf