coolsnowwolf/lede #4177: 安装hd-idle 和相关的luci应用后，OP web界面显示不全并且大部分功能不可用， 原生OP固件无此问题

lixshnew, 2020-04-04:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
在软件包中安装 hd-idle后， OP web页面出现问题，很多页面不可显示，或显示有问题，基本不可用。原生OP固件安装hd-idle正常

### (2) 路由器型号和固件版本
J1900  软路由 （在虚拟机中测试OP，有相同的问题）
OpenWrt R20.3.19 / LuCI Master (git-20.088.29068-cebe120)

对比的原生固件是
OpenWrt 19.07.2 r10947-65030d81f3 / LuCI openwrt-19.07 branch git-20.057.55219-13dd17f

### (3) 详细日志

这是在接口配置时的错误
------------------------------------------------
接口
500 内部服务器错误
对不起，服务器遇到未知错误。
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute arcombine dispatcher target for entry '/admin/network/network'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:55: Failed to execute template 'cbi/map'.
A runtime error occured: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'admin_network/iface_overview'.
A runtime error occured: /usr/lib/lua/luci/model/network.lua:349: attempt to call field 'checkmac' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function

Promix953, 2020-04-04:
肯定100%复现啊，官方源依赖不同，装了luci-compat
不要用自带源安装软件，不兼容，容易出问题，需要的功能直接编译进去，或者单独编译hd-idle

lixshnew, 2020-04-05:
重新编译hd-idle问题解决。请问是否有文档说明如何单独编译一个app? 比如hd-idle.
请close此问题。谢谢。

closed 2020-04-06 by coolsnowwolf