coolsnowwolf/lede #4178: SSR+ 新添加trojan服务器节点时未填写完整信息放弃退出后，页面显示错误

eskimokk, 2020-04-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：SSR+ 新添加trojan服务器节点未填写完整信息放弃退出后，页面显示错误

### (2) 路由器型号和固件版本

A：斐讯K3， OpenWrt R20.3.11 / LuCI Master (git-20.062.50551-a6189c5)

### (3) 详细日志

A： 服务器节点最下端显示错误信息，无法恢复

500 内部服务器错误
对不起，服务器遇到未知错误。

/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute arcombine dispatcher target for entry '/admin/services/shadowsocksr/servers'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:55: Failed to execute template 'cbi/map'.
A runtime error occured: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'cbi/tblsection'.
A runtime error occured: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'cbi/dvalue'.
A runtime error occured: /usr/lib/lua/luci/model/cbi/shadowsocksr/servers.lua:95: bad argument #1 to 'upper' (string expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function

eskimokk, 2020-04-05:
编辑配置文件 /etc/config/shadowsocksr ， 去掉最后一个配置不全的内容后，问题解决。

显然，应该解决：
1. 添加服务器节点时，不点击保存的情况下，不应该存入配置文件
2. 服务器节点信息完整性，应该检查再放入配置文件
3. load配置文件时，不完整，应该有错误处理机制，直接assert 是不对的

coolsnowwolf, 2020-04-08:
你版本太老了

closed 2020-04-08 by coolsnowwolf