coolsnowwolf/lede #4179: WNDR3800在Ubuntu18和Actions上均编译失败

ZForest000, 2020-04-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败

### (2) 路由器型号和固件版本

A：WNDR3800

### (3) 详细日志

A：
```
make[8]: *** [/home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/backports-5.4.27-1/compat] Error 2
scripts/Makefile.build:555: recipe for target '/home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/backports-5.4.27-1/compat' failed
make[7]: *** [_module_/home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/backports-5.4.27-1] Error 2
Makefile:1534: recipe for target '_module_/home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/backports-5.4.27-1' failed
make[6]: *** [modules] Error 2
Makefile.build:13: recipe for target 'modules' failed
make[5]: *** [modules] Error 2
Makefile.real:93: recipe for target 'modules' failed
make[4]: *** [modules] Error 2
Makefile:120: recipe for target 'modules' failed
make[3]: *** [/home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/backports-5.4.27-1/.built] Error 2
make[4]: Leaving directory '/home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/backports-5.4.27-1'
Makefile:538: recipe for target '/home/runner/work/lede/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/backports-5.4.27-1/.built' failed
make[3]: Leaving directory '/home/runner/work/lede/lede/package/kernel/mac80211'
time: package/kernel/mac80211/compile#0.87#0.37#1.19
make[2]: *** [package/kernel/mac80211/compile] Error 2
package/Makefile:111: recipe for target 'package/kernel/mac80211/compile' failed
make[1]: *** [/home/runner/work/lede/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[2]: Leaving directory '/home/runner/work/lede/lede'
package/Makefile:107: recipe for target '/home/runner/work/lede/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/home/runner/work/lede/lede'
/home/runner/work/lede/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
##[error]Process completed with exit code 2.
```

lanzengquan, 2020-04-05:
跑了一晚上  和你的错误代码一样

Aes64X, 2020-04-05:
https://github.com/coolsnowwolf/lede/pull/4176

这个pr应该是修复的，尚未合并，各位可以先试试

AmadeusGhost, 2020-04-05:
> 这个pr应该是修复的，尚未合并，各位可以先试试

这个跟mips没有关系
出错的可以这样试试：
```
make clean
rm .config
```

Aes64X, 2020-04-05:
> > 这个pr应该是修复的，尚未合并，各位可以先试试
> 
> 这个跟ar71xx没有关系
> 出错的可以这样试试：
> 
> ```
> make clean
> rm .config
> ```

嗯啊。。。。看错了，是BCM的

ybist, 2020-04-05:
俺也一样

`Building backport-include/backport/autoconf.h ... done.
  CC [M]  /mnt/sdb/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/backports-5.4.27-1/compat/main.o
In file included from ./include/linux/kmod.h:22:0,
                 from ./include/linux/module.h:13,
                 from /mnt/sdb/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/backports-5.4.27-1/backport-include/linux/module.h:3,
                 from /mnt/sdb/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/backports-5.4.27-1/compat/main.c:1:
/mnt/sdb/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/backports-5.4.27-1/backport-include/linux/gfp.h: In function 'backport_page_frag_alloc':
/mnt/sdb/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/backports-5.4.27-1/backport-include/linux/gfp.h:18:9: error: implicit declaration of function '__alloc_page_frag'; did you mean '__alloc_pages_node'? [-Werror=implicit-function-declaration]
  return __alloc_page_frag(nc, fragsz, gfp_mask);
         ^~~~~~~~~~~~~~~~~
         __alloc_pages_node
/mnt/sdb/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/backports-5.4.27-1/backport-include/linux/gfp.h:18:9: warning: return makes pointer from integer without a cast [-Wint-conversion]
  return __alloc_page_frag(nc, fragsz, gfp_mask);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:304: recipe for target '/mnt/sdb/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/backports-5.4.27-1/compat/main.o' failed
make[9]: *** [/mnt/sdb/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/backports-5.4.27-1/compat/main.o] Error 1
`

ZForest000, 2020-04-05:
> > 这个pr应该是修复的，尚未合并，各位可以先试试
> 
> 这个跟ar71xx没有关系
> 出错的可以这样试试：
> 
> ```
> make clean
> rm .config
> ```

我rm -rf lede 之后，重新拉取源代码使用WNDR3800的默认配置编译，还是会报一样的错误

AmadeusGhost, 2020-04-05:
找到问题了。先测试下再说

ZForest000, 2020-04-05:
> 找到问题了，先测试下再说

是和这个mwlwifi: Update the 88W8964's firmware to 9.3.2.12 and fix backports …这个提交有关吗，我在我的分支上回退了这个提交，正在编译。

AmadeusGhost, 2020-04-05:
> > 找到问题了，先测试下再说
> 
> 是和这个mwlwifi: Update the 88W8964's firmware to 9.3.2.12 and fix backports …这个提交有关吗，我在我的分支上回退了这个提交，正在编译。

不是

AmadeusGhost, 2020-04-05:
试下pr #4184 ，我编译机硬盘满了得清下才能测。

nealwx, 2020-04-05:
我回退到4/3 号 d0922b9398eecddff7aa751dcbe3b229cfbc033f 也是有同样的问题。
平台是x86

ZForest000, 2020-04-05:
> 试下pr #4184 ，我编译机硬盘满了得清下才能测。
默认配置编译成功了

ZForest000, 2020-04-05:
> 我回退到4/3 号 [d0922b9](https://github.com/coolsnowwolf/lede/commit/d0922b9398eecddff7aa751dcbe3b229cfbc033f) 也是有同样的问题。
> 平台是x86

我看大雕的 x86 Actions 编译是成功的

PR #4184 (2020-04-05): ar71xx: fix alloc_page_frag issue
cherry-pick commit from 819800ab4f90cf65dc62045a0f3d06cca57e437f , fix #4179
closed 2020-04-05 by coolsnowwolf